OctoPrint/OctoPrint #2859: [Request] Show the current baudrate

DominikPalo, 2018-10-21:
It would be nice if OctoPrint shows somewhere the baudrate of the conenction (when it is used in the auto mode) because now it is not possible (at least I didn't found a way how to) get the current baudrate.
[image]

foosel, 2018-10-30:
Partially implemented as a side effect of the fix/implementation of #2740. The commit above fully implements it that the current settings are now reflected in the selection boxes after a connection has been established:

[image]

Pushed to `maintenance`, soon `devel`, to be released with 1.3.10

foosel, 2018-11-14:
You aren't wrong and that should be implemented. The problem is that it works here, both of these were set to AUTO before connecting:

[image]

Does it work in safe mode? Have you tried clearing your browser's cache?

foosel, 2018-11-14:
Ooooh, wait. Did that connection happen through auto connect on server startup or through a manual connect?

DominikPalo, 2018-11-18:
It was a manual connect. See my video (recorded in the incognito mode, so it can't be caused by the browser's cache) https://gyazo.com/d9114fd8ac759e6e09ca217da04b08c2

foosel, 2018-12-10:
This not working under 1.3.10 is still unreproduced. Marking as such.

foosel, 2019-01-28:
@Edo78 exact steps with which you experience this issue. The more details the better. I still cannot reproduce this.

Edo78, 2019-01-28:
I did nothing special:

- Downloaded octopi
- flashed on the sd
- configured the wifi
- booted the Pi
- if I remember correctly on the first login it tells me there's an update so I update
- rebooted the Pi and reloaded in the browser
- connected to the printer with AUTO/AUTO
- keep seeing AUTO/AUTO since then

I even tried the incognito mode in chrome but I stile get AUTO/AUTO.

If I set the proper values then I see them, otherwise it's always been AUTO/AUTO

foosel, 2019-03-19:
Now that was weird. I couldn't reproduce it at all - until today. Now I finally could reproduce it and looking at the code have no idea why it ever worked for me to begin with. I'm... very confused by this 🤨

In any case, I hope the above commit now fixes it for good - it does for me. Already pushed to `maintenance`, soon `devel`, to be part of 1.3.11.

commit bc66d29d (2019-03-19): 🐛 Fix reporting back of active port & baudrate

Closes #2859
closed 2019-05-14 by foosel