OctoPrint/OctoPrint #2860: crash inside the serial connection loop

TheNore, 2018-10-23:
#### What were you doing?

1. Sent Print to server in Slic3r PE (1.41.1-beta+win64)
2. Printed successfully until Octoprint Froze. I restarted server via SSH terminal. Octoprint ran as normal.
3. Pi Di dnot need to be rebooted.

N/A

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

Has happened in safe mode - before this session.

#### Version of OctoPrint

1.3.9

#### Operating System running OctoPrint

Raspbian GNU/Linux 9 (stretch)

#### Printer model & used firmware incl. version
Prusa i3 MK3 3.4.1

#### Browser and version of browser, operating system running browser
chrome  70.0.3538.67

#### Link to octoprint.log
https://drive.google.com/open?id=151ZerDYFe3F9h1Q_bhGF34Jz6hIv0Vsf

#### Link to contents of terminal tab or serial.log
Serial.log not enabled
Terminal.log - https://drive.google.com/open?id=1M-nBye6dBAPjNIu1WGNw-QvJQHHPyH_sQZ77lRzVjC0

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2018-10-29:
So... this confuses me a lot:

```
2018-10-23 14:34:28,322 - octoprint.util.comm - ERROR - Something crashed inside the serial connection loop, please report this in OctoPrint's bug tracker:
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 1639, in _monitor
    self._handle_ok()
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 2087, in _handle_ok
    self._continue_sending()
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 2209, in _continue_sending
    elif job_active and self._send_from_job():
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 2653, in _send_from_job
    result = self._sendCommand(line, tags={"source:file", "filepos:{}".format(pos), "fileline:{}".format(lineno)})
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 2867, in _sendCommand
    enqueued_something = process(cmd, cmd_type, gcode, subcode, on_sent=on_sent, tags=tags) or enqueued_something
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 2842, in process
    if self._enqueue_for_sending(cmd, command_type=cmd_type, on_sent=on_sent, tags=tags):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 2893, in _enqueue_for_sending
    self._send_queue.put((command, linenumber, command_type, on_sent, False, tags), item_type=command_type, target=target)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/comm.py", line 3923, in put
    self._unblocked.wait()
AttributeError: SendQueue instance has no attribute '_unblocked'
```

because as visible here https://github.com/foosel/OctoPrint/blob/master/src/octoprint/util/comm.py#L3886 `SendQueue` in fact *does* have an attribute `_unblocked` - and if it didn't that would also immediately fall on its face right on connection attempts, not only sometime down the road.

Is there any way to intentionally reproduce this, any specific GCODE that preceeds it, something that might allow to debug into this and see closer what's going on?

Also, just to rule out any kind of weird version mismatch shenanigans, I see this is a self-installed version, how have you updated it?

TheNore, 2018-10-29:
I agree, I was betting that it was a plugin issue, I have narrowed it down all issues with them, and now i got this in the middle of a print I am not sure either. It is self installed and fully up to date. (1.3.9) I do not update the Pi OS as I have a 3.5" TFT screen mounted on my printer connected to the PI running touchUI. I use all of the built in update mechanisms for the octoprint system.  I have not received this error at all since this happened. I am a systems engineer, and VERY particular about stability on my machines. From what I remember, there was no custom or different GCODE, i was using a beta version of Slic3r at the time Igenerated the code. Could it have been a firmware or beta slic3r hiccup on the MK3? I have updated the MK3 and the slic3r twice since I have received this error, and like I said, nothing has crashed since then.

foosel, 2018-10-29:
> Could it have been a firmware or beta slic3r hiccup on the MK3?

Hm... unlikely. I'm not sure how either would be able to make the application think that an attribute on an internal class doesn't exist that indeed exist. I've seen such behaviour in the past when I manually upgraded stuff but for some reason still had left overs from old versions lying around, but that also didn't happen out of the blue but was immediately an issue, and as I said, you shouldn't even have been able to connect to the printer if this error was present from the get go. So something happened while the connection to the printer was already running, and that did some voodoo with the send queue instance. 

A plugin *could* mess with things here, but I'm not aware of any that touch the send queue (and they really shouldn't in any case). 

It's a bit of a mystery really.

TheNore, 2018-10-29:
Understood, If this happens again, what would you like me to do ?

foosel, 2018-10-30:
Ideally give me a full step by step recount of what you did just before, the more accurate the better. But other than that I have no idea right now. What I'd need to do to get to the bottom of this is to actually set a breakpoint in it and go there with a debugger, see what state the SendQueue is in at the time of this happening, maybe that might get me a clue. But that's not something that can easily be done in a production setup, hence I'd need a way to reproduce this first in order to be able to do that in my development setup here.

What could possibly also shed some light is a copy of your `/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/comm.py` and `/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint/util/comm.pyc` files - maybe something in there is weird and not how it should be. Might also make sense now, just to verify they look like they should.

TheNore, 2018-11-08:
I understand what you are saying as far as replicating the error in a lab
environment - I am a systems engineer and enjoy debugging and tweaking
these issues to perfection. Anything I can provide or test I will.
would you like me to send you the contents of those directories?

On another note, this happened again (Came to mid print stop and server
quit Pic attached) but there was a different error in the log - I have
since enabled the "tab order 0.5.0" plugin but see no error involving that.
in addition - i have not updated to 1.3.10RC1, and 192.168.111.2 is my host
computer that I have a web browser session open to octoprint

2018-11-07 20:36:18,368 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 20:51:18,370 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 21:06:18,372 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 21:21:18,375 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 21:36:18,377 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 21:51:18,379 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 22:06:18,380 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 22:07:43,123 - octoprint.server.util.sockjs - WARNING - Could
> not send message to client ::ffff:192.168.111.2: Existing exports of
> data: object cannot be re-sized
> 2018-11-07 22:07:43,124 - tornado.general - ERROR - Uncaught exception,
> closing connection.
> Traceback (most recent call last):
>   File
> "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/tornado/iostream.py",
> line 523, in _handle_events
>     self._handle_write()
>   File
> "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/tornado/iostream.py",
> line 847, in _handle_write
>     assert self._write_buffer_size >= 0
> AssertionError
> 2018-11-07 22:07:43,132 - tornado.application - ERROR - Exception in
> callback None
> Traceback (most recent call last):
>   File
> "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/tornado/ioloop.py",
> line 888, in start
>     handler_func(fd_obj, events)
>   File
> "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/tornado/stack_context.py",
> line 277, in null_wrapper
>     return fn(*args, **kwargs)
>   File
> "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/tornado/iostream.py",
> line 523, in _handle_events
>     self._handle_write()
>   File
> "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/tornado/iostream.py",
> line 847, in _handle_write
>     assert self._write_buffer_size >= 0
> AssertionError
> 2018-11-07 22:07:43,139 - octoprint.server.util.sockjs - INFO - Client
> connection closed: ::ffff:192.168.111.2
> 2018-11-07 22:07:44,334 - octoprint.server.util.sockjs - INFO - New
> connection from client: ::ffff:192.168.111.2
> 2018-11-07 22:21:18,381 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 22:32:21,253 - octoprint.util.comm - INFO - Finished in
> 20357.290 s.
> 2018-11-07 22:32:21,254 - octoprint.util.comm - INFO - Changing monitoring
> state from "Printing" to "Finishing"
> 2018-11-07 22:32:21,262 - octoprint.filemanager.analysis - INFO - Starting
> analysis of local:Pixel_Stand_mm_0.2mm_PLA_MK3.gcode
> 2018-11-07 22:32:21,410 - octoprint.util.comm - INFO - Changing monitoring
> state from "Finishing" to "Operational"
> 2018-11-07 22:32:33,942 - octoprint.filemanager.analysis - INFO - Analysis
> of entry local:Pixel_Stand_mm_0.2mm_PLA_MK3.gcode finished, needed 12.68s
> 2018-11-07 22:36:18,383 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 22:51:18,384 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 23:06:18,387 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 23:06:42,775 - octoprint.util.comm - INFO - Changing monitoring
> state from "Operational" to "Printing"
> 2018-11-07 23:06:42,793 - octoprint.util.comm - INFO - M110 detected,
> setting current line number to 0
> 2018-11-07 23:21:18,390 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 23:36:18,392 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 23:51:18,393 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-08 00:06:18,394 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3

foosel, 2018-11-08:
That particular error *could* be something like #1302 and/or a side effect of #2850. 

In any case it doesn't sound related to the print fail you observed, the print that happened during finished successfully according to your log, about half an hour later:

```
 2018-11-07 22:32:21,253 - octoprint.util.comm - INFO - Finished in 20357.290 s.
```

Anything else in that log around the time the failure occurred? If it didn't log the stack trace from your OP though I don't think it's the same issue, but possibly another symptom of a core issue underlying everything (possibly something as stupid as file system corruption).

Also, because I've heard from @kennethjiang that there are apparently weird crashs reported when OctoPrint Anywhere is installed: did you have that installed and enabled when this recent issue happened?

TheNore, 2018-11-08:
No i did not have that installed, I will send the full log below. Soi weird
I saw the finished declaration, but the printer was just sitting at its
last position in the picture with fans full on, and heatblock disabled due
to "timeout"

I agree, i dont think its the same issue. Is there a way to check if file
corruption has occurred?

2018-11-06 23:45:47,255 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 00:00:47,259 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 00:15:47,262 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 00:30:47,263 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 00:45:47,265 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 01:00:47,266 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 01:15:47,267 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 01:30:47,270 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 01:45:47,272 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 02:00:47,275 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 02:15:47,276 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 02:30:47,277 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 02:45:47,280 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 03:00:47,282 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 03:15:47,284 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 03:30:47,285 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 03:45:47,286 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 04:00:47,289 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 04:15:47,290 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 04:30:47,291 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 04:45:47,294 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 05:00:47,297 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 05:15:47,300 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 05:30:47,303 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 05:45:47,305 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 06:00:47,305 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 06:15:47,307 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 06:30:47,310 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 06:45:47,313 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 07:00:47,314 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 07:15:47,316 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
[... 423 lines cut ...]
> heartbeat <3
> 2018-11-07 22:51:18,384 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 23:06:18,387 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 23:06:42,775 - octoprint.util.comm - INFO - Changing monitoring
> state from "Operational" to "Printing"
> 2018-11-07 23:06:42,793 - octoprint.util.comm - INFO - M110 detected,
> setting current line number to 0
> 2018-11-07 23:21:18,390 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 23:36:18,392 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-07 23:51:18,393 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-08 00:06:18,394 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-08 00:21:18,396 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-08 00:36:18,397 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-08 00:51:18,398 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3
> 2018-11-08 01:06:18,399 - octoprint.server.heartbeat - INFO - Server
> heartbeat <3

foosel, 2019-03-08:
@TheNore and new findings here?

TheNore, 2019-03-11:
Thanks for the follow up Octopi+plugin + mk3 firmware upgrade resolved
this issue.

closed 2019-03-12 by foosel