OctoPrint/OctoPrint #2864: Octoprint crash anytime after some time

Trikos, 2018-10-30:
What were you doing?
I start my printer, turn on Octoprint, wait until it is on then connect to printer, load a file, start a print. After 10 mins more or less If I try do some operation into Octoprint it doesnt respond and refreshing the page the server is crashed and not respond!
This happen every time even if Im not printing after 10 mins it crash and I cant used it anymore until next restart of Raspberry!

What did you expect to happen?
That Octo stay always alive and be ready to serve me 💃 

What happened instead?
Octo dead

Did the same happen when running OctoPrint in safe mode?
I didnt try it, I will do.

Has happened in safe mode - before this session.
(TO TRY)

Version of OctoPrint
1.3.9

Operating System running OctoPrint

Printer model & used firmware incl. version
Anet A8, with Marvin 1.1.9

Browser and version of browser, operating system running browser
Chrome 70.0.3538.77 (Official Build ) (64 bit)

Link to octoprint.log

Link to contents of terminal tab or serial.log

Link to contents of Javascript console in the browser

I have read the FAQ.

ntoff, 2018-10-31:
you need to provide the logs, can't do much without logs.

Trikos, 2018-10-31:
> you need to provide the logs, can't do much without logs.

Ok, when I ccome back I upload the logs.
Octo logs is enough? Or you need the JavaScript and the serial logs also?

foosel, 2018-10-31:
> Octo logs is enough? Or you need the JavaScript and the serial logs also?

Everything that's asked for in the template please.

foosel, 2018-11-15:
@Trikos Ping? Still waiting for info here.