OctoPrint/OctoPrint #2871: [RC Feedback] Feedback on the 1.3.10rc1 Release Candidate

foosel, 2018-11-06:
Please provide *general* feedback on your experience with the 1.3.10rc1 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

If you run into any obvious bugs, please open a new ticket and follow ["How to file a bug report"](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report).

hashashin, 2018-11-06:
Hi @foosel, seems like there's a problem with the gcode viewer in this version:
2018-11-06 13:19:26,075 - tornado.access - WARNING - 403 GET /downloads/files/local/honor7A.gcode (2a01:c50e:221e:7d00:90e5:9d8e:5a18:154e) 13.35ms
This happens with any file, reverting to 1.3.9 make it work again.
Salud.

zeroflow, 2018-11-06:
Raspberry Pi 3 + MKS sBase: Didn't work, reported as [issue 2872](https://github.com/foosel/OctoPrint/issues/2872)

foosel, 2018-11-06:
@hashashin thanks, can confirm. Weird, I could still it was working last week and I didn't change the authentication stuff since then, at least not that I can remember. Will take a look.

@zeroflow will get back to you in the ticket.

ctgreybeard, 2018-11-06:
After upgrade the Connect/Disconnect button shows Disconnect even though the printer is not connected.

I ran the upgrade and immediately after the Reload the button says Disconnect even though the printer isn't yet connected.
[image]
Clicking the button does not connect to the printer. It seems non-functional.

A reboot of the Pi doesn't bring it back. I don't seem to be able to connect to the printer any more.

foosel, 2018-11-06:
@ctgreybeard what does the Terminal tab say? Also check your powering, I see an undervoltage indicator up there.

ctgreybeard, 2018-11-06:
The terminal tab is blank and not allowing input. The power indicator isn't there now. I have it powered via a 2.4A adapter and although I haven't checked it in a while I was watching for power problems after upgrading the adapter and I currently get "throttled=0x0" for vcgencmd.

foosel, 2018-11-06:
@ctgreybeard Anything in the JS console?

ctgreybeard, 2018-11-06:
Nothing that I can see. Some view models that don't bind.

foosel, 2018-11-06:
Could you please open a ticket on that and provide all the information that is requested (and check safe mode)? I cannot reproduce anything that sounds what you subscribe here, so I fear I need some logs, screenshots and all that, and that should go into its own ticket. Thanks!

ctgreybeard, 2018-11-06:
OK, I'll open a separate ticket a bit later today.

kmanley57, 2018-11-06:
I tried it on my Taz 6 about 4 hours ago and it would not connect to the printer, but the 3 Lulzbot Minis and a Taz 5 worked fine. So I reverted back for now on the Taz 6.

foosel, 2018-11-06:
@kmanley57 might be #2876, fixed for 1.3.10rc2

jneilliii, 2018-11-06:
Please blacklist MQTT Publish Plugin version 0.6.0 on this release, it crashed hard.  I think I have others but my pi died during testing and can't double-check until I get home to re-build. My suspicion is that the Tasmota-MQTT plugin may also be one causing problems.

foosel, 2018-11-06:
@jneilliii done

trendelkamp, 2018-11-06:
Besides #2876 everything is working fine for me.

JohnOCFII, 2018-11-06:
Other than #2876 and the GCode Viewer issue mentioned above, rc1 seems to be working well with my Prusa i3 MK3 with firmware 3.4.0.  Tested Print History Plugin, OctoSlack Plugin, Timelapse, file uploading, and one 23 minute print.

ntoff, 2018-11-07:
Upon reboot I get hit with an anonymous usage enable / disable wizardy thingy. Clicking either enable or disable does.... something? nothing? There's no feedback as to whether or not my clicks are working. I'm not sure if that's intended behaviour but it'd be nice to see a little "tracking disabled" or "tracking enabled" text box or something, anything.

ramsesiden, 2018-11-07:
I installed this version, I have Klipper installed raspberri pi3b+, once installed it couldn't connect anymore to mcu, timeout after timeout, even swapping port or changing port to tmp and usb001, never worked, I had to rollback to .09

foosel, 2018-11-07:
@ramsesiden probably #2872 - but just to confirm could you provide the terminal output or a `serial.log` from a successful connection under 1.3.9?

@ntoff hm... the question now is how that got there though... In any case, added a sanity check on that field for 1.3.10rc2.

@jneilliii done!

ntoff, 2018-11-07:
@foosel probably when I manually deleted the custombackground plugin after it broke. You know how I love to manually break stuff to see what happens. Just so we're clear, this also broke 1.3.9 (and probably every other version), it wasn't unique to 1.3.10. I only just discovered it after updating to 1.3.10rc1. I'd say it's probably 90% chance it was all my own doing.

arminth, 2018-11-07:
@foosel 
Hi Gina,
I can confirm the issue that 1.3.10rc1 can't connect to klipper /tmp/printer anymore. Will trry to roll back to 1.3.9 and get you the connect-log.

arminth, 2018-11-07:
Rollback done!
Here you are.

`Connecting to: /tmp/printerChanging monitoring state from "Offline" to "Opening serial port"Connected to: Serial<id=0x6e2922f0, open=True>(port='/tmp/printer', baudrate=250000, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitorChanging monitoring state from "Opening serial port" to "Connecting"Send: N0 M110 N0*125Recv: okChanging monitoring state from "Connecting" to "Operational"`

foosel, 2018-11-07:
@arminth thanks! That confirms it then that it's #2872 and already fixed for rc2. Thanks! 

The only thing that confuses me that I've definitely seen successful connects to Klipper in the new tracking stats... Only 3 instances, but that's around 3 more than I would expect if Klipper *never* sends a `start` message (because then I'd never be able to detect Klipper in the first place). Odd. In any case, will be fixed in rc2.

ramsesiden, 2018-11-07:
> @ramsesiden probably #2872 - but just to confirm could you provide the terminal output or a `serial.log` from a successful connection under 1.3.9?
> 
> @ntoff hm... the question now is how that got there though... In any case, added a sanity check on that field for 1.3.10rc2.
> 
> @jneilliii done!

do you want me to test again to pull rc2 ?

foosel, 2018-11-07:
@ramsesiden no, just against 1.3.9 stable would suffice. I just want to see how your firmware behaves on connection to confirm that it's indeed not sending start.

ramsesiden, 2018-11-07:
Changing monitoring state from "Operational" to "Offline"
Connecting to: /tmp/printer
Connection closed, closing down monitor
Changing monitoring state from "Offline" to "Opening serial port"
Connected to: Serial<id=0x66b0bbf0, open=True>(port='/tmp/printer', baudrate=250000, bytesize=8, parity='N', stopbits=1, timeout=15.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from "Opening serial port" to "Connecting"
Send: N0 M110 N0*125
Recv: ok
Changing monitoring state from "Connecting" to "Operational"
Send: N0 M110 N0*125
Recv: ok
Send: N1 M115*39
Recv: ok FIRMWARE_VERSION:v0.6.0-583-g37b9a24 FIRMWARE_NAME:Klipper
Send: M21
Recv: ok
Send: M105
Recv: ok T0:25.2 /0.0 B:25.5 /0.0

foosel, 2018-11-07:
Yep, ok, that confirms #2872 as well, thanks!

ThaliaFromPrussia, 2018-11-07:
* Everything works fine so far. Only that I am missing the favicon.

[image]
Octoprint 1.3.9__________________________Octoprint 1.3.10rc1

* I really love the new Pi Support Plugin!

* Just tried the backup function. Worked. When I download the file, I get this. It was downloaded successfully. Is this intended?

[image]

Thank you for all your work you put in 🐙 -print! 💚

foosel, 2018-11-08:
@ThaliaFromPrussia What browser is this? Might be that it has issues with the SVG favicon (IE does I think, switching back to the "uncolored one" should work then).

> * Just tried the backup function. Worked. When I download the file, I get this. It was downloaded successfully. Is this intended?

Hm... nah, that is now intended. I'll try to reproduce this.

EddyMI3d, 2018-11-08:
Reffering to the favicon issue, I found this: https://caniuse.com/#feat=link-icon-svg
For I'm using Waterfox (Firefox), footnote 4 seems to be the problem.

foosel, 2018-11-08:
Ah, good point... I'll take care of that.

foosel, 2018-11-08:
@ThaliaFromPrussia @EddyMI3d the missing favicon in FF will be fixed in RC2, see #2879.

foosel, 2018-11-08:
@ThaliaFromPrussia I can't reproduce the reload prompty on backup download - can you reproduce it reliably? Otherwise my money would be on some fluke...

FanDjango, 2018-11-08:
Hi @foosel: 
Re issue #2873: GCODE viewer not functioning due to 403 when trying to load the GCODE file: Just for testing I made the change you published in module flask.py and can confirm that the GCODE viewer then works. BUT: This breaks being able to download logs, using the firmware update etc. Are you sure that the fix is complete?

foosel, 2018-11-08:
@FanDjango I noticed this myself earlier today and pushed a fix for that in commit 1667f9bd520eeb5d6b8700686ba4ff5bcc07320e 😄

FanDjango, 2018-11-08:
@foosel: I added that further change (seems logical) and the issue is resolved.

FanDjango, 2018-11-08:
Running ANET A8 + Marlin Bugfix 2.0 from 2018-11-06 and continuing the testing.
Firmware Updater Plugin no longer works:
2018-11-08 17:27:19,577 - octoprint.plugins.firmwareupdater - ERROR - Flashing failed. Unexpected error.
Traceback (most recent call last):
  File "/home/linaro/OctoPrint/venv/local/lib/python2.7/site-packages/octoprint_firmwareupdater/__init__.py", line 282, in _flash_avrdude
    p = sarge.run(avrdude_command, cwd=working_dir, async=True, stdout=sarge.Capture(), stderr=sarge.Capture())
  File "/home/linaro/OctoPrint/venv/local/lib/python2.7/site-packages/sarge/__init__.py", line 1462, in run
    p.run(input=input, async_=async_)
  File "/home/linaro/OctoPrint/venv/local/lib/python2.7/site-packages/sarge/__init__.py", line 1071, in run
    self.run_node(node, input=input, async_=False)
  File "/home/linaro/OctoPrint/venv/local/lib/python2.7/site-packages/sarge/__init__.py", line 1187, in run_node
    result = getattr(self, method)(node, input, async_)
  File "/home/linaro/OctoPrint/venv/local/lib/python2.7/site-packages/sarge/__init__.py", line 1333, in run_command_node
    node.cmd.run(input=input, async_=async_)
  File "/home/linaro/OctoPrint/venv/local/lib/python2.7/site-packages/sarge/__init__.py", line 655, in run
    self.process = p = Popen(self.args, **self.kwargs)
TypeError: __init__() got an unexpected keyword argument 'async'

foosel, 2018-11-08:
That's something covered in the "Heads-up" in the release notes - the plugin needs an update due to a backwards incompatible change in a dependency I needed to update. Best open a ticket on the plugin's repository.

hamster65, 2018-11-08:
Cool! I have been waiting for the new version :) Thanks!!

The first update attempt did not work.
[updatelog.log](https://github.com/foosel/OctoPrint/files/2562624/updatelog.log)

The second try finished without errors. After a full reboot, the webpage loads and it seems to be connected, but all controls are disabled. The connection dialogue stays visible, the button reads "Trennen" (disconnect), but I cannot click it. The terminal tab shows no messages.

Could this be related to those messages?
tornado.access - WARNING - 409 POST /api/printer/command (127.0.0.1) 54.01ms

foosel, 2018-11-08:
> The first update attempt did not work.

The log appears to be incomplete 🤔

> The second try finished without errors. After a full reboot, the webpage loads and it seems to be connected, but all controls are disabled. The connection dialogue stays visible, the button reads "Trennen" (disconnect), but I cannot click it. The terminal tab shows no messages.

Already covered in the list of "known issues" right in the initial post ;) See #2876

duncanlovett, 2018-11-08:
Since upgrading, I cannot connect to octoprint any more. I get the message:

The OctoPrint server is currently not running

If you just started up your Raspberry Pi, please wait a couple of seconds, then try to refresh this page.

If the issue persists, please log into your Raspberry Pi via SSH and check the following:

Verify that the process is running: ps -ef | grep -i octoprint | grep -i python should show a python process:
pi@octopi:~ $ ps -ef | grep -i octoprint | grep -i python
pi 1441 1 6 11:12 ? 00:00:15 /home/pi/oprint/bin/python 
/home/pi/oprint/bin/octoprint --host=127.0.0.1 --port=5000
If it isn't, the question is why. Take a look into ~/.octoprint/logs/octoprint.log, there might be an error logged in there that helps to determine what's wrong.
You might also want to try if you can restart the server (if no obvious error is visible): sudo service octoprint restart.
If all that doesn't help to trouble shoot the issue, you can seek support on the OctoPrint Community Forum. Please provide your OctoPi and OctoPrint versions as well as your octoprint.log and explain what you already tried and observed as detailed as possible.

hamster65, 2018-11-08:
> The log appears to be incomplete thinking
I copied everything I thought relevant out of my terminal window.

> Already covered in the list of "known issues" right in the initial post ;) See #2876
Ups, I indeed missed that. But I think my problem is different: I was able to connect with the OctoClient iPhone App. Its status said "detecting baudrate" for a long time and then changed to "Operational". I see the temperatures and I can home. Serial.log shows communications. But at the same time, the octoprint webinterface still shows the connection dialog and it appears to be offline. When I click on "Trennen", it disconnects but again nothing changes on the webinterface.

foosel, 2018-11-08:
@duncanlovett sorry, but I need more than that. E.g. the results of the steps suggested on that very page you quoted. And the logs. "Doesn't work anymore" isn't going to solve this, especially considering that I'm currently seeing around 380 instances online that enabled the new tracking plugin and where the update apparently worked fine.

@hamster65 Please don't copy-paste a log from the output in your SSH client, download the full log file from the server ;) You can do that also through the UI of OctoPrint itself, when it's running, otherwise use an SCP client (e.g. WinSCP under Windows).

> But I think my problem is different: I was able to connect with the OctoClient iPhone App

Doesn't sound like a different issue at all since the problem lies with the *web interface*, not the backend. Any API clients like e.g. such an app won't be affected.

hamster65, 2018-11-08:
That explains it, I guess. Logs attached.
[logs.zip](https://github.com/foosel/OctoPrint/files/2563225/logs.zip)

foosel, 2018-11-08:
@hamster65 Thanks.

So... funny thing... this error there on the first attempt, I have code in OctoPrint to detect it. However:

```
2018-11-08 16:48:37,391 ! OSError: [Errno 2] Datei oder Verzeichnis nicht gefunden: '/home/pi/oprint/local/lib/python2.7/site-packages/sarge-0.1.4-py2.7.egg'
```

I did not expect German locale in that error message, so the mechanism in place to detect this error and work around it doesn't take hold ;) That's not a new issue in 1.3.10 though, and I usually keep the fixes in newer RCs to regressions. I'll have to think a) how to detect this kind of error reliably without relying on localized language and b) whether I dare to put a potential fix into rc2 or not.

In any case, funny bug.

arminth, 2018-11-08:
> @arminth thanks! That confirms it then that it's #2872 and already fixed for rc2. Thanks!
>
> The only thing that confuses me that I've definitely seen successful connects to Klipper in the new [...]

Gina,

maybe it's because we are using an MKS GEN 1.4  running Klipper?
I just received a FYSETC F6 board. If I find the time over the weekend, I will install Klipper, do a test with it and let you know.
Tschüssi!
Armin

GhostlyCrowd, 2018-11-08:
I am getting this for some reason.

`T0 reported as invalid, reverting to T0`

T0 exists in my firmware and i only have one

foosel, 2018-11-09:
**Anyone running into any kind of issues, I can't look into them if you don't [open a proper bug report](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report) for them**. That means logs (`octoprint.log`, JS console output *and if the printer is involved also* `serial.log`), steps to reproduce, your hardware, all that. In short, [a proper fully filled out bug report](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report). Just throwing "X doesn't work" at me won't allow me to look into that, and I'm starting to feel like a broken record having to constantly reiterate this 😕 I want to fix anything that is observed in this RC, that is the whole point of putting it out in the first place, but *you* need to help me with that and not have me pull every tiny bit of information out of your nose...

ThaliaFromPrussia, 2018-11-09:
> 
> 
> @ThaliaFromPrussia I can't reproduce the reload prompt on backup download - can you reproduce it reliably? Otherwise my money would be on some fluke...

[image]
It did it again and also after a full restart. I wanted to download the log file and it did it again.
In safe mode it didn't.
[octoprint.log](https://github.com/foosel/OctoPrint/files/2565357/octoprint.log)

foosel, 2018-11-09:
> In safe mode it didn't.

Then it's one of your installed third party plugins - if an issue can't be reproduced in safe mode but can when not in safe mode, in 99% of cases a third party plugin is to blame (that's btw also why the bug template asks whether an issue reproduces in safe mode - if it doesn't I usually can't fix it since it's caused by a third party plugin and the author of THAT needs to fix it ;))

benlye, 2018-11-09:
Looks like I need to update the firmware updater plugin due to the Sarge change.  Seems simple enough, but I guess the best thing for compatibility is to use `octoprint.util.commandline.CommandlineCaller` and it looks like that class has no documentation.  Is there an example somewhere I could take a look at?

foosel, 2018-11-09:
@benlye oh, thanks for the heads-up, that lack of documentation should be fixed. In the meantime take a look at the [software update plugin](https://github.com/foosel/OctoPrint/blob/master/src/octoprint/plugins/softwareupdate/updaters/update_script.py) (search for `CommandLineCaller` and `caller`), it's using the class for running update scripts, the [PipCaller subclass](https://github.com/foosel/OctoPrint/blob/master/src/octoprint/util/pip.py) and its use by the [plugin manager plugin](https://github.com/foosel/OctoPrint/blob/master/src/octoprint/plugins/pluginmanager/__init__.py) for installing/uninstalling plugins (search for `LocalPipCaller` and `self._pip_caller`), or the [plugin dev cli](https://github.com/foosel/OctoPrint/blob/master/src/octoprint/cli/dev.py) that uses it for some stuff (search for `CommandLineCaller` and `self.caller`).

It has the advantage that a lot of the boilerplate needed to not only run commands but actually do something with their output (e.g. forwarding it to the frontend) is already taken care of. Downside is that if you need to run something non-blocking, you'll have to wrap it in your own thread (I might add another parameter to work around that though). Otherwise it just forwards any additional keyword args to sarge which in turn forwards stuff to the internally used Popen.

EddyMI3d, 2018-11-09:
@ThaliaFromPrussia: I had that issue too and found out that in my case it is this: https://github.com/fabianonline/OctoPrint-Telegram

ThaliaFromPrussia, 2018-11-10:
@EddyMI3d Thank you. I thought about it as well because it's no longer working every time.

duncanlovett, 2018-11-10:
It's ok I just flashed the sd card back to the previous version and all works fine again. I'll try the newer version next week when I have more time/

Cheers

d> @duncanlovett sorry, but I need more than that. E.g. the results of the steps suggested on that very page you quoted. And the logs. "Doesn't work anymore" isn't going to solve this, especially considering that I'm currently seeing around 380 instances online that enabled the new tracking plugin and where the update apparently worked fine.

tedder, 2018-11-10:
Has anyone been able to reproduce the "[can't get past the 'enable or disable anonymous usage' page](https://discourse.octoprint.org/t/anonymous-usage-setup/5151)" issue? I don't see a ticket, so I assume not.

tech-rat, 2018-11-11:
Upgrade smooth on Raspberry Pi 3 and Prusa MK3
Have the error where GCODE viewer won't load.
My Octoprint complains on each print that movement was detected but a G90/91 command was not detected however my output from slic3r has G90 near the top so I assume its a bug.

Thanks
Tech-Rat

foosel, 2018-11-12:
> My Octoprint complains on each print that movement was detected but a G90/91 command was not detected however my output from slic3r has G90 near the top so I assume its a bug.

That doesn't sound like anything that core OctoPrint does, sounds more like Octolapse.

racenviper, 2018-11-12:
RC has been working fine except for gcode viewer, this does not affect me as most of my prints exceed the max size for the viewer.

tech-rat, 2018-11-12:
> > My Octoprint complains on each print that movement was detected but a G90/91 command was not detected however my output from slic3r has G90 near the top so I assume its a bug.
> 
> That doesn't sound like anything that core OctoPrint does, sounds more like Octolapse.

Yes, sorry Octolapse. I can activate it and when I click print it gives the error I mentioned. 
Takes the snapshots and renders when fine.

Tech-Rat

[4 more replies]

closed 2018-11-13 by foosel