OctoPrint/OctoPrint #2872: [1.3.10rc1] Timeout when connecting to printer

zeroflow, 2018-11-06:
#### What were you doing?

1. Update OctoPrint to 1.3.10rc1
2. Restart the server as prompted
3. Click "Connect"

Hint: Going back to 1.3.9 makes it work again.

#### What did you expect to happen?

Connection to the printer established

#### What happened instead?

Nothing / Serial timeout in the console after a few seconds

#### Did the same happen when running OctoPrint in safe mode?

Yes, safe mode shows the same behavior.

#### Version of OctoPrint

1.3.10rc1

#### Operating System running OctoPrint

OctoPi 0.14.0

#### Printer model & used firmware incl. version

Anycubic Kossel, MKS sBase board, Smoothieware (~ June 2018)

#### Browser and version of browser, operating system running browser

Google Chrome latest, Windows 10 Fall18 Update

#### Link to octoprint.log

https://gist.github.com/zeroflow/1efe8597188962995618d9ca0886fa34

#### Link to contents of terminal tab or serial.log

Changing monitoring state from "Offline" to "Detecting serial port"
Serial port list: ['/dev/ttyACM0']
Connecting to: /dev/ttyACM0
Changing monitoring state from "Detecting serial port" to "Opening serial port"
Connected to: Serial<id=0x69afabd0, open=True>(port='/dev/ttyACM0', baudrate=250000, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from "Opening serial port" to "Connecting"
Connection closed, closing down monitor
Changing monitoring state from "Connecting" to "Offline"

#### Link to contents of Javascript console in the browser

[image]

#### Screenshot(s)/video(s) showing the problem:

No applicable / or neccesary.

I have read the FAQ.

foosel, 2018-11-06:
Can you provide an `octoprint.log` and `serial.log` from a working connection attempt under 1.3.9?

zeroflow, 2018-11-06:
Uploaded here: https://gist.github.com/zeroflow/dc1f63f4009600b6acbe9bf7c25b1a08

foosel, 2018-11-06:
Do you have Settings > Serial Connection > Firmware & protocol > Wait for `start` on connect enabled by any chance?

zeroflow, 2018-11-06:
No, Wait for `start` on connect is not enabled.

foosel, 2018-11-06:
I can reproduce it, the missing `start` at connection init is the trigger. Shouldn't make stuff stall, but for some reason it does. On it.

foosel, 2018-11-06:
Fixed by the above commit, will be part of 1.3.10rc2

foosel, 2018-11-13:
1.3.10rc2 was just released.

closed 2018-11-13 by foosel

cmock, 2018-11-13:
I can confirm 1.3.10rc2 connects fine to my Klipper printer now (originally reported in #2875).

commit 34319547 (2018-11-06): Fix connection fail if fw doesn't send start

Closes #2872