OctoPrint/OctoPrint #2873: [1.3.10rc1] GCODE viewer not functioning due to 403 when trying to load the GCODE file

foosel, 2018-11-06:
### Problem

As reported by @hashashin in https://github.com/foosel/OctoPrint/issues/2871#issuecomment-436234068:

>Hi @foosel, seems like there's a problem with the gcode viewer in this version:
2018-11-06 13:19:26,075 - tornado.access - WARNING - 403 GET /downloads/files/local/honor7A.gcode (2a01:c50e:221e:7d00:90e5:9d8e:5a18:154e) 13.35ms
This happens with any file, reverting to 1.3.9 make it work again.
Salud.

foosel, 2018-11-06:
Fixed by the above commit, will be part of 1.3.10rc2

ChrisHeerschap, 2018-11-11:
Ah, looks like this is what I was running into. Will confirm with RC2.

foosel, 2018-11-13:
1.3.10rc2 was just released.

closed 2018-11-13 by foosel

ChrisHeerschap, 2018-11-13:
Glorious! So nice to have the GCode viewer back, thank you!!

commit 4caf4ef0 (2018-11-06): Fix GCODE viewer not being able to load files

Side effect of the ForcedLogin plugin and a bug in
get_flask_user_from_request.

Closes #2873