OctoPrint/OctoPrint #2874: Trouble running in venv on ArchLinux

HarlemSquirrel, 2018-11-06:
I am trying to run OctoPrint for development and followed the instructions on the main README. When attempting to run the server I get a number of `TypeError: character mapping must return integer, None or unicode` errors.

I am new to Python development so I am sure that I missed something. Here's my terminal output:

```
➤  virtualenv2 venv                 
New python executable in /home/hs/code/OctoPrint/venv/bin/python2
Not overwriting existing python script /home/hs/code/OctoPrint/venv/bin/python (you must use /home/hs/code/OctoPrint/venv/bin/python2)
Installing setuptools, pip, wheel...done.

➤  venv/bin/pip install .           
Processing /home/hs/code/OctoPrint
Collecting flask<0.11,>=0.10.1 (from OctoPrint==1.3.9)
  Using cached https://files.pythonhosted.org/packages/db/9c/149ba60c47d107f85fe52564133348458f093dd5e6b57a5b60ab9ac517bb/Flask-0.10.1.tar.gz
Collecting Jinja2<2.9,>=2.8.1 (from OctoPrint==1.3.9)
  Using cached https://files.pythonhosted.org/packages/67/ea/92b1d9d8f2dc43302df7f5271b9500bbfc237386782343561a5f62beb306/Jinja2-2.8.1-py2.py3-none-any.whl
Collecting werkzeug<0.9,>=0.8.3 (from OctoPrint==1.3.9)
  Using cached https://files.pythonhosted.org/packages/a1/1c/02e6127ec997c08266b03d7fd340ba7675dafe0e28886a569664fd019e56/Werkzeug-0.8.3.tar.gz
Collecting tornado==4.5.3 (from OctoPrint==1.3.9)
  Using cached https://files.pythonhosted.org/packages/e3/7b/e29ab3d51c8df66922fea216e2bddfcb6430fb29620e5165b16a216e0d3c/tornado-4.5.3.tar.gz
Collecting PyYAML<3.11,>=3.10 (from OctoPrint==1.3.9)
  Using cached https://files.pythonhosted.org/packages/00/17/3b822893a1789a025d3f676a381338516a8f65e686d915b0834ecc9b4979/PyYAML-3.10.tar.gz
Collecting Flask-Login<0.3,>=0.2.11 (from OctoPrint==1.3.9)
  Using cached https://files.pythonhosted.org/packages/42/3c/ead3f50b8a39b6dd3499ae6f0f5b13b955130c92a7479a287e2e07921faf/Flask-Login-0.2.11.tar.gz
Collecting Flask-Principal<0.4,>=0.3.5 (from OctoPrint==1.3.9)
  Using cached https://files.pythonhosted.org/packages/47/fa/932178a740e699a1072aa1e73dc99dee559b4500cd0ab3c0d98931589843/Flask-Principal-0.3.5.tar.gz
Collecting Flask-Babel<0.10,>=0.9 (from OctoPrint==1.3.9)
  Using cached https://files.pythonhosted.org/packages/24/6b/dd9031c446e05600b740086274a9939799f26401b8b96e124ccf71e6a9b2/Flask-Babel-0.9.tar.gz
Collecting Flask-Assets<0.11,>=0.10 (from OctoPrint==1.3.9)
  Using cached https://files.pythonhosted.org/packages/95/de/88eb544c93313031e7b85758a1443730b92e7b1d01ec7222ddfea5f382e7/Flask-Assets-0.10.tar.gz
Collecting markdown<2.7,>=2.6.9 (from OctoPrint==1.3.9)
  Using cached https://files.pythonhosted.org/packages/6d/7d/488b90f470b96531a3f5788cf12a93332f543dbab13c423a5e7ce96a0493/Markdown-2.6.11-py2.py3-none-any.whl
Collecting pyserial<3.5,>=3.4 (from OctoPrint==1.3.9)
  Using cached https://files.pythonhosted.org/packages/0d/e4/2a744dd9e3be04a0c0907414e2a01a7c88bb3915cbe3c8cc06e209f59c30/pyserial-3.4-py2.py3-none-any.whl
Collecting netaddr<0.8,>=0.7.19 (from OctoPrint==1.3.9)
  Using cached https://files.pythonhosted.org/packages/ba/97/ce14451a9fd7bdb5a397abf99b24a1a6bb7a1a440b019bebd2e9a0dbec74/netaddr-0.7.19-py2.py3-none-any.whl
Collecting watchdog<0.9,>=0.8.3 (from OctoPrint==1.3.9)
  Using cached https://files.pythonhosted.org/packages/54/7d/c7c0ad1e32b9f132075967fc353a244eb2b375a3d2f5b0ce612fd96e107e/watchdog-0.8.3.tar.gz
Collecting sarge==0.1.4 (from OctoPrint==1.3.9)
  Using cached https://files.pythonhosted.org/packages/bc/9a/a715357f0f0a394edf964fb52b8b39551b629f4d1c996ade5f600fb3d278/sarge-0.1.4.tar.gz
Collecting netifaces<0.11,>=0.10.6 (from OctoPrint==1.3.9)
  Downloading https://files.pythonhosted.org/packages/70/dd/411bb4cda549136cf9ff5dbd4f33f898fd90b25ac73fd4ea42a3d3e8fd15/netifaces-0.10.7-cp27-cp27mu-manylinux1_x86_64.whl
Collecting pylru<1.1,>=1.0.9 (from OctoPrint==1.3.9)
  Downloading https://files.pythonhosted.org/packages/c0/7d/0de1055632f3871dfeaabe5a3f0510317cd98b93e7b792b44e4c7de2b17b/pylru-1.0.9.tar.gz
Collecting rsa<3.3,>=3.2.3 (from OctoPrint==1.3.9)
  Downloading https://files.pythonhosted.org/packages/df/40/7e146be0b15022a84992a485a8503ddb20bebd18a68340f3af5abde534f4/rsa-3.2.3-py2.py3-none-any.whl (44kB)
    100% |████████████████████████████████| 51kB 6.9MB/s 
Collecting pkginfo<1.5,>=1.4.2 (from OctoPrint==1.3.9)
  Downloading https://files.pythonhosted.org/packages/a3/fe/f32a48d48f40a7209be9825fba2566cab92364787cf37de2e08300dd6ce7/pkginfo-1.4.2-py2.py3-none-any.whl
Collecting requests<3,>=2.18.4 (from OctoPrint==1.3.9)
  Downloading https://files.pythonhosted.org/packages/f1/ca/10332a30cb25b627192b4ea272c351bce3ca1091e541245cccbace6051d8/requests-2.20.0-py2.py3-none-any.whl (60kB)
    100% |████████████████████████████████| 61kB 6.9MB/s 
Collecting semantic_version<2.5,>=2.4.2 (from OctoPrint==1.3.9)
  Downloading https://files.pythonhosted.org/packages/eb/d0/f9cd54744c360be93e802152fdc507b672a49909ea52be14cf3adba39280/semantic_version-2.4.2.tar.gz
Collecting psutil<5.5,>=5.4.6 (from OctoPrint==1.3.9)
  Downloading https://files.pythonhosted.org/packages/e3/58/0eae6e4466e5abf779d7e2b71fac7fba5f59e00ea36ddb3ed690419ccb0f/psutil-5.4.8.tar.gz (422kB)
    100% |████████████████████████████████| 430kB 7.7MB/s 
Collecting Click<6.3,>=6.2 (from OctoPrint==1.3.9)
  Downloading https://files.pythonhosted.org/packages/a0/90/0f5ea412a4d5b8e31cb514208e02824b45a8f4de52654733f0a862353fd9/click-6.2-py2.py3-none-any.whl (70kB)
    100% |████████████████████████████████| 71kB 11.2MB/s 
Collecting awesome-slugify<1.7,>=1.6.5 (from OctoPrint==1.3.9)
  Downloading https://files.pythonhosted.org/packages/34/39/79ef4e640c3651b40de7812f5fcd04698abf14de4f57a81e12b6c753d168/awesome-slugify-1.6.5.tar.gz
Collecting feedparser<5.3,>=5.2.1 (from OctoPrint==1.3.9)
  Downloading https://files.pythonhosted.org/packages/91/d8/7d37fec71ff7c9dbcdd80d2b48bcdd86d6af502156fc93846fb0102cb2c4/feedparser-5.2.1.tar.bz2 (192kB)
    100% |████████████████████████████████| 194kB 8.2MB/s 
Collecting chainmap<1.1,>=1.0.2 (from OctoPrint==1.3.9)
  Downloading https://files.pythonhosted.org/packages/6a/53/795ce18439a067eea1b233152985c0e67fd7dcc4edf102fb30f8cca6bad0/chainmap-1.0.2-py2.py3-none-any.whl
Collecting future<0.16,>=0.15.2 (from OctoPrint==1.3.9)
  Downloading https://files.pythonhosted.org/packages/5a/f4/99abde815842bc6e97d5a7806ad51236630da14ca2f3b1fce94c0bb94d3d/future-0.15.2.tar.gz (1.6MB)
    100% |████████████████████████████████| 1.6MB 6.0MB/s 
[... 449 lines cut ...]
    return _compile(pattern, flags, kwargs)
  File "/home/hs/code/OctoPrint/venv/lib/python2.7/site-packages/regex.py", line 486, in _compile
    parsed = _parse_pattern(source, info)
  File "/home/hs/code/OctoPrint/venv/lib/python2.7/site-packages/_regex_core.py", line 388, in _parse_pattern
    branches = [parse_sequence(source, info)]
  File "/home/hs/code/OctoPrint/venv/lib/python2.7/site-packages/_regex_core.py", line 413, in parse_sequence
    element = parse_paren(source, info)
  File "/home/hs/code/OctoPrint/venv/lib/python2.7/site-packages/_regex_core.py", line 823, in parse_paren
    subpattern = _parse_pattern(source, info)
  File "/home/hs/code/OctoPrint/venv/lib/python2.7/site-packages/_regex_core.py", line 388, in _parse_pattern
    branches = [parse_sequence(source, info)]
  File "/home/hs/code/OctoPrint/venv/lib/python2.7/site-packages/_regex_core.py", line 410, in parse_sequence
    sequence.append(parse_escape(source, info, False))
  File "/home/hs/code/OctoPrint/venv/lib/python2.7/site-packages/_regex_core.py", line 1186, in parse_escape
    return parse_property(source, info, ch == "p", in_set)
  File "/home/hs/code/OctoPrint/venv/lib/python2.7/site-packages/_regex_core.py", line 1341, in parse_property
    prop = lookup_property(prop_name, name, positive != negate, source)
  File "/home/hs/code/OctoPrint/venv/lib/python2.7/site-packages/_regex_core.py", line 1603, in lookup_property
    value = standardise_name(value)
  File "/home/hs/code/OctoPrint/venv/lib/python2.7/site-packages/_regex_core.py", line 1593, in standardise_name
    return ascii_upper("".join(ch for ch in name if ch not in "_- "))
  File "/home/hs/code/OctoPrint/venv/lib/python2.7/site-packages/_regex_core.py", line 1586, in ascii_upper
    return s.translate(upper_trans)
TypeError: character mapping must return integer, None or unicode
```

foosel, 2018-11-06:
Oooooh, that is fascinating! I'm seeing a similar error on the travis build and so far haven't been able to reproduce locally. That you are running into the same one makes it look like it isn't an issue isolated to Travis but something else. I'll look into it. Probably some dependency update maybe...

foosel, 2018-11-06:
Try `pip install regex==2018.11.3` before installing OctoPrint, that should fix it. The `regex` dependency pulled in by another third party dependency got an update over night, that that [seems to be broken](https://bitbucket.org/mrabarnett/mrab-regex/issues/301/typeerror-character-mapping-must-return).

I'm about to push a hotfix pinning it to that version too.

foosel, 2018-11-06:
Hotfixed on `master`, `staging/maintenance` and `maintenance`, further branches to follow as needed.

jneilliii, 2018-11-06:
Thanks @foosel, I was just installing on a fresh windows machine and was seeing the same thing, glad I found this post.

foosel, 2018-11-06:
Reopening so it is easier to find for people until the package is (hopefully) fixed upstream..

Solution see [here](https://github.com/foosel/OctoPrint/issues/2874#issuecomment-436280639).

foosel, 2018-11-07:
`regex-2018.11.7` got released that fixes this.

closed 2018-11-07 by foosel