OctoPrint/OctoPrint #2877: [1.3.10rc1] Guest mode no longer working

truglodite, 2018-11-06:
Guest mode was working, now it's not.

1. Updated to 1.3.10rc1, reboot, logout.
2. Now a basic login page appears, rather than the guest mode webpage with camera stream and basic info.

#### What did you expect to happen?
After logging out, I expect to see the guest mode webpage.

#### What happened instead?
I see the basic login page

#### Did the same happen when running OctoPrint in safe mode?
Yes

#### Version of OctoPrint
1.3.10rc1

#### Operating System running OctoPrint
Latest Raspbian and Octopi

#### Printer model & used firmware incl. version
Customized anet a8 w/ Marlin 2.xx from last week.

#### Browser and version of browser, operating system running browser
Chrome 70, Win10

#### Link to octoprint.log
https://pastebin.com/VQY75Njc

#### Link to contents of terminal tab or serial.log
*not enabled*

#### Link to contents of Javascript console in the browser
*meh, maybe later*

#### Screenshot(s)/video(s) showing the problem:
I don't want bullet holes in my monitor!

I have read the FAQ.

jubaleth, 2018-11-06:
Hi truglodite,

This would be a feature, not a bug :) There is a new plugin called ForcedLogin, it requires that a user logs in before they can see/do anything. If you wish to restore the old functionality, simply disable the plugin. 

This was also one of the points in the release notes and latest blog post =)

Cheers!

closed 2018-11-06 by jubaleth

truglodite, 2018-11-07:
Sorry jubaleth, I just didn't see it in the plugin list on the settings left pane, LOL!

jubaleth, 2018-11-07:
That happens, no need to apologize =)