OctoPrint/OctoPrint #2881: Doesn't work on python 3, yet happily installs on python 3 (and then fails)

fake-name, 2018-11-11:
Basically, there is no `python_requires` specified in octoprint's `setup.py`. As such, it will happily install on python versions where it won't run, and there are no other alternative version checks (`python_requires` was added in 2016). 

The short is you need to add `python_requires='<3'` to the main call to `setup()` in `setup.py`.

See https://stackoverflow.com/a/42792413/268006

Also, some clear messaging ("Octoprint **does not** support python 3", or something similar somewhere google would index it) would also help. Right now googling "octoprint python 3" gets me a bunch of old forum posts, so determining if python 3 is supported is a little challenging.

ntoff, 2018-11-11:
>Also, some clear messaging ("Octoprint does not support python 3", or something similar somewhere google would index it) would also help. 

Didn't look very hard, did you? https://github.com/foosel/OctoPrint#dependencies

>OctoPrint currently only supports Python 2.7.

fake-name, 2018-11-11:
@ntoff  - Apparently not.

closed 2018-11-26 by GitIssueBot