OctoPrint/OctoPrint #2884: Never pass the Please Login screen

gscorrea, 2018-11-11:
I just upgraded to 1.3.10RC1.
I am controlling 2 printers through same Raspberry PI/Octoprint.
Printer 1 on port 5000
Printer 2 on port 5001

When opening webpage for printer 1 (port 5000) I get the "Please Login" Screen, enter username/password and click on "Login In" and immediately get the same screen back, never getting to the printer screen.

When opening webpage for printer 2 (port 5001) I get the "Please Login" Screen, enter username/password and click on "Login In" and immediately get the regular page for the printer.

Please advise if there is already a reported bug and what kind of logs should I post to help in this issue.

gscorrea, 2018-11-11:
[octoprint.log](https://github.com/foosel/OctoPrint/files/2569944/octoprint.log)

closed 2018-11-11 by gscorrea

foosel, 2018-11-12:
> Please advise if there is already a reported bug

Nope, none so far...

> and what kind of logs should I post to help in this issue.

*Everything* that is stated in the [ticket template](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#what-should-i-include-in-a-bug-report) which was also prefilled into the new issue form before you deleted it ;)