OctoPrint/OctoPrint #2885: GUI reload asked by the GUI cause small hangs

MrKekson, 2018-11-12:
So my problem is, when i initially connect to the GUI, it causes small hangs while it's loading the js + webcam. after the load, the hangs did not occur (api calls are smaller on cpu load maybe).

So my question is, can we get rid of the please reload message? Most of the times, i can see everything working in the backgrounds eg: timers updating, graphs move, everything seems functional, but that message reload the entire GUI, causing small ~1 sec hangs, what will cause the head to stop, and leak small blobs if the buffer runs out of commands. I'm not sure about the cause, but i think it's related to auth or large file sending or something.

If it's loaded, then i experience no hangs, even with full hd cam stream, just while reloading.

I use klipper, what makes the raspi more cpu spike sensitive, but still, the api is functioning well, so i don't understand why do we have to reload the entire front-end.

foosel, 2018-11-15:
I'd honestly like to find out why you even get this request to reload when you do, because it should only ever popup if a disconnect from the websocket is detected *and* the version, installed plugins or active config has changed in the meantime:

``` javascript
            // if the version, the plugin hash or the config hash changed, we
            // want the user to reload the UI since it might be stale now
            var versionChanged = oldVersion !== VERSION;
            var pluginsChanged = oldPluginHash !== undefined && oldPluginHash !== self._pluginHash;
            var configChanged = oldConfigHash !== undefined && oldConfigHash !== self._configHash;
            if (versionChanged || pluginsChanged || configChanged) {
                showReloadOverlay();
            }
```

And to figure that out I'll need JS console and `octoprint.log` from a reproduction (and possibly some stuff to try down the road).

MrKekson, 2018-11-15:
I have a problem with the CR10 bed level helper plugin, where even if i
update it, it thinks it's still outdated? Can this be related to my
problem?

The problem with reproducing it is that it's pretty random. I can give you
a log when it happens, and if you give me some js, or tell me what should i
check while it happens, i can do that.

But your spinet gave me a few ideas, so i'll try to extract some data what
might help.

foosel, 2019-03-08:
@MrKekson have you been able to reproduce this and extract any kind of logs?

MrKekson, 2019-03-08:
Sadly no. I mean it does happens time to time, on my mobile, but mobile
phones break all connections sometimes to save battery.
On my PC since i've uninstalled the cr10 bed level helper plugin, it does
not require me to reload, so most likely something went wrong on the
install of that plugin.

I think you can close this issue, but if i experience it again, i will grab
the logs.

foosel, 2019-03-08:
Alright, will do so then. Thanks!

closed 2019-03-08 by foosel