OctoPrint/OctoPrint #2886: Firefox on Linux

azalanono, 2018-11-12:
#### What were you doing?
Octoprint is already running, my PC (openSUSE 13.2 x64, KDE, FireFox 63.0.1 (64-Bit), too.
I enter the URL to my octoprint server, get a connection, but I am NOT logged in automatically

1. ... Open web browser FireFox 63.0.1 (64-Bit)
2. ... Enter URL/IP of octoprint server
3. ... See that you are not logged in.

I check "Remember Login" and login again. This works (I am logged in), but on next connect I am again not logged in. This worked in the past!
Is for sure depending on the browser, too, but should be fixed.  The browser knows the login data. Wen clicking into the field of "username", I can select the saved one, and then the password is also inserted correctly.

#### What did you expect to happen?
Same behavior as 1 year ago (with older Firefox).

#### What happened instead?
See above
#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint
OctoPrint 1.3.9 auf OctoPi 0.13.0

#### Operating System running OctoPrint
Raspbian GNU/Linux 8 (jessie)

#### Printer model & used firmware incl. version
Marlin based  Not important to this issue.

#### Browser and version of browser, operating system running browser
openSUSE 13.2 x64, FireFox 63.0.1 (64-Bit) on KDE

#### Link to octoprint.log
none

#### Link to contents of terminal tab or serial.log
none

#### Link to contents of Javascript console in the browser
none

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2018-11-12:
Please provide _all_ requested information. Without that information this ticket cannot be looked into and will be closed. Thanks.

azalanono, 2018-11-12:
Hello, foosel.
Very constructive.

What is a octoprint.log/serial log good for when the problem is in your WEB interface?
Tell me what commands I have to type in, then I will return you the output.

foosel, 2018-11-12:
It's all explained in the [contribution guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report). `octoprint.log` is always needed since it also contains information on the communication *with the UI*. Contents of the JS console is *always important in case of issues with the UI*. Checking in safe mode is *always important* to rule out any of your third party plugins to be at fault.

I can't run after this kind of information individually on each and every ticket, I'd not do anything else at this point if I did.

And seriously, check your attitude, because I've honestly have had it with your constant "do my bidding" approach and outright passive aggressive insults on IRC and don't need that here as well.

closed 2018-12-05 by foosel