OctoPrint/OctoPrint #2887: FR: Display if JavaScript errors

OutsourcedGuru, 2018-11-12:
#### Feature Request
As someone who helps others to troubleshoot their OctoPrint installation, it would be nice if they had some early warning of JavaScript errors.

Certainly, I can try to talk them through the process of visiting their browser's Developer's Console to look for these errors but sometimes for the new people, they don't understand.

The new undervoltage indicator in the upper menu area is similar to what I'm suggesting. Perhaps a visual Yield sign indicator (if JavaScript errors were seen) could be somehow bubble up to the menu. 

Hovering over the Yield sign indicator would include a suggestion to check for errors in the Console log.

[image]

#### How do JavaScript errors affect users?
In most browsers, when a JavaScript error is seen in the bundled JavaScript code (from the plugin collection), it just stops running at this point.

The unfortunate side-effect then is that important functionality like the CLOSE/SAVE buttons at the bottom of the Settings dialog box are victims to small errors as seen above.

#### Incremental fix
Perhaps moving the button binding for these two buttons so that they occur before the bundled plugins JavaScript could be better (or possibly not since some of the functionality is potentially missing). And yet, it might help other features that are unnecessarily affected. _("One bad apple spoils the bushel.")_

#### Did the same happen when running OctoPrint in safe mode?
Almost always, the problem is in a third-party plugin so it would work of course for the user.

#### Link to contents of Javascript console in the browser
Is is possible to copy JavaScript errors to the `octoprint.log`? Or better, could they be copied to a new log file called `javascript-errors.log`?

foosel, 2018-11-13:
I was actually thinking about something like even forwarding the JS console to either `octoprint.log` or another server side `js.log` or something like this, to make it easier to access it in the future. Not sure yet though how best to go about it, haven't yet had time to look into possible libraries or general approaches for that.

BillyBlaze, 2018-11-13:
What about introducing a library like [loglevel](https://github.com/pimterry/loglevel) and use the plugin [loglevel-plugin-remote](https://github.com/kutuluk/loglevel-plugin-remote)?

foosel, 2018-11-13:
Ooooh, I wasn't aware of that plugin. I already use `loglevel` :)

Would need some "catchall" thing too though. Last time I checked that seemed to still be a bit iffy depending on the browser, but that was two or three years ago, so things might have changed.

OutsourcedGuru, 2019-03-04:
Take a look at this: [OctoPrint-ConsoleLog](https://github.com/OutsourcedGuru/OctoPrint-ConsoleLog).

I'm not completely convinced that a plugin is the way to go on this. Overclassing the console log should probably be done at the top of the packed third-party plugins' JavaScript and then reporting should probably happen at the end of that stack. Presumably that would be the best way of capturing all the errors.