OctoPrint/OctoPrint #2888: [Request] M600 event trigger

tech-rat, 2018-11-12:
#### What were you doing?
Prusa uses the M600 command at specific layer change(s) to swap filament. I would like a M600 event to be generated so that my custom event handler can be called which will notify me via Slack to change the filament.

1. ...
2. ...
3. ...

#### What did you expect to happen?
An event to be generated whan a M600 command is sent to the printer.

#### What happened instead?
Currently nothing

#### Did the same happen when running OctoPrint in safe mode?
N/A

#### Version of OctoPrint
Version 1.3.10rc1

#### Operating System running OctoPrint
Rasberry Pi running: Raspbian GNU/Linux 8

#### Printer model & used firmware incl. version
Prusa MK3

#### Browser and version of browser, operating system running browser
Chrome on Windows 10

#### Link to octoprint.log
N/A

#### Link to contents of terminal tab or serial.log
N/A

#### Link to contents of Javascript console in the browser
N/A

#### Screenshot(s)/video(s) showing the problem:
N/A

I have read the FAQ.

Thanks for your help and support. I love this project.

Tech-Rat

ianwiltdotcom, 2021-07-20:
This issue can be closed now I think

foosel, 2021-07-27:
It'll be closed when 1.7.0 is released which actually implements this :)

cp2004, 2021-10-11:
1.7.0 has been released 🎉

closed 2021-10-11 by cp2004