OctoPrint/OctoPrint #2893: OctoPrint cannot connect to Prusa i3 MK2.5 with firmware 3.4.1

yschroeder, 2018-11-14:
#### What were you doing?

1. Update firmware of Prusa i3 MK2.5 to version 3.4.1
2. Click `Connect` in Octoprint to connect to the printer

#### What did you expect to happen?

Octoprint connects to printer

#### What happened instead?

Connection to printer timed out

#### Did the same happen when running OctoPrint in safe mode?

Yes.

#### Version of OctoPrint

1.3.9

#### Operating System running OctoPrint

OctoPi

#### Printer model & used firmware incl. version

Prusa i3 MK2.5, Firmware 3.4.1

#### Browser and version of browser, operating system running browser

Firefox 63.0.1, Arch Linux 

#### Link to octoprint.log

https://gist.github.com/yschroeder/3f157f65d7407d5663f9f05fdd0cf191

#### Link to contents of terminal tab or serial.log

https://gist.github.com/yschroeder/0abd10a1cc06b7d479ac7499839ff9cc

#### Workaround

If the baudrate is set to `AUTO` the connection works:
https://gist.github.com/yschroeder/40ffabf06ef68c1fa0a0bc8fba4c65e8

I guess Octoprint sends the `M110` too quickly and the printer is not ready yet to accept commands. Or the printer is not fast enough... however I think Octoprint should be more resilient to connection delays (as it is already when probing baudrates).

The issue is discussed here:
https://github.com/prusa3d/Prusa-Firmware/issues/1232

foosel, 2018-11-15:
I've reclassified this as a feature request. Also see [my comment in the firmware ticket](https://github.com/prusa3d/Prusa-Firmware/issues/1232#issuecomment-439055030) as to why I'm kinda reluctant to change this on OctoPrint's side when it's once again a firmware quirk added in an upgrade that's making a workaround necessary.

matthew-humphrey, 2019-01-13:
I'm seeing this with 1.3.10 as well.