OctoPrint/OctoPrint #2895: Issue with daemon after upgrading to 1.3.9

shed909, 2018-11-15:
#### What were you doing?

1. Updated Ocotoprint via the web interface.

2. The service restarted and brought me to the "Octoprint not started" troubleshooting page.

3. SSH'd in. The service status is active (exited), and a restart does not resolve the issue.

4. Updated pip and re-installed with "~/oprint/bin/pip install https://get.octoprint.org/latest".

4. Ran "source ~/oprint/bin/activate", then "sudo service octoprint stop", then "~/oprint/bin/octoprint serve". (from what I could tell, the logs did not have any errors).

5. Lost SSH access. Was unable to access server via SSH for longer than a few seconds until i stopped the octoprint server by issuing a command via SSH.

6. Logged in and checked the logs:
https://gist.github.com/shed909/46ef1dfe136f8d47ebe92d0fa9de2b2f

Note that I also had a similar issue coming from 1.3.7 to 1.3.8.

#### What did you expect to happen?
For the service to update and restart automatically.

#### What happened instead?
The service was restarted in a errored state, causing broken pipes via SSH and the service to exit immediately after starting.

#### Did the same happen when running OctoPrint in safe mode?
Yes.

#### Version of OctoPrint
1.3.9 (updating from 1.3.8)

#### Operating System running OctoPrint
OctoPi version    : 0.15.1

#### Printer model & used firmware incl. version
LulzBot TAZ6

#### Browser and version of browser, operating system running browser
Chrome MacOS 10.14.1

#### Link to octoprint.log
https://gist.github.com/shed909/46ef1dfe136f8d47ebe92d0fa9de2b2f

#### Link to contents of terminal tab or serial.log
Same as log above.

#### Link to contents of Javascript console in the browser
Not applicable. 

#### Screenshot(s)/video(s) showing the problem:
[image]

I have read the FAQ.

foosel, 2018-11-15:
I have to admit that I'm struggling a bit to follow your bug report, so let's figure out if I understand the issue you are reporting here correctly.

  1. you updated OctoPrint through the web interface
  2. the server restarted
  3. but after the restart it wasn't accessible anymore

Correct?

If so, some follow-up questions/additional needed info:

  * Please provide the `plugin_softwareupdate_console.log` log file.
  * Are those logs you provided the *full* contents of your `octoprint.log`? Because that looks rather like snippets. If those are snippets you created, please provide the *full* log file. Don't edit it in any way.
  * > The service restarted and brought me to the "Octoprint not started" troubleshooting page.

    How did you know that the service restarted? Normally it would have prompted you to reload in OctoPrint after it detected the server being up again after the update - was this the case here? If not, what happened instead?
  * Please provide the output of `~/oprint/bin/octoprint serve --host 127.0.0.1` (the latter is what OctoPi actually fires up)
  * > Lost SSH access. Was unable to access server via SSH for longer than a few seconds until i stopped the octoprint server by issuing a command via SSH.

    OctoPrint doesn't have any kind of influence on your SSH server, other than possibly taking up some system resources. Do I understand this correctly that this was the case here, as in, you were still connected via SSH, but stuff got incredibly slow? Otherwise I don't understand how you could issue a command via SSH if you "lost SSH access".
  * Could you clarify what the SSH session in your screenshot is showing? Because that shows a running server and I'm confused how that relates to the browser tab next to it. I also don't know if that was from a manual `octoprint serve` command or not. Please clarify.
  * As mentioned, from the looks of that attached screenshot your server is running. Please run `netstat -tulpen` and provide its output to figure out if its actually listening for connections.
  * Have you done any kind of modifications to that OctoPi image? Installed additional software, `apt dist-upgrade`d the distro, installed new firmware, anything like that?
  * From what version were you updating? I'm guessing 1.3.8, but I'd like to get this verified.
  * Please run a [file system check](https://raspberrypi.stackexchange.com/a/61739).

shed909, 2018-11-15:
> I have to admit that I'm struggling a bit to follow your bug report, so let's figure out if I understand the issue you are reporting here correctly.
> 
> 1. you updated OctoPrint through the web interface
> 2. the server restarted
> 3. but after the restart it wasn't accessible anymore
> 
> Correct?

Correct.
> 
> If so, some follow-up questions/additional needed info:
> 
> * Please provide the `plugin_softwareupdate_console.log` log file.

plugin_software_console.log:
https://gist.github.com/shed909/ad806d2a6a273b672b73d6e6a1c9be27

> * Are those logs you provided the _full_ contents of your `octoprint.log`? Because that looks rather like snippets. If those are snippets you created, please provide the _full_ log file. Don't edit it in any way.

The logs I provided were from the point that I ran octoprint serve.
Here's a full one for you:
https://gist.github.com/shed909/b2e8b82c4a782c4dd7993ecef45fbaca

> * > The service restarted and brought me to the "Octoprint not started" troubleshooting page.
>
>
>   How did you know that the service restarted? Normally it would have prompted you to reload in [...]

I knew the service wasn't started as the service status was "active (exited)".

> * Please provide the output of `~/oprint/bin/octoprint serve --host 127.0.0.1` (the latter is what OctoPi actually fires up)

Here's the output from ~/oprint/bin/octoprint serve --host 127.0.0.1:
https://gist.github.com/shed909/2c27397286a5c5b3f34b5b3873d653fe

> * > Lost SSH access. Was unable to access server via SSH for longer than a few seconds until i stopped the octoprint server by issuing a command via SSH.
>
>
>   OctoPrint doesn't have any kind of [...]

Correct. Obviously OctoPrint doesn't directly influence this occurrence, rather it was a side effect that I included to help you understand the situation.
As I said I unable to access server via SSH for longer than a few seconds, but luckily that few seconds allowed me to issue a one-liner to stop the service.

> * Could you clarify what the SSH session in your screenshot is showing? Because that shows a running server and I'm confused how that relates to the browser tab next to it. I also don't know if that was from a manual `octoprint serve` command or not. Please clarify.

This was from a manual octoprint serve command. Eg:

```
source ~/oprint/bin/activate
sudo service octoprint stop
~/oprint/bin/octoprint serve 
```
As you say, it shows the server is running, but as soon as I issue the above, the Pi grinds to a holt until I can issue a one liner to stop the service quick enough via SSH.

On checking the octoprint service logs, this is what it shows:
```
journalctl -u octoprint -b
-- Logs begin at Thu 2016-11-03 17:16:42 UTC, end at Thu 2018-11-15 22:07:44 UTC. --
Nov 15 21:17:03 octopi systemd[1]: Starting LSB: OctoPrint daemon...
Nov 15 21:17:03 octopi systemd[1]: Started LSB: OctoPrint daemon.
Nov 15 21:17:03 octopi octoprint[382]: Starting OctoPrint Daemon: OctoPrint.
Nov 15 21:17:14 octopi systemd[1]: Stopping LSB: OctoPrint daemon...
Nov 15 21:17:15 octopi octoprint[631]: Stopping OctoPrint Daemon: OctoPrint.
Nov 15 21:17:15 octopi systemd[1]: Stopped LSB: OctoPrint daemon.
```
> * As mentioned, from the looks of that attached screenshot your server is running. Please run `netstat -tulpen` and provide its output to figure out if its actually listening for connections.

```
Proto Recv-Q Send-Q Local Address           Foreign Address         State       User       Inode      PID/Program name
tcp        0      0 0.0.0.0:8080            0.0.0.0:*               LISTEN      1000       970        424/./mjpg_streamer
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      0          10994      -
tcp6       0      0 :::8080                 :::*                    LISTEN      1000       971        424/./mjpg_streamer
tcp6       0      0 :::80                   :::*                    LISTEN      0          13319      -
tcp6       0      0 :::22                   :::*                    LISTEN      0          10996      -
tcp6       0      0 :::443                  :::*                    LISTEN      0          13320      -
udp        0      0 0.0.0.0:60914           0.0.0.0:*                           0          13321      -
udp        0      0 0.0.0.0:68              0.0.0.0:*                           0          10791      -
udp        0      0 0.0.0.0:45155           0.0.0.0:*                           108        1861       -
udp        0      0 0.0.0.0:5353            0.0.0.0:*                           108        1859       -
udp6       0      0 :::43728                :::*                                108        1862       -
udp6       0      0 :::5353                 :::*                                108        1860       -
```

> * Have you done any kind of modifications to that OctoPi image? Installed additional software, `apt dist-upgrade`d the distro, installed new firmware, anything like that?

This was a fresh image at the time that the update issue occurred.
Since then all I have done is updated pip.

> * From what version were you updating? I'm guessing 1.3.8, but I'd like to get this verified.

Yep. Updating from 1.3.8 to 1.3.9.

> * Please run a [file system check](https://raspberrypi.stackexchange.com/a/61739).
Output from running fsck twice:

```
sudo fsck /dev/mmcblk0p1
fsck from util-linux 2.29.2
fsck.fat 4.1 (2017-01-24)
/dev/mmcblk0p1: 163 files, 44283/86872 clusters
```

foosel, 2018-11-26:
I'm sorry, I have absolutely no clue. From your logs it doesn't look like anything is amiss. I cannot reproduce this in any shape or form either and have also not heard of any other instance of this problem. I'll leave this open an mark it as unreproduced, maybe someone else runs into similar issues and that will allow to figure out what's going on here, but that's the only thing I can offer right now.

foosel, 2018-11-26:
Ah, and of course this only comes to mind after hitting send on the above: One thing you could try is running with the `--debug` flag - maybe something in the debug output will indicate something.

shed909, 2018-11-29:
Thanks for all your work @foosel .
I've started with a fresh image, but I'll come back here if I see the issue occur again.

foosel, 2018-11-30:
Ok, closing this then for now. Can always be reopened.

closed 2018-11-30 by foosel