OctoPrint/OctoPrint #2896: [Request] Confirmation dialog when removing a plugin

yet-another-average-joe, 2018-11-15:
Hello, and thanks for this great piece of software !

A confirmation dialog box would be usefull when deleting or even deactivating a plugin.

ntoff, 2018-11-16:
Removing one I can see a confirmation being handy, but do you really need one when de-activating? I'm pretty sure there are warnings when you try to disable one of the system critical plugins, but as far as disabling 3rd party plugins, if you accidentally disable one you don't want to disable, just re-enable it.

Also if you're removing multiple plugins, a dialog for each would be super annoying so some way to suppress it, or even select multiple plugins for mass removal, would need to be thought up first.

ntoff, 2018-11-17:
Then perhaps you should post an issue to that plugin's github page, to see if the issue can be fixed.

yet-another-average-joe, 2018-11-17:
Yes, I did it already. It is TouchUI. I found a solution that works, at least for me.

A question : could a plugin display its own warning dialog box when the user flips the activation switch, or hits the delete button ? Can a plugin do something, or is it completely passive ? I searched the OctoPrint Wiki for some sort of deactivate/uninstall event that a plugin could handle, but I found nothing...

dovecode, 2019-05-28:
I think we can close this issue now that the changes are in `maintenance`, right (cf. 79558fd)?

foosel, 2019-05-28:
Actually, I'll mark it as solved and close it once 1.3.12 (the release that includes that change) is released.

dovecode, 2019-05-28:
Ah, ok. That makes sense. Thanks.

foosel, 2019-10-22:
1.3.12 has been released.

closed 2019-10-22 by foosel

yet-another-average-joe, 2019-10-22:
Thanks for this enhancement !