OctoPrint/OctoPrint #2897: reverse proxy with domaine name error with 1.3.10RC

devildant, 2018-11-16:
hello,
i use an reverse proxy for octoprint, in 1.3.9 no problem, but with 1.3.10RC i take this error : 
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/views.py", line 404, in index
    if plugin.will_handle_ui(request):
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/forcelogin/__init__.py", line 29, in will_handle_ui
    result = passive_login()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/util/flask.py", line 531, in passive_login
    additional_private=ipCheckTrusted)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/net.py", line 59, in is_lan_address
    subnets.add(to_ipnetwork(v6))
  File "/home/pi/oprint/local/lib/python2.7/site-packages/netaddr/ip/sets.py", line 293, in add
    addr = IPNetwork(addr)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/netaddr/ip/__init__.py", line 928, in __init__
    implicit_prefix, flags)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/netaddr/ip/__init__.py", line 823, in parse_ip_network
    raise TypeError('unexpected type %s for addr arg' % type(addr))
TypeError: unexpected type <type 'NoneType'> for addr arg

devildant, 2018-11-16:
[octoprint.log](https://github.com/foosel/OctoPrint/files/2589029/octoprint.log)

devildant, 2018-11-16:
i found that : https://github.com/drkjam/netaddr/issues/65

foosel, 2018-11-16:
Can you please provide the *full* `octoprint.log`? So far you only posted snippets. Please do not edit it, just upload it in full.

devildant, 2018-11-16:
yes of course, I just had to anonymize because my domain name was visible in the log.
[octoprint.log](https://github.com/foosel/OctoPrint/files/2589244/octoprint.log)

devildant, 2018-11-16:
I understood how to solve the problem.
the simple addition of ipCheck: true in the config.yml file fixes the problem

foosel, 2018-11-16:
Thanks. So this is a bit weird because the referenced line numbers in the stack trace don't match the source code in 1.3.10rc2. 

I can see what might be happening here though and fix that.

Could you shoot me the output of `python -c "import netifaces; print(repr([netifaces.ifaddresses(interface) for interface in netifaces.interfaces()]))"` from the python environment OctoPrint is running in (so in your case probably rather `~/oprint/bin/python -c "import netifaces; print(repr([netifaces.ifaddresses(interface) for interface in netifaces.interfaces()]))"`)? If the returned info contains sensitive data, please shoot me a private mail at gina -at- octoprint.org with it instead of redacting it, the info might be crucial to understand why that error arose).

> the simple addition of ipCheck: true in the config.yml file fixes the problem

With that you actually just disabled the ip check (it expects a dictionary here, not a single boolean, so you basically gave it invalid config data which it now ignores). You didn't fix the problem, you hid it, and hiding problems isn't the goal here ;)

devildant, 2018-11-16:
mmm i reinstall octoprint with the 1.3.10RC2 with that : 
~/oprint/bin/pip install https://github.com/foosel/OctoPrint/archive/1.3.10rc2.zip
i take same error with different stack : 

```
2018-11-16 12:55:29,090 - octoprint - ERROR - Exception on /api/login [POST]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/__init__.py", line 221, in login
    additional_private=s().get(["server", "ipCheck", "trustedSubnets"]))
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/net.py", line 59, in is_lan_address
    subnets.add(to_ipnetwork(v6))
  File "/home/pi/oprint/local/lib/python2.7/site-packages/netaddr/ip/sets.py", line 293, in add
    addr = IPNetwork(addr)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/netaddr/ip/__init__.py", line 928, in __init__
    implicit_prefix, flags)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/netaddr/ip/__init__.py", line 823, in parse_ip_network
    raise TypeError('unexpected type %s for addr arg' % type(addr))
TypeError: unexpected type <type 'NoneType'> for addr arg
```

the command : python -c "import netifaces; print(repr([netifaces.ifaddresses(interface) for interface in netifaces.interfaces()]))"
return an error : 

```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: No module named netifaces
```
for the other command, i will send you the result to your email.

foosel, 2018-11-16:
Weird, that's still a line number that doesn't make sense. The code at that location doesn't contain the cited command. Huh.

> the command : python -c "import netifaces; print(repr([netifaces.ifaddresses(interface) for interface in netifaces.interfaces()]))"
> return an error

Not surprising, you need to run it in OctoPrint's python environment - that's why I added the second one which takes care of this ;)

> for the other command, i will send you the result to your email.

Thank you, got it. Weirdly though I cannot make that data cause an error. In any case, I've made the whole `is_lan_address` function way more resilient against any kind of errors that arise inside of it, so with rc3 that should no longer happen. Since you are so far the only person who can reliably reproduce this with their setup, once RC3 hits, please upgrade to it, remove your `ipCheck` adjustment from `config.yaml` again, restart and check whether it behaves. It *should* also show you a notification that you are connecting from an external address, if it doesn't there is probably an error logged in `octoprint.log` in which case please share that too.

devildant, 2018-11-16:
> Weird, that's still a line number that doesn't make sense. The code at that location doesn't contain the cited command. Huh.
>
>
> Not surprising, you need to run it in OctoPrint's python environment - [...]

no problem, as soon as the RC3 will be available, I will test, and I will make a return on this post.
I encounter another problem with the return of the plugins. When I go through the reverse proxy, I have the call network, but at the level of the view, they does not updates, if RC3 I still have this problem, I will create another bug report

foosel, 2018-11-16:
No, please create it now ;) If there's even the chance of an issue I need to look into that now, I'd rather not have to keep putting out RCs solved issue by solved issue, it's a tremendous amount of work each time, so the goal is to keep the number of RCs as low as possible and that means fixing ALL issues that are observed per RC if possible.

devildant, 2018-11-16:
ha ok, I'm doing it right now :)

foosel, 2018-11-20:
@devildant I was just wondering, since you updated to `staging/maintenance` anyhow, could you check whether this issue is resolved for you?

devildant, 2018-11-20:
@foosel oops, i forgot to test this case: p i do it right after my current print

devildant, 2018-11-20:
@foosel , i removed ipCheck config, i get the staging/maintenance, and, i tried to login with my phone (3g), no error.
This seems good :)

foosel, 2018-11-20:
Perfect, marking this as solved then, will close it when RC3 is out.

louispires, 2018-11-23:
@devildant what is the workaround for this issue to get it working on RC2?

Should I just add the ipCheck: true to the config?

foosel, 2018-11-23:
Rather set `server.ipCheck.enabled` to `false`:

``` yaml
server:
  ipCheck:
    enabled: false
```

foosel, 2018-11-28:
1.3.10rc3 was just released.

closed 2018-11-28 by foosel