OctoPrint/OctoPrint #2898: no status plugins on main view 1.3.10RC2

devildant, 2018-11-16:
#### What were you doing?
just logged in to octoprint, all the plugin are blocked on the main page

#### What did you expect to happen?
see the checkbox check for the plugin or specifique status
#### What happened instead?
nothing the plugins status on main page is not update

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint

1.3.10RC2

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser
LAST CHROME, and safari on ios

#### Link to octoprint.log
2018-11-16 14:16:03,188 - octoprint.plugins.navbartemp - INFO - Checking SoC internal temperature
2018-11-16 14:16:18,812 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.1.10
2018-11-16 14:16:20,496 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.1.10
2018-11-16 14:16:22,082 - octoprint.plugins.sdswitchfastcopy - INFO - test mount OK
2018-11-16 14:16:22,876 - octoprint.plugins.sdswitchfastcopy - INFO - test mount OK
2018-11-16 14:16:23,868 - tornado.access - WARNING - 404 GET /auth/status?plugin=true (::ffff:192.168.1.10) 12.33ms
2018-11-16 14:16:25,693 - octoprint.server.util.sockjs - INFO - User devildant logged in on the socket from client ::ffff:192.168.1.10

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser
Starting dependency resolution...
packed_core.js?d307d348:15581 ... dependency resolution done
packed_core.js?d307d348:15885 Initial application setup done, connecting to server...
packed_core.js?d307d348:13640 Connected to the server
packed_core.js?d307d348:15863 Finalizing application startup
packed_core.js?d307d348:15747 Going to bind 41 view models...
packed_core.js?d307d348:15800 Did not bind view model TrackingViewModel to target #wizard_plugin_tracking since it does not exist
packed_plugins.js?8758b863:8203 checking 192.168.1.43
packed_core.js?d307d348:15800 Did not bind view model CuraViewModel to target #wizard_plugin_cura since it does not exist
packed_core.js?d307d348:15800 Did not bind view model SoftwareUpdateViewModel to target #softwareupdate_confirmation_dialog since it does not exist
packed_core.js?d307d348:15800 Did not bind view model SoftwareUpdateViewModel to target #wizard_plugin_softwareupdate since it does not exist
packed_core.js?d307d348:2409 User devildant logged in
packed_core.js?d307d348:15839 ... binding done
packed_plugins.js?8758b863:7340 check status
packed_core.js?d307d348:15859 Application startup complete
packed_core.js?d307d348:11122 ConfiguredLoggers has not changed. Not saving.
VM3294:1 GET http://192.168.1.47/auth/status?plugin=true 404 (NOT FOUND)
(anonymous) @ VM3294:1
send @ packed_libs.js?331a1bb5:9601
ajax @ packed_libs.js?331a1bb5:9207
(anonymous) @ packed_plugins.js?8758b863:22
setInterval (async)
AuthDockerViewModel.self.onStartupComplete @ packed_plugins.js?8758b863:21
callViewModelIf @ packed_core.js?d307d348:15092
(anonymous) @ packed_core.js?d307d348:15038
Pn @ packed_libs.js?331a1bb5:11208
(anonymous) @ packed_libs.js?331a1bb5:11229
callViewModelsIf @ packed_core.js?d307d348:15036
callViewModels @ packed_core.js?d307d348:15030
bindViewModels @ packed_core.js?d307d348:15842
setTimeout (async)
(anonymous) @ packed_core.js?d307d348:15881
fire @ packed_libs.js?331a1bb5:3269
fireWith @ packed_libs.js?331a1bb5:3399
done @ packed_libs.js?331a1bb5:9306
(anonymous) @ packed_libs.js?331a1bb5:9549
load (async)
send @ packed_libs.js?331a1bb5:9568
ajax @ packed_libs.js?331a1bb5:9207
OctoPrintClient.ajax @ packed_client.js?1e5e38ec:143
OctoPrintClient.get @ packed_client.js?1e5e38ec:156
OctoPrintSettingsClient.get @ packed_client.js?1e5e38ec:1240
SettingsViewModel.self.requestData @ packed_core.js?d307d348:4270
fetchSettings @ packed_core.js?d307d348:15868
setTimeout (async)
(anonymous) @ packed_core.js?d307d348:15938
fire @ packed_libs.js?331a1bb5:3269
fireWith @ packed_libs.js?331a1bb5:3399
done @ packed_libs.js?331a1bb5:9306
(anonymous) @ packed_libs.js?331a1bb5:9549
load (async)
send @ packed_libs.js?331a1bb5:9568
ajax @ packed_libs.js?331a1bb5:9207
OctoPrintClient.ajax @ packed_client.js?1e5e38ec:143
OctoPrintClient.ajaxWithData @ packed_client.js?1e5e38ec:152
OctoPrintClient.post @ packed_client.js?1e5e38ec:164
OctoPrintClient.postJson @ packed_client.js?1e5e38ec:177
OctoPrintBrowserClient.passiveLogin @ packed_client.js?1e5e38ec:555
LoginStateViewModel.self.requestData @ packed_core.js?d307d348:2397
onServerConnect @ packed_core.js?d307d348:15918
(anonymous) @ packed_core.js?d307d348:15936
fire @ packed_libs.js?331a1bb5:3269
fireWith @ packed_libs.js?331a1bb5:3399
deferred.(anonymous function) @ packed_libs.js?331a1bb5:3737
DataUpdater.self._onConnectMessage @ packed_core.js?d307d348:13644
(anonymous) @ packed_client.js?1e5e38ec:388
Pn @ packed_libs.js?331a1bb5:11208
(anonymous) @ packed_libs.js?331a1bb5:11229
OctoPrintSocketClient.propagateMessage @ packed_client.js?1e5e38ec:387
(anonymous) @ packed_client.js?1e5e38ec:482
(anonymous) @ packed_libs.js?331a1bb5:11225
_t @ packed_libs.js?331a1bb5:11213
(anonymous) @ packed_libs.js?331a1bb5:11224
(anonymous) @ packed_libs.js?331a1bb5:11229
onMessage @ packed_client.js?1e5e38ec:481
EventTarget.dispatchEvent @ packed_libs.js?331a1bb5:22527
(anonymous) @ packed_libs.js?331a1bb5:23240
SockJS._transportMessage @ packed_libs.js?331a1bb5:23238
EventEmitter.emit @ packed_libs.js?331a1bb5:22443
WebSocketTransport.ws.onmessage @ packed_libs.js?331a1bb5:25314

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2018-11-16:
> just logged in to octoprint, all the plugin are blocked on the main page
> [...]
> see the checkbox check for the plugin or specifique status
> [...]
> nothing the plugins status on main page is not update

I don't understand what you are talking about. What does "all the plugins are blocked on the main page" mean? What checkbox check? What plugin status on what main page? Please clarify and provide screenshots. Also: full `octoprint.log`, *no snippets please*.

> **Did the same happen when running OctoPrint in safe mode?**

Need that info.

> **Operating System running OctoPrint**

Need that too.

> LAST CHROME, and safari on ios

Please give me version numbers, not "latest" - latest can change at any given moment.

devildant, 2018-11-16:
[image]

foosel, 2018-11-16:
Where can I find these plugins to test myself?

devildant, 2018-11-16:
you can test with shutdown printer (sd switch is still in development :))

foosel, 2018-11-16:
I cannot reproduce this with shutdown printer currently. Checkbox gets populated. In general not a good way though to send settings via the push socket - that's for real time data, not for settings. Still, getting through fine here. 

Can you try disabling the force login plugin (it's bundled)? I'd like to know if disabling this fixes the issue for you.

devildant, 2018-11-16:
yes, if I disable the forced login plugin, everything works

devildant, 2018-11-16:
Thank you for your help :)
(sorry if i'm not clear sometimes. I'm not good in english)

foosel, 2018-11-16:
All good :)

I think I have an idea what's going on here. The problem is that the plugin fires off a plugin message before the socket is authenticated. With the force login plugin disabled that doesn't matter, but if it is enabled it swallows up any messages until things are authenticated. I'll look into solving this - while it would be better if the plugin actually utilized the REST API for fetching the data in question here instead of triggering a shove of the data down the socket via REST, it is still a compatibility issue with plugins that do that and hence should be tackled.

foosel, 2018-11-16:
> the concern is the other plugins, there are many who use this method: p

which is why I'm looking into this. The problem is that I cannot reliably reproduce the issue yet (seems to be a race condition too), so testing the fix will be a bit tricky.

devildant, 2018-11-16:
I can give you the plugins I use, together it produces the problem at home
tplink plugin
enclosure plugin
shutdownprinter plugin
navbar temperature plugin
automatic shutdown plugin
I have other plugin but these affect the main page

devildant, 2018-11-16:
do you want me to wait before fixing my "shutdown printer" plugin?

jneilliii, 2018-11-16:
Since one of those is my plugin I was curious if I'm not doing things right @foosel. I haven't seen any issues with the RC with them, but haven't specifically looked yet.

foosel, 2018-11-19:
So I just added two commits to the `staging/maintenance` branch that fix the problem at least for Shutdown Printer. I actually had two issues (hence two commits) caused by the introduction of Forcelogin, the other was that the server side socket object didn't even register itself with the plugin subsystem as message forwarder if no auth had yet been received on the socket, this is now fixed as well.

To an observer it should now basically behave as before, even with forcelogin enabled. Messages will be go out on the socket in the order they were sent. If forcelogin is enabled a message backlog of max. 100 items will make sure that any messages sent (by plugins, core won't send anything yet) before the user has authed on the socket will just be delayed.

@devildant if you want to checkout `staging/maintenance` and give this a whirl, that would be great.

devildant, 2018-11-19:
@foosel  I test this, tonight, as soon as I arrive at home: p

devildant, 2018-11-19:
Re @foosel , I tested the staging / branch maintenance, for my plugin, it works, but I still have problems with the plugin tplink.
the state of the plug is not up to date (it is held in the ON position when the plug is off).
normally we use a similar technique it seems to me, it should work.
One thing I do not understand is that the problem seems to occur if multiple plugins use the WebSocket (at page load).
separately, it seems to work

foosel, 2018-11-19:
Taking a look at the TP Smartplug plugin then. This abuse of the socket as a back channel when a perfectly fine REST request context is available and hence a regular HTTP response could just be sent really has to stop though ;)

Actually thinking about how to detect this and log about this code smell :P

devildant, 2018-11-19:
yes i see.
after I admit that it is tempting to use the websocket it avoids polling :p
on my side I fixed my two plugin, I use the websocket only when the realtime is necessary (during updates via API, or when it is information recovered server side)
I do not have push yet for shutdown printer, if you still want to test.
I remain available for tests, I run just a little print of 1 day tonight: p

foosel, 2018-11-19:
@devildant 

> I admit that it is tempting to use the websocket it avoids polling :p

Yes, but if you have a request on startup anyhow, it's just wrong on so many levels to not actually reply to that request but instead push the response to *every connected client* via the socket - which is what is happening here. 

--- 

*short mini rant incoming*

General rule of thumb: If you want to query data from the frontend, use REST. If you need to push data to the frontend when you don't have a request running, or *want to trigger the frontend to do some request* you use the socket. Do not use the socket as a replacement back channel for REST requests. Not only does that lead to a non functional API on your end that makes it incredibly difficult for third party clients to integrate with your plugins, it's also horribly to debug and way more fragile than just using the API for what it was made.

Also: Don't send commands to query data, that's what the GET endpoint is for. Commands on the SimpleApi should only be used for things that change state (e.g. to toggle some switch, abort an auto shutdown, ...), if you need to query data use the GET endpoint and *do not modify internal state* when replying to it. 

*that's it, you can open your eyes again*

---

RE that TPLink issue, can confirm, seems to be a bug in the frontend component of the plugin though - the "off" state gets pushed through the socket just fine, the icon briefly switches to red, then immediately switches back. I haven't yet found the source for this switch back but my money would be with the view model having a bug. Speaking of which...

@jneilliii as I just went through your code to try to figure out why that happens, you have a couple of issues in your viewmodel. If you declare something as observable, you need to make 100% sure to never compare it but rather the result of *calling* it (e.g. not `plug.emeter != data.emeter` but rather `plug.emeter() != data.emeter` or even better `plug.emeter() !== data.emeter` to avoid JS's attempts at type conversion). Also you never must set it but rather call it with the value to set (e.g. not `plug.emeter = data.emeter` but rather `plug.emeter(data.emeter)`). Otherwise you'll kill the observable, or compare a value with the function that is the observable, not the value stored in the observable.

jneilliii, 2018-11-19:
Thanks @foosel. I've been trying to transition the state out of settings, just haven't had the time to do so. The other part you mention about observables relative to comparing and setting was initially done the way you mention but the knockout started complaining about it not being a function and couldn't bind the view model.

foosel, 2018-11-20:
> So makes sense to use API GET for initial load, and socket for subsequent updates.

Exactly. At least if you are doing an `GET` request anyhow and hence have a "free" back channel. Or - depending on message size - it can also make sense to have a message on the socket trigger a GET request, this if for example how the `SettingsUpdate` handling works in OctoPrint.

foosel, 2018-11-28:
1.3.10rc3 was just released.

closed 2018-11-28 by foosel