OctoPrint/OctoPrint #2901: Broken after update 1.3.8->1.3.9 AnalysisQueue too many arguments

MarcusWolschon, 2018-11-16:
#### What were you doing?
1. check for updates on Octoprint 1.3.8 via web interface
2. perform the update to 1.3.9
3. perform reload as requested

#### What did you expect to happen?

1.3.9 web interface to apear

#### What happened instead?

Error page
"The OctoPrint server is currently not running"

#### Did the same happen when running OctoPrint in safe mode?

YES

#### Version of OctoPrint

1.3.9

#### Operating System running OctoPrint

OctoPi

#### Printer model & used firmware incl. version

Ultimaker II extended

#### Browser and version of browser, operating system running browser

Firefox
#### Link to octoprint.log

`2018-11-16 20:45:09,728 - octoprint.startup - INFO - ******************************************************************************
2018-11-16 20:45:09,730 - octoprint.startup - INFO - Starting OctoPrint 1.3.9
2018-11-16 20:45:09,731 - octoprint.startup - INFO - ******************************************************************************
2018-11-16 20:45:09,787 - octoprint.util.connectivity_checker - INFO - Connectivity changed from offline to online
2018-11-16 20:45:09,983 - octoprint.startup - INFO - Blacklist processing done
2018-11-16 20:45:10,026 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2018-11-16 20:45:12,126 - octoprint.plugin.core - INFO - Found 15 plugin(s) providing 15 mixin implementations, 13 hook handlers
2018-11-16 20:45:12,224 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2018-11-16 20:45:12,336 - octoprint.server - INFO - Intermediary server started
2018-11-16 20:45:12,337 - octoprint.plugin.core - INFO - Loading plugins from /home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins, /home/pi/.octoprint/plugins and installed plugin packages...
2018-11-16 20:45:12,632 - octoprint.plugin.core - INFO - Found 15 plugin(s) providing 15 mixin implementations, 13 hook handlers
2018-11-16 20:45:12,663 - octoprint.cli.server - ERROR - Uncaught exception
Traceback (most recent call last):
  File "/home/pi/oprint/bin/octoprint", line 9, in <module>
    load_entry_point('OctoPrint==1.3.9', 'console_scripts', 'octoprint')()
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/__init__.py", line 599, in main
    octo(args=args, prog_name="octoprint", auto_envvar_prefix="OCTOPRINT")
  File "/home/pi/oprint/lib/python2.7/site-packages/click-6.2-py2.7.egg/click/core.py", line 716, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/oprint/lib/python2.7/site-packages/click-6.2-py2.7.egg/click/core.py", line 696, in main
    rv = self.invoke(ctx)
  File "/home/pi/oprint/lib/python2.7/site-packages/click-6.2-py2.7.egg/click/core.py", line 1037, in invoke
    return Command.invoke(self, ctx)
  File "/home/pi/oprint/lib/python2.7/site-packages/click-6.2-py2.7.egg/click/core.py", line 889, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/oprint/lib/python2.7/site-packages/click-6.2-py2.7.egg/click/core.py", line 534, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/oprint/lib/python2.7/site-packages/click-6.2-py2.7.egg/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/cli/__init__.py", line 175, in octo
    ctx.invoke(serve_command, **kwargs)
  File "/home/pi/oprint/lib/python2.7/site-packages/click-6.2-py2.7.egg/click/core.py", line 534, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/oprint/lib/python2.7/site-packages/click-6.2-py2.7.egg/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/cli/server.py", line 188, in serve_command
    ignore_blacklist)
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/cli/server.py", line 109, in run_server
    octoprint_server.run()
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/server/__init__.py", line 284, in run
    analysisQueue = octoprint.filemanager.analysis.AnalysisQueue(analysis_queue_factories)
TypeError: __init__() takes exactly 1 argument (2 given)`

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2018-11-19:
Cannot reproduce. Looks like something got messed up in the code copy on the card, like some old bits are still running that have a different signature.

What version of OctoPi is this?

MarcusWolschon, 2018-11-19:
Where can I look up the version of the OctoPi distribution?
Is there a way to reinstall Octoprint itself without loosing custom modules and configuration?

foosel, 2018-11-19:
https://discourse.octoprint.org/t/how-can-i-find-out-the-version-of-octoprint-or-octopi-i-am-running/204

https://discourse.octoprint.org/t/how-do-i-backup-my-octoprint-settings-on-octopi/1489

MarcusWolschon, 2018-11-30:
pi@octopi:~ $ ~/oprint/bin/octoprint --version
octoprint, version 1.3.9

MarcusWolschon, 2018-11-30:
Do I have to do anything to
/home/pi/oprint/lib/python2.7/site-packages/octoprint/ 
?
I don't know what a site-package is.

foosel, 2018-12-02:
I asked for the Octo*Pi* version, not the Octo*Print* version. ```cat /etc/octopi_version```

Your installation doesn't live in `~/OctoPrint`, that's just the source folder. At this point I'd honestly try to just kill the install and redo it on the card:

```
rm ~/oprint
virtualenv ~/oprint
source ~/oprint/bin/activate
cd ~/OctoPrint
pip install .
sudo service octoprint restart
```

MarcusWolschon, 2018-12-09:
$ cat /etc/octopi_version
0.13.0

Reinstall worked.
I get the fresh-install wizard.
Copied my old config in. It came up in "Sicherheitsmodus"
Restarted without "Sicherheitsmodus".
Had to reinstall the code for the plugins but at least their config was retained.
Thanks a lot!

closed 2018-12-10 by foosel