OctoPrint/OctoPrint #2902: Cura 3.6.0 makes a doubled gcode  whenprinting to 1.3.10rc2

dimkin-eu, 2018-11-18:
#### What were you doing?

1. open STL in cura
2. [adjust positions/speed/fill/etc]
3. press "Print with OctoPrint"

#### What did you expect to happen?
correct file sent to OctoPrint

#### What happened instead?
filesize is double comparing to just saved from Cura and uploaded manually
after model  printed, printer started to print it again, dragging into heatbad :(

not reproduceable in safe mode, also, after returning to non-safe mode everything is fine
havent saved doubled file, shame on me
feel free to close ticket

#### Did the same happen when running OctoPrint in safe mode?

not reproduceable, also, after returning to non-safe mode everything is fine
havent saved doubled file, shame on me
feel free to close ticket

#### Version of OctoPrint

OctoPrint 1.3.10rc2 running on OctoPi 0.15.1

#### Printer model & used firmware incl. version
 CR10 + TH3D U1.R2.4b (10/19/2018) 

#### Browser and version of browser, operating system running browser

Mozilla/5.0 (Macintosh; Intel Mac OS X 10_14_1) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/70.0.3538.102 Safari/537.36

#### Link to octoprint.log

https://pastebin.com/nz1pVQ5Y

#### Link to contents of terminal tab or serial.log

2018-11-18 09:25:16,631 - serial.log is currently not enabled, you can enable it via Settings > Serial Connection > Log communication to serial.log
2018-11-18 10:34:59,425 - serial.log is currently not enabled, you can enable it via Settings > Serial Connection > Log communication to serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

[image]

I have read the FAQ.

foosel, 2018-11-19:
> filesize is double comparing to just saved from Cura and uploaded manually

That sounds more like a bug in Cura than anything else really, if it works when uploaded through OctoPrint's own UI.

closed 2018-11-19 by foosel

jarrodasmith, 2018-11-21:
It's a bug in the Octoprint 3.5.2 Cura plugin.  Identified and will be fixed in 3.5.3.

https://github.com/fieldOfView/OctoPrintPlugin/issues/94