OctoPrint/OctoPrint #2903: [1.3.10rc2] Backup plugin fails when timelapse directory is external USB

autonumous, 2018-11-20:
#### What were you doing?

Using the new Backup and restore feature of 1.3.10rc2

On a clean install, this feature works as expected, a zip file is created, 
log  ~/.octoprint/logs/octoprint.log shows
`octoprint.log.2018-11-15:2018-11-15 01:13:31,935 - octoprint.plugins.backup - INFO - Creating backup zip at /home/pi/.octoprint/data/backup/.backup-20181115-011331.zip (excluded: -)...`

However when having timelapse on an external USB, nothing happens when clicking backup button.  
the above does not appear in the log.

exact steps to follow in order to reproduce your problem:
1) clean install of octopi 0.15.1 on 16GB  card
2) initial login
 log in to GUI, go through initial options, 
3) update to 1.3.10rc2
4) log in to gui, & test backup 
 Settings -> backup and restore -> ‘create backup  now’
 new backup zip file created. (can verify in log) 
5) login via cli (user pi)
6) stop octoprint
sudo service octoprint stop
7) mount Existing USB
sudo mkdir /media/timelapse
sudo mount /dev/sda1 /media/timelapse
8) remove existing timelapse dir
cd ~/.octoprint && rm -rf timelapse
9) link to usb mount
ln -s /media/timelapse ~/.octoprint/timelapse
10) start octoprint
sudo service octoprint start

11) login to GUI, create backup as above 
The file is not created, nothing logged regarding backup in  ~/.octoprint/logs/octoprint.log
Tried both exclude options (timelapse and uploads)

12) tried mounting USB directly onto ~/.octoprint/timelapse same result.

#### What did you expect to happen?
backup zip file to be created

#### What happened instead?
when using USB, irrespective of exclude options, backup not performed.
 
#### Did the same happen when running OctoPrint in safe mode?
Yes

#### Version of OctoPrint

1.3.10rc2 (& rc1)

#### Operating System running OctoPrint

octopi 0.15.1

#### Printer model & used firmware incl. version

Marlin 1.1.9

#### Browser and version of browser, operating system running browser

Safari macos

#### Link to octoprint.log

will make one available asap

#### Link to contents of terminal tab or serial.log

can make available is if required

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2018-11-20:
What are the permissions on the mounted USB volume? Based on the commands you showed above, it should only be readable by the pi user, not writable, correct?

Trying to reproduce this, at least just with a regular symlink to another folder on the disk it doesn't reproduce, so it must be something related to the USB stick mount.

autonumous, 2018-11-20:
thanks @foosel for  your response.  long reply below sorry...

> What are the permissions on the mounted USB volume? Based on the commands you showed above, it should only be readable by the pi user, not writable, correct?
> 

On mine its writeable

```
pi@octopi:~ $ mount |grep sda 
pi@octopi:~ $
pi@octopi:~ $ sudo mount /dev/sda1 /media/timelapse
pi@octopi:~ $ mount |grep sda
/dev/sda1 on /media/timelapse type ext4 (rw,relatime,data=ordered)
pi@octopi:~ $

```

although i should point out, these steps i posted were only to simulate the issue, i normally mount my usb via an /etc/fstab entry, i.e. 
`UUID=4ee699d8-94a6-435a-bfeb-b392050e7e60 /media/timelapse ext4    defaults,noatime  0  0`

As for logs, difficult to show something that doesn't happen.

```
pi@octopi:~/.octoprint $ sudo service octoprint start
pi@octopi:~/.octoprint/logs $ tail -f ~/.octoprint/logs/octoprint.log
2018-11-20 20:11:46,591 - octoprint.events - INFO - Processing startup event, this is our first event
2018-11-20 20:11:46,592 - octoprint.events - INFO - Adding 0 events to queue that were held back before startup event
2018-11-20 20:11:46,595 - octoprint.filemanager - INFO - Adding backlog items from all storage types to analysis queue...
2018-11-20 20:11:46,600 - octoprint.filemanager - INFO - Added 0 items from storage type "local" to analysis queue
2018-11-20 20:11:46,617 - octoprint.plugins.discovery - INFO - Registered 'OctoPrint instance on octopi' for _http._tcp
2018-11-20 20:11:46,630 - octoprint.plugins.discovery - INFO - Registered 'OctoPrint instance on octopi' for _octoprint._tcp
2018-11-20 20:11:46,642 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on octopi for SSDP
2018-11-20 20:11:46,656 - octoprint.server - INFO - Listening on http://127.0.0.1:5000
2018-11-20 20:11:46,704 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from disk, was still valid
2018-11-20 20:11:47,142 - octoprint.plugins.announcements - INFO - Loaded channel _important from https://octoprint.org/feeds/important.xml in 0.47s
2018-11-20 20:11:47,467 - octoprint.plugins.pluginmanager - INFO - Loaded plugin notices data from https://plugins.octoprint.org/notices.json
2018-11-20 20:11:47,653 - octoprint.plugins.announcements - INFO - Loaded channel _releases from https://octoprint.org/feeds/releases.xml in 0.18s
2018-11-20 20:11:47,933 - octoprint.plugins.announcements - INFO - Loaded channel _blog from https://octoprint.org/feeds/octoblog.xml in 0.19s
2018-11-20 20:11:48,180 - octoprint.plugins.announcements - INFO - Loaded channel _plugins from https://plugins.octoprint.org/feed.xml in 0.16s
2018-11-20 20:11:48,485 - octoprint.plugins.announcements - INFO - Loaded channel _octopi from https://octoprint.org/feeds/octopi.xml in 0.17s
2018-11-20 20:11:49,950 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python2 -m pip" as command to invoke pip
2018-11-20 20:11:58,466 - octoprint.util.pip - INFO - pip installs to /home/pi/oprint/lib/python2.7/site-packages/ (writable -> yes), --user flag needed -> no, virtual env -> yes
2018-11-20 20:11:58,467 - octoprint.util.pip - INFO - ==> pip ok -> yes
2018-11-20 20:11:58,488 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2018-11-20 20:12:00,513 - octoprint.util.pip - INFO - Using "/home/pi/oprint/bin/python2 -m pip" as command to invoke pip
2018-11-20 20:12:00,515 - octoprint.util.pip - INFO - pip installs to /home/pi/oprint/lib/python2.7/site-packages (writable -> yes), --user flag needed -> no, virtual env -> yes
2018-11-20 20:12:00,516 - octoprint.util.pip - INFO - ==> pip ok -> yes
2018-11-20 20:12:12,049 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.100.150
2018-11-20 20:12:14,881 - octoprint.server.util.flask - INFO - Passively logging in user dummy from ::ffff:192.168.100.150
2018-11-20 20:12:25,793 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.100.150
2018-11-20 20:12:26,847 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.100.150
2018-11-20 20:12:26,953 - octoprint.server.util.flask - INFO - Passively logging in user dummy from ::ffff:192.168.100.150
```
at this point I’m hitting the "Create backup now” button, but no logs.

so edit the backup, to log something, so i know the button is being pressed,

vi ~/oprint/lib/python2.7/site-packages/octoprint/plugins/backup/__init__.py

add the following bold line
```
 def create_backup(self):
                **self._logger.info(u"DEBUG - BackUp Requested.")**
                backup_file = "backup-{}.zip".format(time.strftime("%Y%m%d-%H%M%S"))
```
restart & tail log while madly hitting "Create backup now” button, 

```
pi@octopi:~/.octoprint/logs $ sudo service octoprint restart
pi@octopi:~/.octoprint/logs $ tail -f ~/.octoprint/logs/octoprint.log
[... 264 lines cut ...]
2018-11-20 20:35:46,840 - octoprint.server - INFO - Shutting down intermediary server...
2018-11-20 20:35:47,311 - octoprint.server - INFO - Intermediary server shut down
2018-11-20 20:35:47,314 - octoprint.events - INFO - Processing startup event, this is our first event
2018-11-20 20:35:47,315 - octoprint.events - INFO - Adding 0 events to queue that were held back before startup event
2018-11-20 20:35:47,317 - octoprint.filemanager - INFO - Adding backlog items from all storage types to analysis queue...
[... cut ...]
2018-11-20 20:36:39,959 - octoprint.server.util.flask - INFO - Passively logging in user dummy from ::ffff:192.168.100.150
2018-11-20 20:36:54,784 - octoprint.plugins.backup - INFO - DEBUG - BackUp Requested.
2018-11-20 20:36:54,806 - octoprint.plugins.backup - INFO - Creating backup zip at /home/pi/.octoprint/data/backup/.backup-20181120-203654.zip (excluded: -)...
2018-11-20 20:36:54,909 - octoprint.plugins.backup - INFO - ... done creating backup zip.
^C
pi@octopi:~/.octoprint/logs $
```

3 entries for backup

autonumous, 2018-11-21:
Actually my last comment not entirely accurate. Still seeing issue on my other image.   More testing later...

foosel, 2018-11-21:
I find it really odd that you have write access out of the box. Have you added the `pi` user to any specific additional groups by any chance? The `id` command should give some insights here.

In any case, I'd love to try something. Are you comfortable with installing a development branch (`staging/maintenance`)? I've just added some error logging to the backup creation thread (something which I really should have done from the get go). So whatever goes wrong there for you should now at least produce a log entry if you run that and might give us some more insights into what's happening.

Alternatively you might also see some stack trace if you stop the service and run OctoPrint from console:

```
sudo service octoprint stop
~/oprint/bin/octoprint serve --host 127.0.0.1
```

autonumous, 2018-11-21:
@foosel,  no changes it was a brand new octopi instance written to a brand new as card. 
 
Sure i’ll try that commit later.

autonumous, 2018-11-22:
Due to other commitments, was unable to test last night, hope to be able to spend time to tonight....

autonumous, 2018-11-24:
I think there was  too much to backup.  

```
2018-11-24 00:15:10,638 - octoprint.server.heartbeat - INFO - Server heartbeat <3
Exception in thread Thread-32:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/backup/__init__.py", line 643, in _create_backup
    zip.writestr("plugin_list.json", json.dumps(plugins))
  File "/usr/lib/python2.7/zipfile.py", line 803, in __exit__
    self.close()
  File "/usr/lib/python2.7/zipfile.py", line 1349, in close
    " would require ZIP64 extensions")
LargeZipFile: Central directory offset would require ZIP64 extensions

2018-11-24 00:30:10,640 - octoprint.server.heartbeat - INFO - Server heartbeat <3
```

foosel, 2018-11-26:
> OSError: [Errno 13] Permission denied: '/home/pi/.octoprint/timelapse/lost+found'

Ok, I think if issues like these are getting reported on the frontend instead of being silently eaten up it should suffice, or what do you think?

> Central directory offset would require ZIP64 extensions

I've actually run into this myself in a test related to this issue and fixed it in a recent commit. With the fix in place I successfully created a 3.3GB backup. See bd241e8e6c22354d8ef777fa88e3aca6cc8d45e1 

Therefore tentatively marking this as solved.

autonumous, 2018-11-28:
I’ve just tried with that commit. It’s successfully backed up 3.5GB to the USB

I’ll close this issue

Thanks

closed 2018-11-28 by autonumous