OctoPrint/OctoPrint #2904: Suggestion: Add ipv4 and ipv6 switches to oserv

tkurbad, 2018-11-21:
#### What were you doing?

1. ...
2. ...
3. ...

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2018-11-22:
> seems to prefer IPv6 over IPv4

Just for the record, it prefers a dual stack and if that doesn't work out it goes for v4.

tkurbad, 2018-11-22:
> > seems to prefer IPv6 over IPv4
> 
> Just for the record, it prefers a dual stack and if that doesn't work out it goes for v4.

That's odd, because when I just started it with the `--port 5000` option, it stated to be listening on `http://[::]:5000`
No sign of IPv4 there. But maybe, that's just a logging/display problem. Anyway, since I added `--host 0.0.0.0` to the command line, everything runs smoothly again, which solved my problem, but - even more in the light of your comment, @foosel - doesn't explain the connection problems I've had before.

foosel, 2018-11-22:
So, here's what happens internally. It checks if there is ipv6 support[*] *and* dual stack support according to some OS flags. If both is the case it will configure the intermediary server socket to bind to both and also tell tornado to use a v6 address which will make it bind to both (and at least in my test cases that also worked out fine, as visible in `netstat`). It will display the v6 address in the log though. 

If there is no dual stack support it will fall back to v4.

[*]: That check in 1.3.9 still has a bug since the socket property used for that apparently doesn't actually report on OS support but rather library support - the documentation sadly was unclear there. 1.3.10 actually does attempt to create a v6 socket once on initial v6 check and if that fails assumes v6 support is lacking.

tkurbad, 2018-11-22:
Thank you for explaining this in detail. If that in general works as expected, I think we can close this issue. Also, the `--host 0.0.0.0` works just fine for me.

On a side note, I'm beginning to suspect that I've got a serious IPv6 routing problem at hand:
My 3d printer is located at my employer's, a research institute that shares the internet connection of the local university (Tübingen, Germany, to be exact). Since the university still didn't manage the transition to IPv6, I disabled the protocol on all of our switches and routers. Nonetheless, all modern OSes run dual stack per default, so the problem might be that just disabling the protocol on a managed switch doesn't prevent IPv6 machines from **seeing** each other, while it inhibits **communication** between them nonetheless.
I'll need to investigate this further...

foosel, 2018-11-22:
Ah. Yeah, that might be an issue here. I also noticed some really odd behaviour at least with dual stacked windows clients that will happily tackle back and forth between v4 and v6 for connections against the same host from a browser session.

tkurbad, 2018-11-22:
Ok, I've been able to verify that disabling IPv6 on a HP Procurve switch leads to problems with a dual stack setup of octoprint.
Would you consider adding the protocol parameters nonetheless?

foosel, 2018-11-22:
Yes. They are probably easier for some people to remember than `--host 0.0.0.0` (and shorter too)

OutsourcedGuru, 2019-03-01:
Thought I'd add:

I have a Netgear Internet router as well as two Orbi (Netgear) internal wi-fi routers. My Internet provider is Cox. It seems like over the last year everything internally has transitioned from IPv4-as-default to IPv6-as-default (at least with respect to the DNS lookup as provided by the collection of three Netgear routers).

That said, I now notice over time that IPv6 really isn't as reliable as the IPv4. For example, my IPv6-based `ssh` session to the Raspberry will freeze up during a `tail` command to the octoprint.log. The web interface will be fine and the job printing. I will open another Terminal and try to connect via `ssh`. No response. A `ping` will not work. Fetching the log from the web interface will show me that almost everything is using IPv6-based addresses.

While in this weird state, I note that if I then connect an Ethernet cable to the Raspberry Pi I can then `ssh` into it and everything's fine. I run an `htop` and this is not related to a too-busy system.

In summary, I think the Netgear router's implementation of DNS on the IPv6 side is buggy. So I agree with the author of this request, I would like it if there were easy ways of just toggling that off. Thanks.

jubaleth, 2019-03-01:
> While in this weird state, I note that if I then connect an Ethernet cable to the Raspberry Pi I can then `ssh` into it and everything's fine. I run an `htop` and this is not related to a too-busy system.

This sounds a lot like an issue with the wireless link, be it between your pi and AP, your computer and AP, or both. Unfortunately, there's really not enough information here to make a realistic diagnosis...

> In summary, I think the Netgear router's implementation of DNS on the IPv6 side is buggy.

Are you assigning a name to your rpi (either ipv6 or v4, or both) manually in your router's DNS server? If you're relying on the .local name, then the DNS server on your router has nothing to do with it. The .local name is broadcast on the multicast address (mDNS), your OS will keep track of what announcements it sees, there is no additional query to the DNS server (at least not a successful one, unless you've added a record by hand). Either way, from what you describe, DNS definitely doesn't sound like the problem here. Especially since you said you can reach the rpi over ipv6, but then have connectivity issues later.

OutsourcedGuru, 2019-03-01:
@jubaleth No, I'm not assigning a name in the DNS router. I think I agree with your `.local` logic, though; why would the Netgear care about such things?

I'm on macOS Mojave 10.14.2 so you'd think this would just work. Something in this path (Bonjour broadcasts to macOS DNS or arp cache) within IPv6 appears to be falling on its face.

I fear that if this is an Apple bug, it's likely not to be fixed soon or even discovered for what it is. Or is this somehow Debian/Bonjour/IPv6?

I don't ultimately have to know what's causing it and why. I'm just throwing my two cents into the discussion that sometimes it's just best to toggle this off.

foosel, 2019-03-01:
If you are running OctoPrint on OctoPi images, this switch discussed here will not do anything anyhow since you aren't accessing OctoPrint directly but rather through haproxy and it depends on how *that* is bound. And on recent OctoPi images it's bound to both v4 and v6 (and OctoPrint itself is listening on 127.0.0.1 only and hence v4). 

Any switches added to the serve or daemon command will really only have an effect on how you can access OctoPrint *directly* instead of through a reverse proxy.

OutsourcedGuru, 2019-03-01:
Thanks for the clarification. You're talking about the OctoPrint service itself.

foosel, 2019-03-01:
In this repository we are always talking about the OctoPrint service itself ;)

foosel, 2019-03-11:
The above commit adds `-4` (basically an alias for `--host 0.0.0.0`) and `-6` (host `::` and disabled dual stack binding).

Pushed to `maintenance`, soon `devel`, to be released with 1.3.11.

tkurbad, 2019-03-12:
Thank you very much, @foosel!

foosel, 2019-05-15:
1.3.11 was released yesterday.

closed 2019-05-15 by foosel