OctoPrint/OctoPrint #2905: [1.3.10rc2] Communication Timeout after some random time for no apparent reason

FanDjango, 2018-11-23:
After noticing communications timeouts happening everytime some random time after printing an object had ended, I decided to look closer at what is happening here. In order to find out more, I used SAFE MODE and did the following:

#### What were you doing?

Fresh new start of System, of Octoprint Server in SAFE MODE. 
Connect to printer
Wait a long time but do nothing, no printing, no commands, just wait.

#### What did you expect to happen?

Connection to printer should remain "connected".

#### What happened instead?

After some random time, a connection timeout happened. Reproducible but random time.

#### Did the same happen when running OctoPrint in safe mode?

Yes

#### Version of OctoPrint

[1.3.10rc2]

I never noticed this in prior versions. Opportunities for this same error to occur were present in prior versions, as I often am not there when a print ends and there were never communication timeouts before.

#### Operating System running OctoPrint

Octoprint [1.3.10rc2] on Tinkerboard S with Debian Strecth current version and all updates till today.

#### Printer model & used firmware incl. version

Anet A8 mod with Geetech GT2560 + BLTouch + Filament Runout Sensor + Marlin [bugfix-2.0.x] 2018-11-21

#### Browser and version of browser, operating system running browser

NA

#### Link to octoprint.log

https://pastebin.com/AGdHtjzK

#### Link to contents of terminal tab or serial.log

https://pastebin.com/kpwks8KZ

#### Link to contents of Javascript console in the browser

NA

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2018-11-23:
Hm... looks like my initial hunch is wrong, so a bit more digging is needed.

From your `serial.log` I see that your regular temperature reports come ever 5s. Then just before the timeout is reported there is a gap of 9s. That would match a communication timeout of 3 x 3s, which is configured according to `octoprint.log`. 

```
2018-11-23 10:29:58,428 - Recv:  T:100.00 /100.00 B:40.00 /40.00 @:9 B@:14
2018-11-23 10:30:03,429 - Recv:  T:100.00 /100.00 B:40.00 /40.00 @:9 B@:14
2018-11-23 10:30:08,428 - Recv:  T:100.00 /100.00 B:40.00 /40.00 @:9 B@:15
2018-11-23 10:30:13,429 - Recv:  T:100.00 /100.00 B:40.00 /40.00 @:9 B@:15
2018-11-23 10:30:22,442 - No response from printer after 3 consecutive communication timeouts, considering it dead. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
```

Do you have any idea why your printer stops sending temperature reports there in their regular intervals? If you increase the communication timeouts, does the reporting take up again?

In any case, it looks like you changed the default intervals and timeouts - temperature auto report interval with 5s is now larger than the busy timeout of 3s, so that is probably also a factor here. Does the same behaviour manifest with stock values (temperature auto report = 2s)?

FanDjango, 2018-11-23:
`In any case, it looks like you changed the default intervals and timeouts - temperature auto report interval with 5s is now larger than the busy timeout of 3s, so that is probably also a factor here.`

Darn, I knew this would come up and forgot to mention this in the initial issue submit: I started playing around with this (2s -> 5s) to see what is going on. By the way: The frequency of the busy message in Marlin Configuration_adv.h is still 2s.

```
//
// Host Keepalive
//
// When enabled Marlin will send a busy status message to the host
// every couple of seconds when it can't accept commands.
//
#define HOST_KEEPALIVE_FEATURE        // Disable this if your host doesn't like keepalive messages
#define DEFAULT_KEEPALIVE_INTERVAL 2  // Number of seconds between "busy" messages. Set with M113.
#define BUSY_WHILE_HEATING            // Some hosts require "busy" messages even during heating
```

`Does the same behaviour manifest with stock values (temperature auto report = 2s)?`

Yes, I will go back to the 2 seconds and I will submit a new serial log showing the same thing.

FanDjango, 2018-11-23:
But I agree about the 9 seconds. Hmmm. That could mean Marlin has a hickup, or the Debian underlying OctoPrint? Sheesh, this will become difficult, I will see what I can do... I will check in sylog if there was anything heavy going on at that time but it's a quick little board with no other duties.

FanDjango, 2018-11-23:
I'll close for now and reopen if I feel it points to OctoPrint. The new scenario with 2s and all temperatures = 0 is running and it will take a while... but with your explanation of the timing issue (above) I feel the cause is to be found somewhere else.

FanDjango, 2018-11-23:
LOL It didn't take long:
```
2018-11-23 14:37:44,149 - Recv:  T:72.91 /0.00 B:37.72 /0.00 @:0 B@:0
2018-11-23 14:37:46,149 - Recv:  T:72.68 /0.00 B:37.76 /0.00 @:0 B@:0
2018-11-23 14:37:48,149 - Recv:  T:72.39 /0.00 B:37.65 /0.00 @:0 B@:0
2018-11-23 14:37:50,149 - Recv:  T:72.11 /0.00 B:37.59 /0.00 @:0 B@:0
2018-11-23 14:37:52,149 - Recv:  T:71.91 /0.00 B:37.59 /0.00 @:0 B@:0
2018-11-23 14:37:54,149 - Recv:  T:71.67 /0.00 B:37.59 /0.00 @:0 B@:0
2018-11-23 14:37:56,148 - Recv:  T:71.41 /0.00 B:37.59 /0.00 @:0 B@:0
2018-11-23 14:37:58,148 - Recv:  T:71.16 /0.00 B:37.59 /0.00 @:0 B@:0
2018-11-23 14:38:00,149 - Recv:  T:70.97 /0.00 B:37.54 /0.00 @:0 B@:0
2018-11-23 14:38:09,163 - No response from printer after 3 consecutive communication timeouts, considering it dead. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
2018-11-23 14:38:10,220 - Changing monitoring state from "Operational" to "Offline (Error: Too many consecutive timeouts, printer still connected and alive?)"
2018-11-23 14:38:10,225 - Connection closed, closing down monitor
```
I'll have to check what Marlin is doing. 9 seconds again? Is this a coincidence?

closed 2018-11-23 by FanDjango

foosel, 2018-11-23:
Depends on what your comm timeout was. But if it detected busy capability it probably sat at 3 again, and 3 x 3 = 9. If you set the comm timeout when busy to something like 5s you might see 15s. Or if you don't execute a command before that allows detection of busy support and hence sticks at the default communication timeout of 30s you'll probably see the same I saw with the virtual printer, 90s. Or not because maybe the firmware will start responding again before that time.

I'd up those comm timeouts and see what happens, if the firmware gets stucked for good or recovers. Might give some hints as to what's happening.

FanDjango, 2018-11-23:
Sorry to bother you again, I did as you suggested and set 20 seconds. Look:
```
018-11-23 14:59:26,317 - Recv: ok
2018-11-23 14:59:26,318 - Send: M104 S100
2018-11-23 14:59:26,322 - Recv: ok
2018-11-23 14:59:26,323 - Send: M140 S40
2018-11-23 14:59:26,326 - Recv: ok
2018-11-23 14:59:26,327 - Send: M280 P0 S120
2018-11-23 14:59:26,395 - Recv: ok
2018-11-23 14:59:26,396 - Send: G4 P2500
2018-11-23 14:59:28,402 - Recv: echo:busy: processing
2018-11-23 14:59:28,407 - Printer seems to support the busy protocol, will adjust timeouts and set busy interval accordingly
2018-11-23 14:59:28,901 - Recv: ok
2018-11-23 14:59:28,902 - Send: M280 P0 S90
2018-11-23 14:59:28,956 - Recv: ok
2018-11-23 14:59:28,957 - Send: M300 S440 P250
2018-11-23 14:59:28,961 - Recv: ok
2018-11-23 14:59:28,962 - Send: M300 S660 P250
2018-11-23 14:59:28,965 - Recv: ok
2018-11-23 14:59:28,966 - Send: M155 S20
2018-11-23 14:59:28,969 - Recv: ok
2018-11-23 14:59:28,970 - Send: M113 S2
2018-11-23 14:59:28,973 - Recv: ok
2018-11-23 14:59:37,985 - No response from printer after 3 consecutive communication timeouts, considering it dead. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
2018-11-23 14:59:37,989 - Changing monitoring state from "Operational" to "Offline (Error: Too many consecutive timeouts, printer still connected and alive?)"
2018-11-23 14:59:37,992 - Connection closed, closing down monitor
```
Timeout after 10 seconds of silence. What is going on? Shouldn't there be 60 sconds of grace? 20 * 3?

FanDjango, 2018-11-23:
M155 S20 and M113 S2 (i.e. 20 seconds for TEMP and 2 seconds for BUSY) will not work. Why?
I need to set M155 S20 AND M113 S20.
Now I can connect. 
```
018-11-23 15:07:18,429 - Send: M155 S20
2018-11-23 15:07:18,433 - Recv: ok
2018-11-23 15:07:18,434 - Send: M113 S19
2018-11-23 15:07:38,437 - Recv: o� T:97.18 /100.00 B:32.46 /40.00 @:0 B@:127
2018-11-23 15:07:58,436 - Recv:  T:110.77 /100.00 B:35.95 /40.00 @:0 B@:127
2018-11-23 15:08:18,435 - Recv:  T:105.76 /100.00 B:39.01 /40.00 @:0 B@:127
2018-11-23 15:08:38,435 - Recv:  T:99.68 /100.00 B:40.06 /40.00 @:12 B@:55
2018-11-23 15:08:58,434 - Recv:  T:100.64 /100.00 B:40.02 /40.00 @:0 B@:44
2018-11-23 15:09:18,433 - Recv:  T:99.83 /100.00 B:40.02 /40.00 @:13 B@:33
2018-11-23 15:09:38,432 - Recv:  T:99.67 /100.00 B:40.00 /40.00 @:15 B@:30
2018-11-23 15:09:58,432 - Recv:  T:100.11 /100.00 B:40.02 /40.00 @:8 B@:21
```
Look again: The "Ok" is mangled...
I am getting a little confused now. Time for weekend.

foosel, 2018-11-23:
> M155 S20 and M113 S2 (i.e. 20 seconds for TEMP and 2 seconds for BUSY) will not work. Why?

Because that basically makes the printer only respond every 20s but with a timeout of 3s. The other way round will work. That way around won't. Temperature interval should always be lower than busy timeout. Otherwise the printer won't generate replies within the timeout length. 

That mangled `ok` there might have several reasons, not necessarily related to this problem. Unless you can reliably reproduce it I'd ignore it for now.

FanDjango, 2018-11-23:
Thank you very much - I have been looking at the code and have reached that conclusion also. Plus the fact that setting M113 to 3 in the GUI will result in M113 S2, I can see the minus "1" in the code comm.py

Ok, what I have found out is that 
1. I think MARLIN serial output from TEMP AUTOREPORTS which comes from the idle loop might sometimes be mangled by some other process from an INTR handler doing some serial output as well. I have seen a few occurences of temp reports with beginning parts missing. So ignoring this for now and perhaps reporting that to them someday
2. Turning off temperature autoreporting looks like things could go a little more smoothly. I am testing with this setting the next few days.
3. Still need to identify the situations where the time between TEMPERATURE AUTOREPORTS is longer than the timeout, but I cannot do that until a few days have passed where I am testing TEMPERATURE AUTOREPORTS = DISABLED :-)
Thanks for all your time. When I know more I will update you.

P.S. It might alleviate confusion if you sanity check the timeout values (i.e. the one should not be smaller then the other)

foosel, 2018-11-23:
> P.S. It might alleviate confusion if you sanity check the timeout values (i.e. the one should not be smaller then the other)

That's what I meant with 

> I'll take a look though at the used communication timeout, I guess it should use the maximum of temperature (auto) reporting and configured communication timeout, just in case.

:D

FanDjango, 2018-11-24:
For the sake of documentation, I have no more problems since using the following values:

Marlin: Temperature Autoreport is ON, Keepalive (BUSY) messages are ON.

OctoPrint:
Autoreport Interval set to 15
Communication Timeout set to 30 (BUSY protocol)

I have done no further investigation of what kind of "once in a while" delays might be involved in the Marlin temperature reporting when using a shorter timeout.

foosel, 2018-11-26:
I added a `max` check per the above commit, will be part of 1.3.**11** (1.3.10 is already in freeze mode and this is too risky of an adjustment to put into an RC iteration).