OctoPrint/OctoPrint #2906: Hang->Communication Timeout when using Marlin config item: WAIT_FOR_BED_HEATER on probing

FanDjango, 2018-11-23:
Marlin has a configuration option to disable steppers and/or disable heaters on probing to increase the accuracy of BLTouch by removing noise and spikes on the 5V supply, and to (optionally) reheat back to nominal temperature (in case this might have decreased) before continuing. 

This is what that looks like (in Marlin Configuration.h):

```
#define PROBING_HEATERS_OFF       // Turn heaters off when probing §§§§
#if ENABLED(PROBING_HEATERS_OFF)
  //#define WAIT_FOR_BED_HEATER     // Wait for bed to heat back up between probes (to improve accuracy) §§§§
#endif
#define PROBING_FANS_OFF          // Turn fans off when probing §§§§
//#define PROBING_STEPPERS_OFF      // Turn steppers off (unless needed to hold position) when probing
#define DELAY_BEFORE_PROBING 200  // (ms) To prevent vibrations from triggering piezo sensors §§§§
```

So, with OctoPrint, you can turn this stuff on, but you must do without the **#define WAIT_FOR_BED_HEATER** because it causes a little problem.

If you turn it on, OctoPrint will monitor the **echo:busy: processing** message - but while the bed is still heating back up prior to a probe, the message is changed to **Wait for bed heating...** which, it seems, is not accepted by OctoPrint in this context.

Once the bed has heated up, the **echo:busy: processing** messages will resume until the probe has completed.

This especially concerns G28 (for Z, once) and G29 commands. G29 goes into such a phase repeatedly on each of the levelling probe movements, of course only if the bed cools down enough in the short probing phase to make a reheat necessary.

Also note that this parameter of course will also wait for the bed to heat, regardless of the reason why the bed might be too cold - in the case of the below serial log, it is because the bed is not heated up yet from the start of the print in the first place.

#### What were you doing?

Trying to use "WAIT_FOR_BED_HEATER" in Marlin Configuration.h

#### What did you expect to happen?

Nothing bad

#### What happened instead?

Configuration Timeout

#### Did the same happen when running OctoPrint in safe mode?

Yes

#### Version of OctoPrint

[1.3.10rc2]

#### Operating System running OctoPrint

Octoprint [1.3.10rc2] on Tinkerboard S with Debian Stretch current version and all updates till today.

#### Printer model & used firmware incl. version

Anet A8 mod with Geetech GT2560 + BLTouch + Filament Runout Sensor + Marlin [bugfix-2.0.x] 2018-11-21

#### Browser and version of browser, operating system running browser

NA

#### Link to octoprint.log

https://pastebin.com/zNTkLYxE

#### Link to contents of terminal tab or serial.log

Note: After printer connect, manually entered G28 XY (no problem), then G28 Z (problem).

https://pastebin.com/Z84tiaRf

Note: To show the further sequence, I would need to attach the printer via USB to a serial terminal on a laptop and capture the sequence there. Perhaps the above description of events would allow a testing version to be created which also accepts the alternate message and I will gladly test that.

#### Link to contents of Javascript console in the browser

NA

#### Screenshot(s)/video(s) showing the problem:

NA

I have read the FAQ.

foosel, 2018-11-23:
Thank you for the thorough report!

That honestly looks like a firmware bug. Marlin is reporting here that it does comply with the busy protocol, so it should *always* be sending `busy` messages when it is ... well ... busy, regardless of what it is busy with. So while this "Waiting for heater" bit is idling there, Marlin should also be sending `busy` messages.

It doesn't though, hence OctoPrint has to assume something terrible has happened. The only reason for `busy` to stop being sent should really be that the serial line can be used again, hence an `ok` should usually follow the end of `busy` messages right away. Here we get one line of non defined human readable text, then nothing for a long time (and then later some more `busy` but we never see that due to the long delay). The result is that the printer looks pretty dead and hence a communication timeout is triggered. So - OctoPrint is actually behaving correctly here. You could work around this by disabling `busy` protocol support, but the fix for this problem has to be done on the firmware side of things.

I'd suggest to open a bug report for this on the Marlin repository so that this error in complying with the busy protocol can be looked into.

FanDjango, 2018-11-23:
I was sort of expecting that. But with your expert comment, my chances to convince them might have improved substantially. Thanks and I will converse with the Marlin developers about this.

foosel, 2018-11-23:
I'll keep my fingers crossed for you. Feel free to link to this ticket, if push comes to shove I can also weigh in. Closing this though as - as I said - OctoPrint is behaving correctly here.

closed 2018-11-23 by foosel

FanDjango, 2018-11-24:
For the sake of documentation: This Marlin issue will be fixed in Marlin Bugfix-2.0 2018 11 24 or a few days later - you can search for this issue, they accepted it right away. I am testing it right now and it works ok.

foosel, 2018-11-26:
@FanDjango Thanks for the follow-up!