OctoPrint/OctoPrint #2908: [1.3.10rc2] Multiple tracked files can have the same hash

DominikPalo, 2018-11-24:
I've checked the implementation of the tracking plugin and have found, that the sha1 hash of a printed file is computed only from the [file name](https://github.com/foosel/OctoPrint/blob/de42ce73f7c4a73b892e405dc34dcf1e38f36f22/src/octoprint/plugins/tracking/__init__.py#L236) (and the instance id):

```python
sha = hashlib.sha1()
sha.update(payload.get("name"))
sha.update(self._settings.get([b"unique_id"]))
```

Wouldn't be better to include also the file path and location into that hash?
```python
sha = hashlib.sha1()
sha.update(payload.get("origin"))
sha.update(payload.get("path"))
sha.update(payload.get("name"))
sha.update(self._settings.get([b"unique_id"]))
```
Because now I can imagine a situation when two completely different files with the same name that are stored in different locations, e.g.:
```
/foo-v1/bar.gcode
/foo-v2/bar.gcode
```
result in the same hash.

foosel, 2018-11-26:
That's actually a really good point. Thanks for that. 

The filename hashing is only used to be able to potentially track something like "file X was started and cancelled 4h later" for the same instance, but even for that it should definitely be able to differentiate between files in sub folders.

foosel, 2018-11-26:
Fixed by the above commit, to be part of 1.3.10rc3. In the end I went just for `path` instead of `name` - `path` contains `name` and `origin` is already provided as part of the payload anyhow.

foosel, 2018-11-28:
1.3.10rc3 was just released.

commit 9a90093d (2018-11-26): tracking: file path instead of just name in print events

Closes #2908
commit 13260fa2 (2018-11-26): tracking: file path instead of just name in print events

Closes #2908
closed 2018-11-28 by foosel