OctoPrint/OctoPrint #2909: [Request] Do not link to images hosted at imgur.com directly from front page of server

kortschak, 2018-11-25:
#### What were you doing?
Loading local OctoPrint page.

1. Navigate to https://octopi.local/
2. Watch traffic to external sites.
3. ...

#### What did you expect to happen?
Only traffic from octopi.local.

#### What happened instead?
Image and cookie traffic from imgur.com.

#### Did the same happen when running OctoPrint in safe mode?
Yes

#### Version of OctoPrint
OctoPrint 1.3.9

#### Operating System running OctoPrint
 OctoPi 0.15.0

#### Printer model & used firmware incl. version
Prusa i3 Mk3

#### Browser and version of browser, operating system running browser
Not relevant.

#### Link to octoprint.log
Not relevant.

#### Link to contents of terminal tab or serial.log
Not relevant.

#### Link to contents of Javascript console in the browser
Not relevant.

#### Screenshot(s)/video(s) showing the problem:
Not relevant.

I have read the FAQ.

#### Additional information.
This is due to CHANGELOG.md having `![]()`s pointing to imgur.com (also README.md and CONTRIBUTING.md, but they aren't the cause AFAICS). A solution would be to host the images within the repo rather than at imgur.

foosel, 2018-11-26:
> A solution would be to host the images within the repo rather than at imgur.

Then you'd see image and cookie traffic from github.com and open a ticket about that next week ;) The only sensible options to avoid any kind of loading of stuff from the net are 

a) bundling all images with the server
b) encoding the images as `data` URL right into the file
c) not using any images at all in the CHANGELOG.

I'll go with c) since the only reason there was one image in there to begin with was to document a certain kind of log output in the update dialog, which can be done just as fine via a simple text embed.

foosel, 2018-11-26:
Fixed by the above commit, will be part of 1.3.10rc3 (and hence 1.3.10).

foosel, 2018-11-28:
1.3.10rc3 was just released.

commit 8292408c (2018-11-26): Remove image from changelog, replace with text

Closes #2909
closed 2018-11-28 by foosel