OctoPrint/OctoPrint #2910: [1.3.10-rc2] Print from SD causes repeated connect/disconnect GCodes

goeland86, 2018-11-25:
Hi Gina, feedback after I upgraded my printer from 1.3.9 to 1.3.10-rc2 today.

#### What were you doing?
This is all based on Redeem, and the steps described were working with 1.3.9.

1. Upload file to OctoPrint (in my case with Simplify3D's curl script)
2. Refresh SD files to list freshly uploaded files seen by Redeem
3. Select SD file (Redeem prefixes filenames with `/lcl/` to identify them)
4. Hit the Print button

#### What did you expect to happen?

Expected to see OctoPrint start the normal "Print from SD" process begin after issuing `M24`, starting with bed heating (was preheated), then heat up nozzle, then home, and start printing.

#### What happened instead?

As soon as `M24` was sent, OctoPrint said the print was in state Cancelled, and would apparently try to repeatedly reconnect to the serial port, as it was issuing `G28` in a loop, until I manually restarted Redeem through SSH to break the connection.

#### Did the same happen when running OctoPrint in safe mode?

Yes.

#### Version of OctoPrint

OctoPrint 1.3.10-rc2 as downloaded by the update plugin.

#### Operating System running OctoPrint

Umikaze 2.2.1-dev4 version - Replicape's BeagleBone Black image (I'm a Redeem developer)

#### Printer model & used firmware incl. version

Redeem's current `RC` branch, on the Monster Kossel v2 (https://www.thingiverse.com/thing:2854236)

#### Browser and version of browser, operating system running browser

Windows 10 / Firefox 63.0.3 64bit

#### Link to octoprint.log

Octoprint's log: https://pastebin.com/svFhkSPg

#### Link to contents of terminal tab or serial.log

Terminal output in safe mode: https://pastebin.com/cJ1XiWd7

#### Link to contents of Javascript console in the browser

Same as OctoPrint terminal output - restarted OctoPrint instance forced reload of UI.

#### Screenshot(s)/video(s) showing the problem:

Not UI issue.

I have read the FAQ.

foosel, 2018-11-26:
On first glance, it looks like the problem is this line received from the printer:

```
Recv: Not SD printing.
```

Looking further down it appears to be that the response to `M27` by redeem is always the current byte position, followed right after with `Not SD printing.`. 

All other firmwares that I've taken a look at will only ever print `Not SD printing` when no SD file is selected and no print from SD is ongoing. Redeem appears to have a different interpretation and that is one of the problems here (other potential issues I'm still looking into). 

[The only thing that we have resembling a spec](https://reprap.org/wiki/G-code#M27:_Report_SD_print_status) is also pretty clear on that matter:

> Marlin and RepRapFirmware report the number of bytes processed in this format, which can be processed by Pronterface:
>
> ```
> SD printing byte 2134/235422
> ```
>
>If no file is being printed, only this message is reported:
>
> ```
> Not SD printing.
> ```
 
That this didn't cause issues earlier is something I'd rather consider a bug in OctoPrint<=1.3.9 to be honest than a regression in 1.3.10rc2 ;) Still, taking a closer look now.

Why is the firmware sending this? Is this intentional and if so what's the reasoning here?

goeland86, 2018-11-26:
@foosel I do believe in this case it's a bug in Redeem - we've had a number of PR merges recently which may account for this.

I agree with your assessment that it was a bug in OctoPrint <=1.3.9 instead of in 1.3.10-rc2 - I was in a hurry to print something and didn't review the log in detail...

Thanks for the link to the spec you use (even if it's not official, at least we can build up to something that's in common with you). I will consider this issue as closed from OctoPrint's standpoint and raise it within Redeem to fix our firmware.

foosel, 2018-11-26:
> Thanks for the link to the spec you use

"spec" ;) The problem with this page is that it *is* a wiki that pretty much everyone can edit, so I take it more as a guideline and puzzle what is missing together from reading firmware source code.

What I've seen in other firmware (and hence consider the agreed upon standard communication here) is this:

  * no file selected: `Not SD Printing` or `SD printing byte 0/0` (the latter shouldn't exist but I've seen it in the wild)
  * file selected but not yet printing: `SD printing byte 0/235422`
  * file selected and printing: `SD printing byte 2134/235422`

goeland86, 2018-11-26:
Thank you for the further information :+1: 

We'll try to stick to the more common Marlin way of things, as that's been our general approach to GCode matching in general. Makes it easier for slicer compatibility too ;)

I understand what you mean regarding the editable wiki, but generally such changes are easily spotted if they don't match actual behavior from the code.

Once 1.4 comes out, we'll avoid most of the issue by building our own communication API with Redeem, which should simplify things from your end significantly.

foosel, 2018-11-26:
And while we are at it:

  * response to `M23`: 
    ```
    File opened:foo.gco Size:12345
    File selected
    ```
  * SD print finished: `Done printing file` in line
  * response to `M28`: `Writing to file` in line
  * response to `M29`: `Done saving file` in line

foosel, 2018-11-26:
Taking the liberty to close this as you are tracking it on Redeem's side now. Cheers!

closed 2018-11-26 by foosel