OctoPrint/OctoPrint #2911: [Request] End the Feedrate / Flowrate confusion

ragflyer, 2018-11-25:
related closed issue: #2093
I too suffer confusion of terminology and have to look it up every couple of weeks when the default setup does not give a great result rightaway.
It was no surprise to find a closed issue, given the amount of hits you get when searching for the combination of the terms flowrate and feedrate.
What people actually want is change "Speed" and "Extrusion factor".
I agree a standard terminology should not be changed.
May I suggest adding a tooltip or better an additional description to spare new and occasional users from googling so often? Would make a big difference in usability.

ntoff, 2018-11-26:
Give this a try https://github.com/ntoff/OctoPrint-Flowspeedrename You'll have to install it manually since it's not registered. It adds a page in octoprint's settings to let you change the buttons to whatever you want (not entirely sure how, or even if, it'll handle weird characters from languages other than english).

[image]

ragflyer, 2018-12-01:
Thanks, that works for me. Not sure about new users though.

ntoff, 2018-12-01:
and an unintended consequence of not sanitising the input, you can put fontawesome icons in there too

[image]

eg ``<i class="fa fa-arrows"></i> Speed``

JohnEdwa, 2018-12-06:
A setup item which would match the terms with your slicer and/or firmware could make sense (kinda like the camera controls on Fusion 360), but that would still leave the most common "Marlin with Cura" with Feed Rate and Flow. 

The iconography @ntoff added does look pretty spiffy though, maybe something like that could be implemented by default? Feed Rate could use a [speedo/tachometer gauge](https://icons8.com/icon/1412/speed), and Feed Rate is obviously [Line Width](https://icons8.com/icon/46246/line-width), it even has a graphic for underextrusion :P !

Also, I quote foosel, as she does it exactly as I do:
>I just remind myself that the only thing that flows is molten plastic, and that the stepper movements themselves are therefore the other thing ¯\(ツ)/¯

RichyT, 2019-03-14:
Are we stating that Feed rate does not affect extrusion speed? Because I have seen elsewhere stated that it does. What fun when it seems that the people seeking to clarify for the confused themselves get things wrong (one way or the other).

If the feed rate indeed doesn't affect the extrusion speed, then can we have a unified slider that affects both? if you're in the middle of a print, you'll either be under or over extruding if you move one before the other.

JohnEdwa, 2019-03-14:
It doesn't affect extrusion *multiplier*, aka flow rate, but it obviously changes the speed of the extruder as it has to turn faster to extrude the correct amount of plastic.

RichyT, 2019-03-14:
@JohnEdwa, OK. I get what you are saying but be careful with "flow rate". The slider is not specified as a rate but as a multiplier. It's not obvious that the flow increases with the XYZ speed from the layout of the buttons. I stopped writing because I totally agree with your edit.

JohnEdwa, 2019-03-14:
@RichyT The sliders in Octoprint are called "Feed rate: 100%" and "Flow rate: 100%", by any chance, the ones that change the [Feed Rate](https://en.wikipedia.org/wiki/Speeds_and_feeds#Feed_rate), the [Volumetric Flow Rate](https://en.wikipedia.org/wiki/Volumetric_flow_rate), so to me it would be kinda odd to not call the slider called "Flow Rate" the "Flow Rate" slider...

RichyT, 2019-03-14:
@JohnEdwa Ah, but 100% of what? Rate also has connotations of X per second. So the Feed rate increases the mm/s, wouldn't the flow rate increase the mm^3/s? Well no. And perhaps that should be obvious but as stated, the UI design visually associates the slider with the axis and not the extruder and so encourages confusion.

JohnEdwa, 2019-03-14:
@RichyT Well yes. "Flow Rate" multiplies the desired output speed of the extruder so if you set the "Flow Rate" to 110%, it will move the extruder motor 10% faster (measured in mm/s) and therefore push roughly 10% more plastic through the nozzle, which is volume that is measured in mm^3/s.

And I don't see how the UI would "visually associate" Flow Rate with an axis, to me it is quite clear that "Feed Rate" is in the same block as "X/Y Z" and other movement options, while "Flow Rate" is under the "Tool (E)" section with the "Extrude" and "Retract" buttons. [As can be seen in this screeshot.](http://dabbletron.com/wp-content/uploads/2015/09/OctoPrint-control-interface-2.png)

I don't actually have an issue with Octoprint, the UI is clear to me and it could say "asf" and "Dsa" for all I care. The issue is when I use my printer directly as I need to use "Feed", "Flow" and "FR" with no UI context.

RichyT, 2019-03-14:
I don't have anything against the flow rate button but its presence *coupled* with the Feed rate button being grouped under the axis controls invites one to not be sure that the feed rate effects the speed at which the extruder motor turns as well as the axis motors. To put it another way, *why* does the feed rate control *not* extend under the extruder control? There is no good reason but that it doesn't implies that there is a reason.

yoyo42, 2019-04-14:
Unconfuse me: if I set the speed to 110% does the whole job go faster (movement AND extrusion) or is it just movement, so you are effectively _reducing_ the extrusion?

Brandon2255p, 2019-07-29:
If you set the speed (feed rate) to 110% then extrusion is automatically increased in proportion to the speed so that the correct amount is extruded. The flow rate control stays at 100%.

I tested this by only adjusting feed rate to 150 % and there was no under-extrusion

PrestonL, 2020-02-16:
> If you set the speed (feed rate) to 110% then extrusion is automatically increased in proportion to the speed so that the correct amount is extruded. The flow rate control stays at 100%.
> 
> I tested this by only adjusting feed rate to 150 % and there was no under-extrusion

I feel like the interface needs to better reflect this, but I'm not sure the best way.  I'm just glad it's clarified! Maybe "Print Speed" instead of "Feed rate"?

hedva92, 2020-03-08:
is possible to increase feed rate range? I would like to change range up to 300 %.

Theaxiom, 2020-07-09:
@hedva92 you can use G-Code to modify feed rate: https://marlinfw.org/docs/gcode/M220.html

alex9099, 2021-05-30:
So at least on my printer by rotating the knob it modifies a percentage on the display, i guess that changes both the extrusion speed and the movement speed.

IMO these fields could have more "easily understandable" descriptions, at least something on the "?" (where it says it's not possible to grab the current value, etc)

jneilliii, 2021-05-30:
uh, it does?

[image]

[image]

alex9099, 2021-05-30:
I meant to show some more description on what is actually meant by "flow" and "feed". I (and likely more beginners) find those terms confusing