OctoPrint/OctoPrint #2912: Reverse proxy external connection issues 1.3.8.rc2

louispires, 2018-11-26:
#### What were you doing?

1. Via reverse proxy I am constantly being asked for my login details - Cookies doesn't seem to remember me (This is easily replicated by accessing OctoPrint with different devices eg. mobile and laptop)
2. Over Reverse Proxy I also get "Incorrect Password" issues often, even though its auto populated with Chrome...

#### What did you expect to happen?
1. Not to constantly be asked to Login, even with different devices
2. Reverse Proxy to always work and load correctly from outside my network

#### What happened instead?
Failures to load octoprint which eventually works but is painful in the beginning 

#### Did the same happen when running OctoPrint in safe mode?
Unable to test in Safe Mode - Got a 36 hour print ongoing :(

#### Version of OctoPrint
1.3.8.rc2

#### Operating System running OctoPrint
OctoPi 0.15.1

#### Printer model & used firmware incl. version
Ender-3 Pro - Marlin TH3D U1.R2.5

#### Browser and version of browser, operating system running browser
Chrome Windows 10 and Android Pie

#### Link to octoprint.log
https://gist.github.com/louispires/b7a596a84394e197622f1e56d135206e

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser
https://gist.github.com/louispires/de8fc1494b6f5650269bcf9090f9bdb6

#### Screenshot(s)/video(s) showing the problem:
https://imgur.com/a/K8lQr6K

I have read the FAQ.

foosel, 2018-11-27:
From what I'm seeing in your log, for some reason your client IP keeps happily toggling back and forth, which might be killing the session too:

```
2018-11-26 12:18:49,279 - tornado.general - ERROR - Attempted to attach to session 3y3sfyfr (197.234.242.133) from different IP (197.234.242.127)
2018-11-26 12:18:49,560 - tornado.general - ERROR - Attempted to attach to session 3y3sfyfr (197.234.242.133) from different IP (197.234.242.127)
2018-11-26 12:18:49,801 - tornado.general - ERROR - Attempted to attach to session 3y3sfyfr (197.234.242.133) from different IP (197.234.242.127)
[...]
2018-11-26 12:20:36,468 - tornado.general - ERROR - Attempted to attach to session 3y3sfyfr (197.234.242.127) from different IP (197.234.242.145)
2018-11-26 12:20:37,071 - tornado.general - ERROR - Attempted to attach to session 3y3sfyfr (197.234.242.127) from different IP (197.234.242.145)
[...]
2018-11-26 12:22:17,193 - tornado.general - ERROR - Attempted to attach to session mzswztev (197.234.242.145) from different IP (197.234.242.133)
2018-11-26 12:22:18,195 - tornado.general - ERROR - Attempted to attach to session o2ouqiht (197.234.242.145) from different IP (197.234.242.133)
2018-11-26 12:22:19,204 - tornado.general - ERROR - Attempted to attach to session o2ouqiht (197.234.242.145) from different IP (197.234.242.133)
[...]
2018-11-26 12:24:04,193 - tornado.general - ERROR - Attempted to attach to session kab5jg1j (197.234.242.133) from different IP (197.234.242.115)
2018-11-26 12:24:05,173 - tornado.general - ERROR - Attempted to attach to session kab5jg1j (197.234.242.133) from different IP (197.234.242.115)
[...]
2018-11-26 12:25:29,288 - tornado.general - ERROR - Attempted to attach to session kab5jg1j (197.234.242.115) from different IP (197.234.242.133)
2018-11-26 12:25:30,162 - tornado.general - ERROR - Attempted to attach to session kab5jg1j (197.234.242.115) from different IP (197.234.242.133)
[ad infinitum]
```

That is *not* normal and in fact I even had to `grep` the source to even figure out where torndo - or rather sockjs - is even logging this message since I've never before seen it. You might want to figure out why your IP is all over the place. Is there a load balancer in front?

Now, your NGINX configuration is lacking some crucial stuff to make everything work smoothly, especially on the websocket. Using your's stuff is definitely broken due to a lack of info forwarded to the server that it needs to properly serve requests, and the websocket is even broken in a ways the automatic fallback to xhr-polling doesn't even work properly. Take a look [here](https://discourse.octoprint.org/t/reverse-proxy-configuration-examples/1107) and adjust it accordingly. It should look roughly like this:

```
http {
    include mime.types;
    default_type application/octet-stream;
    sendfile on;
    keepalive_timeout 65;

    map $http_upgrade $connection_upgrade {
        default upgrade;
        '' close;
    }

    server {
        listen 80;
        server_name your_server_name.com;
        location / {
            proxy_pass http://10.0.0.169;
            proxy_set_header Host $http_host;
            proxy_set_header Upgrade $http_upgrade;
            proxy_set_header Connection "upgrade";
            proxy_set_header X-Real-IP $remote_addr;
            proxy_set_header X-Forwarded-For $proxy_add_x_forwarded_for;
            proxy_set_header X-Scheme $scheme;

            proxy_http_version 1.1;
        }

        client_max_body_size 1G;
    }
}
```

foosel, 2018-11-27:
What I was trying to say is that your configuration lacks important lines. As in, the way you've set it up will not work and cause a ton of issues. Try the configuration I pasted above that adds `Upgrade`, `Connection`, `X-Forwarded-For` and `X-Scheme` and also sets the proper HTTP version to use. Do not forget to add the `map` that makes `$http-upgrade` work.

The config that you currently have doesn't provide OctoPrint with everything it needs to know to generate proper URLs, and it also kills the websocket. The lines I added above are crucial and must not be left out if you expect things to work. 

And note that this is not a change with 1.3.10, but I'm guessing that you had connectivity problems before which simply didn't manifest as "in your face" as they do now but potentially caused other weird issues.

louispires, 2018-11-27:
I see - Thank you and I will make the changes required and report back shortly.

foosel, 2018-11-27:
Glad it could be solved :)