OctoPrint/OctoPrint #2913: 1.3.10.RC2 Issues

nionio6915, 2018-11-26:
3 issues encountered with RC2-

1. Firmware updater plugin issues: understood that this is known issue at time or wiritng. 

2. Temp callback error: it is understood that it is realted to a 3rd party plugin... but how to get Kunsi's attention and hold it long enought to review?  Attached are logs. 

3. Most importantly,  issues autoconnecting to printer. Printer is conventional MEGA2560+RAMPS1.4 connected via USB to raspi. I have been lulled into confidence with it since it is just so stable in its operation with 1.3.9.

connection would timeout or drop completetly after a short time in operation. 

logs attached: 

https://drive.google.com/open?id=1LOh0SG_RzVVhSp2YKA4QRwKYzCnJgyoT

jneilliii, 2018-11-26:
On your number 2 item listed, I assume that's the MQTT plugin?  I am one of the contributors on that plugin and actually think @tedder already has a fix prepped for release to resolve.

jneilliii, 2018-11-26:
Yeah, I posted a pull request to the firmware updater plugin that does octoprint version checks and changes the line based on that.  Not sure if it's the right way, but it does work.

nionio6915, 2018-11-26:
any thoughts on why the octoprint and the printer are not initiating or dropping communication?

foosel, 2018-11-27:
> any thoughts on why the octoprint and the printer are not initiating or dropping communication?

Based on your `serial.log`, the only problem that is see is that your first connection attempt was at the wrong baudrate (115200) whereas the successful connection was then at the correct baudrate (250000).

The sudden disconnects I think are actually a side effect of the hd44780 plugin error, at least they correlate heavily. I suggest to disable that plugin/start in safe mode and see if you can reproduce this.

As already stated in #2889 the firmwareupdater situation should be resolved with rc3 (I added a backwards compatibility layer so a patch on the plugin's side isn't mandatory even though recommended).

nionio6915, 2018-11-29:
I am a more confused on this now-

1. is the temp error a result of PSU control which HD44780 pulls from or is it HD44780?

2. Do I do anything more on this or go onto RC3 and see what happens? I already rolled this back to 1.3.9 to get printing again. I am happy to upgrade to RC2 and test again but it does not seem fruitful.

foosel, 2018-11-29:
> * is the temp error a result of PSU control which HD44780 pulls from or is it HD44780?

I do not know, which is why I asked you to disable that plugin and see if it still reproduces then.

> * Do I do anything more on this or go onto RC3 and see what happens? I already rolled this back to 1.3.9 to get printing again. I am happy to upgrade to RC2 and test again but it does not seem fruitful.

Upgrade to RC3, disable the HD44780 plugin, see if the issue still reproduces. If no, the plugin is the problem. If yes, provide `serial.log` and `octoprint.log` from that reproduction again.

closed 2018-12-10 by foosel