OctoPrint/OctoPrint #2914: RC2 issues

Charly333, 2018-11-26:
1. Start print, wait for second layer with normal print temp
2. Reduce temp to 200 del

#### What did you expect to happen?
new Temp send is taken over from printer

#### What happened instead?
Print falls with to room temp.

It only could be saved whilst setting temp to old temp of 220 and whilst nozzle is heating up again, switching to 200 as long nozzle didn´t reach that temp.

#### Did the same happen when running OctoPrint in safe mode?

 All printers busy printing for a couple of days

#### Version of OctoPrint

#### Operating System running OctoPrint

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

ntoff, 2018-11-27:
Can't do anything without the logs, please provide a serial log as well as octoprint.log https://discourse.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299

foosel, 2018-11-27:
Thank you for reporting this!

Sadly your report lacks some of the information that is needed in order to look into this issue:

  * whether you have been able to reproduce this issue in safe mode too
  * your [exact version of OctoPrint](https://discourse.octoprint.org/t/how-can-i-find-out-the-version-of-octoprint-or-octopi-i-am-running/204/1)
  * your [exact version of OctoPi](https://discourse.octoprint.org/t/how-can-i-find-out-the-version-of-octoprint-or-octopi-i-am-running/204/1) (should you be running it)
  * the operating system running OctoPrint (name *and* version please)
  * unmodified full length [`octoprint.log`](https://discourse.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299/1)
  * contents of the terminal tab, or even better a [`serial.log` (please enable that first though, it's not being written by default)](https://discourse.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299/1)
  * the contents of [your JS error console](https://webmasters.stackexchange.com/questions/8525/how-to-open-the-javascript-console-in-different-browsers)

(If you need help with locating this stuff, please take a look [into the Contribution Guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#what-should-i-include-in-a-bug-report) which explains everything - and if anything is still unclear then just ask)

Could you please provide that so this issue can be analyzed and ideally reproduced and fixed? Thank you!

closed 2018-12-10 by foosel