OctoPrint/OctoPrint #2916: "One-Off" error wanted to report.

KodySalak, 2018-11-28:
#### What were you doing?

1. Was printing a regular print, and OctoPrint crashed.

#### What did you expect to happen?
For the print to keep going...?
#### What happened instead?
The OctoPrint python client crashed.
#### Did the same happen when running OctoPrint in safe mode?
Does not apply, this was a one-off and I wanted to report it.

#### Version of OctoPrint
1.3.9

#### Operating System running OctoPrint
Windows Server 2012 R2/Python 2.7.14

#### Printer model & used firmware incl. version
Creality Ender 3 / "Marlin V1; Sprinter/grbl mashup for gen6"

#### Browser and version of browser, operating system running browser
Chrome 70.0.3538.110 (Offical) / Windows 10 1803

#### Link to octoprint.log
https://pastebin.com/rWuzvfAq

#### Link to contents of terminal tab or serial.log
NA, was unable to capture this information.

#### Link to contents of Javascript console in the browser
NA, was unable to capture this information.

#### Screenshot(s)/video(s) showing the problem:
NA, was unable to capture this information.

I have read the FAQ.
Yes.

foosel, 2018-11-28:
Thank you for the thorough report! 

Looks like that's an issue with Tornado under Windows, see tornadoweb/tornado#1802. How many clients did you have connected when this happens? I have never before seen this particular error message pop up and as far as I understand its origins it should also only ever be possible to happen if there are significantly more connections held open to the server than is common with OctoPrint.

I'm not sure if that can be fixed in any way, but it can possibly be mitigated by changing the keepalive values as suggested in the linked ticket. I'll take a look into that.

In general though I'd honestly recommend to not use OctoPrint on Windows machines in production - yes, it works thanks to it being cross platform, but running a full fledged Windows machine for it (with potential unexpected reboots and other fun) kinda defeats the whole purpose.

KodySalak, 2018-11-28:
> Huh, then I have no idea how that could even happen. I develop this
> software under Windows, so I should have seen this by now if it was a
> general problem (I have more than one connection open regularly). Odd.

Definitely is, not too sure what’s up! Thought I would give you a heads up
though. :D

KodySalak, 2018-12-11:
@foosel It happened again tonight, logs are here: https://pastebin.com/cY00ijNQ

KodySalak, 2018-12-11:
Same environment variables as before, I think I'm going to update to the RC and see if it stops.

foosel, 2018-12-11:
The RC is no longer, 1.3.10 stable got released yesterday. But I'd be surprised if that solves this.

foosel, 2018-12-12:
I've pushed a commit that sets the idle timeout of tornado to 10min, but only under windows, as suggested in the thread linked above. Available on `maintenance`, soon `devel`, to be rolled out with 1.3.11. That commit will *not* solve the problem as that's an inherent problem with how Tornado works on Windows, but hopefully makes it less likely to run into this.

I have still not been able to reproduce it myself under regular use, so something about your environment must be different. Possibly some network tooling that keeps pinging the server, using up connections. I remember reading that some Netgear routers were fairly aggressive in that regard. If you keep running into this even after 1.3.11 with this patch goes live, you might want to switch to a Linux based setup since this patch is the only thing I'm able to do here. As you can see in the linked thread, the official recommendation from the Tornado people is "don't use Windows" in such situations as well.

KodySalak, 2018-12-12:
Sounds good, Gina! Thank you so much!

foosel, 2019-05-15:
1.3.11 was released yesterday.

closed 2019-05-15 by foosel