OctoPrint/OctoPrint #2917: [RC Feedback] Feedback on the 1.3.10rc3 Release Candidate

foosel, 2018-11-28:
Please provide *general* feedback on your experience with the 1.3.10rc3 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

**If you run into any obvious bugs not yet listed below the following line, please open a new ticket and follow ["How to file a bug report"](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report).**

---

### Currently known bugs

*none*

### Unreproduced

*none*
 
### Unreproduced and lacking necessary information to look into, awaiting full bug report

  * ❓ [Not possible to complete Anonymous Usage Tracking setup wizard](https://discourse.octoprint.org/t/anonymous-usage-setup/5151)
  * ❓ ["temperature changes doesn't work"](https://github.com/foosel/OctoPrint/issues/2889#issuecomment-441449215)

benlye, 2018-11-28:
I can confirm that the current version (v1.0.0) of the Firmware Updater plugin works with RC3 👍

When I've more time I'll make necessary changes to use CommandLineCaller wrapper instead of invoking Sarge directly.

ChrisHeerschap, 2018-11-28:
RC2 has been working well for me with the exception that the fantastic Exclude Region plugin stopped working - see https://github.com/bradcfisher/OctoPrint-ExcludeRegionPlugin/issues/15#issuecomment-439594024 - just tested with RC3 and it looks like it's still broken. Plugin maintainer is aware and investigating, don't know if you can provide any helpful feedback.

Running a test print in RC3 and will do the usual click-around-and-look-for-any-issues.

skohls, 2018-11-28:
Using RC3 on the first print, so far w/o problems.
I installed the development version of the Exclude Region plugin and this seems to work now as well... (at least I can create areas, modify them etc.)
https://github.com/bradcfisher/OctoPrint-ExcludeRegionPlugin/archive/devel.zip

tteckenburg, 2018-11-28:
Hi Gina,

still dinglish... i only keep the firmware for testing ;-)

[image]

ChrisHeerschap, 2018-11-28:
I think it's _way_ more attention getting in German, though. 😆

ChrisHeerschap, 2018-11-28:
Test print ran fine, clicked around everything I could think of, looking good!

bradcfisher, 2018-11-29:
I released a new version (0.1.3) of the [Exclude Region plugin](https://github.com/bradcfisher/OctoPrint-ExcludeRegionPlugin) to address the issues it had with the 1.3.10 release candidates as reported by @ChrisHeerschap.  The behavior was not due to anything wrong in OctoPrint, and was a result of how the plugin was detecting Gcode viewer enhancements that were added in the 1.3.10 releases.

JohnOCFII, 2018-11-29:
Upgrade from rc2 to rc3 went well.  I completed a 3.5 hour print to a Prusa MK3 running (slightly older) firmware 3.4.0.  
- Timelapse worked fine.  
- Print History plugin worked fine.  
- OctoSlack plugin worked fine.
- Telegram plugin worked fine.

Goodeid, 2018-11-30:
The upgrade from rc2 to rc3 was perfect and after numerous prints with both versions there is only one thing that is missing and one anomaly..
You removed the "printed" and :visited" layer display in the Gcode Viewer and I understand the confusion that some people had with them. But I sure miss the "printed" layer display as it was directly above the "Layer Number" and the two provided a quick. simple progress indication.
Second the "Approx Total Print Time" in the State area does not match the "Estimated total print time" on the Gcode Viewer tab. I realize that these are only approximations, but should they not be the same?

foosel, 2018-11-30:
> You removed the "printed" and :visited" layer display in the Gcode Viewer and I understand the confusion that some people had with them. But I sure miss the "printed" layer display as it was directly above the "Layer Number" and the two provided a quick. simple progress indication.

It was simply causing confusion too often, sorry.

> Second the "Approx Total Print Time" in the State area does not match the "Estimated total print time" on the Gcode Viewer tab. I realize that these are only approximations, but should they not be the same?

[image]

Yes and now. They are different implementations, so there can be a difference in the result. It might make sense to remove the GCODE version in a future version and replace it with the server side estimate, however then the layer estimates would no longer add up.

Goodeid, 2018-11-30:
Regarding the "printed" and "visited" layers I fully understand the confusion they caused. I was hoping that the "printed layers" display could return without the "visited layers" portion of the line.

andrivet, 2018-12-01:
Confirm also that Firmware Updater plugin works with RC3. Updating RC2 to RC3 went well.

foosel, 2018-12-03:
1.3.10rc4 is out, new ticket is here: https://github.com/foosel/OctoPrint/issues/2939

closed 2018-12-03 by foosel