OctoPrint/OctoPrint #2919: Print Bed moves all the way down right before print starts.

cartergray22, 2018-11-29:
#### What were you doing?

Trying to print a simple test part using octoprint with Slic3r, Cura, or flashprint.  After the bed and extruder is heated properly and all axes are homed properly, the z axis suddenly goes all the way to the bottom of my printer before extruding material way up in the air above the bed.
Please be as specific as possible here. The maintainers will need to
reproduce your issue in order to fix it and that is not possible if they
don't know what you did to get it to happen in the first place.

1. ...
2. ...
3. ...

#### What did you expect to happen?
I thought the print would work as it shows in the g code visuilizer, g code, or as it does without octoprint.
#### What happened instead?
Explained in "what were you doing"
#### Did the same happen when running OctoPrint in safe mode?
My FF creator pro doesnt connect in safe mode. "no more baudrates found"

#### Version of OctoPrint

1.3.9

#### Operating System running OctoPrint

Windows 7    Octopi 0.15.1

#### Printer model & used firmware incl. version

Flashforge Creator Pro ,  marlin

#### Browser and version of browser, operating system running browser

Google Chrome ,  Edge

#### Link to octoprint.log

<script src="https://gist.github.com/cartergray22/29881bf3fe7ab5b3ec94e1360243b61c.js"></script>
<script src="https://gist.github.com/cartergray22/9b08b521c421759a4ad2bf1644d4541e.js"></script>
<script src="https://gist.github.com/cartergray22/08a6bd952cc91c22582ed6e396d9cbb3.js"></script>

#### Link to contents of terminal tab or serial.log

<script src="https://gist.github.com/cartergray22/4f6ce247a25813ba7e7fd0b69941b5dd.js"></script>

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2018-11-29:
OctoPrint never sends commands on its own, so the instructions to move the print bed must either originate in your GCODE file (please share one) or be generated by some plugin, possibly GPX. 

The only z related movement command I see in the `serial.log` is

```
2018-11-29 04:55:51,050 - Send: N8 G1 Z5 F5000*12
```

but I can't say whether that is the culprit here or not - I know too little about GPX based printers. I also see that right after issuing the heatup command your printer got disconnected due to a timeout. Was that your doing or is that another effect here?

Maybe @markwal could take a look at this and confirm or deny that it might be a GPX issue.

markwal, 2018-11-29:
Yes.  Well... Not GPX exactly but a mismatch between reprap and MakerBot.

MakerBot and derivitives traditionally have start gcode that home and define the center of the bed as 0,0. Reprap defines the lower left hand corner as 0,0. So... What you do is tell the slicer that the 0,0 is the center of the bed (they all default the other way) in settings.  You should also tell Octoprint the same thing but all that practically affects is where the viewer shows the print on the bed.

cartergray22, 2018-11-29:
> Yes.  Well... Not GPX exactly but a mismatch between reprap and MakerBot.
> MakerBot and derivitives traditionally have start gcode that home and define the center of the bed as 0,0. Reprap defines [...]

I have tried all combinations of the origin from front left corner to center of bed in both the slicer software printer profile, and in octoprint itself.  None of these changes are making a difference except for changing what the g code visualizer shows on octoprint as you mentioned above.  Although, in addition to the Z axis moving all the way down, regardless of my positioning in the slicer software and origin setup, the print starts in the back right corner of the bed (very close to home coordinates), leading me to think that it is an origin defining issue, but nothing changes when I try to change the origin.

cartergray22, 2018-11-29:
> Was that your doing or is that another effect here?

Yes, this was my doing.  I was doing this because as soon as heating was finished and the print "should start", the bed then moves from its seemingly correct position to the very bottom before beginning the print.  I would hit cancel print on the machine, but whenever I do that while connected to octoprint it cancels the print as usual but the bed does not ever stop going down, even after it crashes into the endstop.  So long story short, to not crash the printer I just turn it off and back on when I see this same problem happening.  I am away from the printer for a bit, so I will get you the g code and g code scripts as soon as I can.

foosel, 2018-11-30:
As far as I understood @markwal this is outside of my control since it has something to do with the GPX translation happening under the hood. So chances are that I cannot do anything here.

I'll await the GCODE file and scripts, but from the looks of your `serial.log` so far, this isn't core OctoPrint's doing.

cartergray22, 2018-12-05:
Sorry all for the delay, I wasn't able to get to the files until today but here are the files as requested.    

The g code scripts from OctoPrint are as follows:

Before print job starts

After print job completes

After print job is cancelled

G91; Set to Relative position
G1 E-1 F300; retract the filament a bit before lifting the nozzle
G0 Z15; move z axis up 15
G28 X; home X axis
G1 Y150 F5000; move part out for inspection
G90;  Set to Absolute position
M104 S0 ; turn off extruder heat
M140 S0 ; turn off heated bed
M106 S0 ; Turn off fan
M18; Move Freely
After print job is paused

M117 Print Paused
G91
G1 Z10 F1000
G90
Before print job is resumed

M117 Print Resumed
G91
G1 Z10 F1000
G90
Before tool change

After tool change

After connection to printer is established

Before connection to printer is closed

cartergray22, 2018-12-05:
<script src="https://gist.github.com/cartergray22/91a9e29f1a1a074ba38c069e8c966cce.js"></script>

Here is the G Code that I am trying to run on my printer

markwal, 2018-12-05:
The problem is your "start gcode" in Slic3r. You might want to look at my blog post on the matter at the following link.  I'm a little out of date, but the gcode you have doesn't define the coordinate system properly for MakerBot style 0,0 center. 

http://markwal.github.io/programming/2016/02/18/ffcp-start-and-end-gcode.html

closed 2018-12-05 by markwal