OctoPrint/OctoPrint #2920: Failed to install plugins during restore

varazir, 2018-11-29:
#### What were you doing?

1. Go to 'Backup & restore in the settings'
2. Click on 'Browse...'
3. Clicked on Upload and restore
4. All looked OK no errors. 
5. Missing plugins 

E-mailing backups 

#### What did you expect to happen?

#### What happened instead?

#### Did the same happen when running OctoPrint in safe mode?

#### Version of OctoPrint

OctoPrint 1.3.10rc3 

#### Operating System running OctoPrint

OctoPi 0.15.1

#### Printer model & used firmware incl. version

#### Browser and version of browser, operating system running browser

Windows 10, Chrome 

#### Link to octoprint.log

#### Link to contents of terminal tab or serial.log

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2018-11-29:
Please don't forget the `octoprint.log`s from restoring and ideally also backup creation, like we talked about on IRC. Thanks!

varazir, 2018-11-29:
Sent a e-mail with the backup

foosel, 2018-11-30:
Got the backups, thank you, but I'm still lacking *the logs*. I need the logs from the instances, not the ones in the backup (they might not have captured everything). Please provide the logs here as discussed.

varazir, 2018-11-30:
Hmm you mean system logs or OctoPrint logs ? the newer backup have all
the logs that are in the Octoprint/logs . Can dubble check that they
are the same.

foosel, 2018-11-30:
When I say logs I always mean `octoprint.log` (and `serial.log` if applicable) ;) Also says so in the bug reporting template. 

The backups do not contain the restore logs as they were naturally overwritten during restore. And I just realized you also can't produce these since well... they were overwritten. So never mind.

I just checked... you have GCODE System Commands disabled per your config. Fullscreen Webcam had a mismatching identifier on the repository, that will be fixed soon. PidTune and Display ETA aren't registered on the repository. So that explains the non-restored plugins that you so far told me about. 

However, on my reproduction I noticed a TON of wrongly reported compatibility issues that led to a lot of other plugins not being installed. Quick look into the code and I now know what's up, there's a bit of wrong logic in there determining compatibility, and that's causing this issue. So I guess RC4 it will have to be after all 😕 Still, good this was caught in time.

foosel, 2018-11-30:
Yeah, already on that.

foosel, 2018-11-30:
Fixed the issue. Also `logs` won't be part of the backup anymore, neither will be `generated` and `watched`. Makes more sense this way. Fixes will be released with 1.3.10rc4.

varazir, 2018-11-30:
Well I think logs would be good to have in the backup. If you use backup/restore part of troubleshooting.

foosel, 2018-11-30:
The plan was to have the restore log in place, since that is the more informative anyhow. However I just noticed an issue with that. Correcting this is too invasive for an RC fix, so logs will have to wait until 1.3.11.

foosel, 2018-12-03:
1.3.10rc4 is out

commit a3f2e0b5 (2018-11-30): backup: fix plugin compatibility check on restore

Fixes #2920
closed 2018-12-03 by foosel