OctoPrint/OctoPrint #2921: Prusa i3 MK3 crash resume

wbatzle, 2018-11-29:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

wbatzle, 2018-11-29:
I would like to start out by thanking you for all your hard work. This is the best thing since water. I recently had a print crash on my Prusa. I could not tell what was wrong until I went to my office and took a look. The head crashed into the print I was doing and of course stopped with their new crash detection feature. It's a really nice feature. I would love to see you guys add a resume print option to the controls if possible. I would do it myself but I have no friggin clue what I am doing with that kind of stuff. Anyway, thanks again for all your hard work.

Sincerely,
Will

foosel, 2018-11-30:
I agree that this would be neat, however resuming a print is sadly not as trivial as it sounds since OctoPrint doesn't know which of the commands it sent to the printer had already been executed at point of failure (multiple buffers inside the firmware). There's also already a feature request for that in #1876 so I'm closing this one.

closed 2018-11-30 by foosel

joesolly, 2019-05-22:
I think this is doable without worrying about the buffers. The prusa printer pauses when a crash or other fault is detected (which would be nice if octoprint could display the status of) and then will resume the print once resolved. So I think this can be solved by adding support for the prusa pause conditions and adding the appropriate resume command option. Thanks for listening, let me know if I can help out with this.

foosel, 2019-05-22:
That would be a very printer model specific solution however and hence something that should be done via a plugin and not in core OctoPrint.