OctoPrint/OctoPrint #2939: [RC Feedback] Feedback on the 1.3.10rc4 Release Candidate

foosel, 2018-12-03:
Please provide *general* feedback on your experience with the 1.3.10rc4 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

**If you run into any obvious bugs not yet listed below the following line, please open a new ticket and follow ["How to file a bug report"](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report).**

---

### Currently known bugs

*none*

### Unreproduced

*none*
 
### Unreproduced and lacking necessary information to look into, awaiting full bug report

  * ❓ [Not possible to complete Anonymous Usage Tracking setup wizard](https://discourse.octoprint.org/t/anonymous-usage-setup/5151)
  * ❓ ["temperature changes doesn't work"](https://github.com/foosel/OctoPrint/issues/2889#issuecomment-441449215)

ChrisHeerschap, 2018-12-03:
Upgraded and ran a small test print with success. Also tested new version of ExcludeRegion plugin and everything is looking good!

paukstelis, 2018-12-03:
I have issues with all the release candidates that are likely related to web access/authentication setup. There is no issue in 1.3.9.

What happens:
No websocket connection.
Browser console log contains 
`can’t establish a connection to the server at ws://<server>/octoprint/sockjs/322/4imior1c/websocket`
Apache config for reverse proxy looks like this:

> ProxyPass http://127.0.0.1:5000/
>   ProxyPassReverse http://127.0.01:5000/
>   RequestHeader set X-SCRIPT-NAME /octoprint/
>   RequestHeader set X-SCHEME http
>   RewriteEngine on
>   RewriteCond %{HTTP:UPGRADE} ^WebSocket$ [NC]
>   RewriteCond %{HTTP:CONNECTION} Upgrade$ [NC]
>   RewriteRule .* ws://127.0.0.1:5000%{REQUEST_URI} [P]

tteckenburg, 2018-12-04:
Thx Gina, works!!

[image]

Now I 've to upgrade my Fimrware ;)

foosel, 2018-12-04:
@paukstelis I'm not sure I'm seeing the exact same problem. At first I ran into a 500 being reported on the websocket since I had forgotten to enable `proxy_wstunnel`. Now I'm seeing a 404:

```
WebSocket connection to 'ws://octopi3.lan/octoprint/sockjs/750/nkaav5vb/websocket' failed: Error during WebSocket handshake: Unexpected response code: 404
```

which seems to be caused by the `/octoprint/` in the URL not being removed before it gets sent to OctoPrint, which in turn is probably caused by the Apache config - I have to take another look at that, it mirrors the one [recommended here]() but apparently it has a bug for prefix paths. In any case I'm seeing the exact same behaviour on 1.3.9, which makes me think this is not what you are experiencing. 

Can we continue this in [its own bug report over here](https://github.com/foosel/OctoPrint/issues/2941)?

JohnOCFII, 2018-12-07:
Sorry for my late input, I was out of town and away from my computer.  I did the upgrade from rc3 to rc4 last night.  All went smoothly.  I did one successful test print this morning, and have started on a much larger print now.

All seems fine so far.

foosel, 2018-12-10:
1.3.10 has been released, so I'm closing this now.

@stough I was not able to reproduce this, please open a full ticket, provide test files and logs, thanks!

closed 2018-12-10 by foosel