OctoPrint/OctoPrint #2940: Do not pause Octoprint on M0/M1

NewbieYnet414, 2018-12-04:
**Requested feature**
Hello it would be nice to have option disable pausing Octoprint on M0/M1 if printer support M0/M1.

**Is your feature request related to a problem? Please describe.**

**Background story**
I have 3D printing service on dormitory with open access to Průša i3 MK3. Every member of our voluntary organization can request printing but because of lack of knowledge I'm one of few which loads print jobs to Octoprint. And because I'm not always at dormitory or have time go to printer someone else clean heatbed.
**Point**
I installed print queue plugin which send M0 to printer after each print job. And wait for someone to clear heatbed and confirm that on printer. That idea works pretty well as we are printing non-stop but unpausing print jobs from Octoprint is little bit redundant and we loosing time on that.

**Describe the solution you'd like**
Add option to disable pause in Octoprint on M0/M1.

**Describe alternatives you've considered**
M117 could not pause Octoprint but this is not universal.

foosel, 2018-12-04:
[Already implemented](https://github.com/foosel/OctoPrint/issues/1513#issuecomment-386807154).

closed 2018-12-04 by foosel

foosel, 2018-12-05:
[Action commands](http://docs.octoprint.org/en/master/features/action_commands.html) were implemented in OctoPrint for exactly these kinds of cases. If you want to pause via the firmware, then the firmware also needs to properly report back to OctoPrint that it got unpaused. If it does, your described scenario will already work. If it doesn't, how should OctoPrint know that it got unpaused on the firmware side?

If I disable pausing on `M0`/`M1` but forward the command to the printer, OctoPrint will continue to send data to the now blocked printer, fill up its buffer and best case get the communication stuck, worst case make the controller crash due to too much data. Unless the firmware happens to send the `pause` action command to OctoPrint when it receives an `M0`/`M1` (which it probably won't if it doesn't also send the `resume` command on resume).

To summarize: Implementing your request would break things severely and is behaviour that simply doesn't make sense from a protocol and controller behaviour perspective. I rather suggest you weigh in on the feature request on the firmware tracker that you already found to add this basic functionality to the firmware.