OctoPrint/OctoPrint #2942: Random crashes

martinowren, 2018-12-04:
#### What were you doing?
Starting a print from octoprint and print stops. Octoprint still states that it is printing, but not on the printer interface

#### What did you expect to happen?
To not crash and continue printing
#### What happened instead?
Octoprint states its still printing, but nothing is happening and printer display says "Heating Done". 
#### Did the same happen when running OctoPrint in safe mode?
No diffenrece

#### Version of OctoPrint
1.39 

#### Operating System running OctoPrint
Octopi 0.15.1

#### Printer model & used firmware incl. version
Prusa i3 MK3 3.5.0

#### Browser and version of browser, operating system running browser
Chrome/Edge/Firefox Windows 10

#### Link to octoprint.log
https://pastebin.com/TjVYT5J9

foosel, 2018-12-04:
Thank you for reporting this!

Sadly your report lacks some of the information that is needed in order to look into this issue:

  * contents of the terminal tab, or even better a [`serial.log` (please enable that first though, it's not being written by default)](https://discourse.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299/1)
  * the contents of [your JS error console](https://webmasters.stackexchange.com/questions/8525/how-to-open-the-javascript-console-in-different-browsers)
  * screenshots or maybe a video showing the problem

(If you need help with locating this stuff, please take a look [into the Contribution Guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#what-should-i-include-in-a-bug-report) which explains everything - and if anything is still unclear then just ask)

Could you please provide that so this issue can be analyzed and ideally reproduced and fixed? Thank you!

martinowren, 2018-12-04:
JS console:
[image]
Serial.Log: https://pastebin.com/8UEnb86N

foosel, 2018-12-04:
So that looks like the bed never finishes heating. You are saying the whole server crashes though? There's nothing at all in the log, so it's probably the whole python process that dies. This only ever happens on heating? What Pi are you running this on (3B+?)?

Please try the following:

  * ssh into OctoPi (user: pi, password: raspberry (if you haven't changed it))
  * stop the server: `sudo service octoprint stop`
  * check if your Pi is complaining about any issues: `vcgencmd get_throttled` - provide the output here please
  * start `screen`: `screen`
  * inside screen start OctoPrint from the console, in debug and safe mode: `~/oprint/bin/octoprint --host 127.0.0.1 --debug --safe` - a ton of stuff will be scrolling by, that's normal
  * trigger the issue or wait for it to occur
  * check the output of the server in the `screen` session (if you've closed the window you can reattach by SSHing into OctoPi again and running `screen -x`). There'll hopefully be some kind of error there that gives us some more insight into what is happening. Copy everything from the last `Server heartbeat` message until the end of the output and share it here (via pastebin.com or gist.github.com)

martinowren, 2018-12-04:
Hi!
Process happens randomly. While heating or mid print. Doesnt even have to print anything.
vcgencmd get_throttled: throttled=0x50005
start screen: screen does not work. 
Does not recognize the command

foosel, 2018-12-04:
> vcgencmd get_throttled: throttled=0x50005

So before we continue here, get your Pi another power supply. The one you are currently using is not giving it enough power and Pis that are browning out due to such low power situations can cause the weirdest kind of issues, including random crashes. It doesn't make sense to analyse this further until this command is returning `0x0` and hence low power is ruled out as reason for the problem.

See also [here](https://discourse.octoprint.org/t/i-had-a-strange-lightning-bolt-temperature-symbol-with-an-exclamation-mark-popup-in-my-navbar-what-does-that-mean/5256).

> start screen: screen does not work.

Huh? I was fairly sure that's part of the stock OctoPi image. Well, in that case, `sudo apt update && sudo apt install screen`.

martinowren, 2018-12-04:
So I have upgraded the power supply to 2.5A. Should be plenty. Still showing the same 0x50005

foosel, 2018-12-05:
Could also be your cable between supply and Pi. Or that power supply delivers slightly under 5V. If it's saying it's throttled due to undervoltage (which that code means) then something is not right yet.

martinowren, 2018-12-05:
Hi! Cable is changed. I can guarantee that there is not a problem with the power with this psu and cable. Used it on another Rpi.

martinowren, 2018-12-05:
Still 0x50005

foosel, 2018-12-05:
Then either your Pi is broken or there's still something wrong with either your power supply or the USB cord after all.