OctoPrint/OctoPrint #2944: [Devel] AttributeError: 'FilebasedUserManager' object has no attribute 'register_callback'

koenkooi, 2018-12-07:
#### What were you doing?

I am trying to update my octoprint instance running on 'devel' now that that branch has seen some activity. 

#### What did you expect to happen?

A working instance.

#### What happened instead?

Spinning dots in the browser, errors in the log:
```

#### Did the same happen when running OctoPrint in safe mode?

<!-- 
Test if you can reproduce your problem in safe mode. You can find information
on how to enable safe mode in the Contribution Guidelines.

If you can't reproduce in safe mode, this is a bug with one of your
installed third party plugins. Don't open a ticket here!

If you can't test this in safe mode, state why.
-->

#### Version of OctoPrint

<!--
Can be found in the lower left corner of the web interface. ALWAYS INCLUDE.
-->

#### Operating System running OctoPrint

<!--
OctoPi, Linux, Windows, MacOS, something else? With version please.
OctoPi's version can be found in /etc/octopi_version or in the lower left
corner of the web interface.
-->

#### Printer model & used firmware incl. version

<!--
If applicable, always include if unsure.
-->

#### Browser and version of browser, operating system running browser

<!--
If applicable, always include if unsure.
-->

#### Link to octoprint.log

<!--
On gist.github.com or pastebin.com. ALWAYS INCLUDE and never truncate.

The Contribution Guidelines tell you where to find that.
-->

#### Link to contents of terminal tab or serial.log

<!--
On gist.github.com or pastebin.com. If applicable, always include if unsure or
reporting communication issues. Never truncate.

serial.log is usually not written due to performance reasons and must be
enabled explicitly. Provide at the very least the FULL contents of your
terminal tab at the time of the bug occurrence, even if you do not have
a serial.log (which the Contribution Guidelines tell you where to find).
-->

#### Link to contents of Javascript console in the browser

<!--
On gist.github.com or pastebin.com or alternatively a screenshot. If applicable -
always include if unsure or reporting UI issues.

The Contribution Guidelines tell you where to find that.
-->

#### Screenshot(s)/video(s) showing the problem:

<!--
If applicable. Always include if unsure or reporting UI issues.
-->

I have read the FAQ.

koenkooi, 2018-12-11:
Just tested with 17ad63dabf90da8beb2b01f7fdf7c93681f54c81, issue is stil present.

foosel, 2018-12-12:
Should be fixed in e14cb44

koenkooi, 2018-12-13:
Fix confirmed!
```
Dec 13 09:31:12 raspberrypi2 octoprint[11267]: 2018-12-13 09:31:12,782 - octoprint.startup - INFO - ******************************************************************************
Dec 13 09:31:12 raspberrypi2 octoprint[11267]: 2018-12-13 09:31:12,786 - octoprint.startup - INFO - Starting OctoPrint 1.4.0.dev1705+ge14cb44.dirty
Dec 13 09:31:12 raspberrypi2 octoprint[11267]: 2018-12-13 09:31:12,787 - octoprint.startup - INFO - ******************************************************************************
```

closed 2018-12-13 by koenkooi