OctoPrint/OctoPrint #2945: "Suppress temperature messages" suppresses M115 output

OutsourcedGuru, 2018-12-08:
#### What were you doing?

1. Go to Terminal tab
2. Click all three checkboxes to quiet the output, mostly importantly "Suppress temperature message"
3. Enter `M115`

#### What did you expect to happen?

I expected to see the output text:

```
Send: M115
Recv: FIRMWARE_NAME:Marlin 1.1.7-C2 (Github) SOURCE_CODE_URL:https://github.com/Robo3D/Marlin-C2 PROTOCOL_VERSION:C2 MACHINE_TYPE:RoboC2 EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff EMERGENCY_CODES:M108,M112,M410
```
#### What happened instead?

I did not see the output coming back from the firmware. It was accidentally hidden by the regex.

#### Did the same happen when running OctoPrint in safe mode?

Yes.

#### Suggested fix (confirmed on mine)
Updated the regex from what it was to this:

```
(Send: (N\d+\s+)?M105)|(Recv:\s+(ok\s+)?.* (B|T\d*):\d+)
```

It's subtle, but there is now a single space character before the `B|T` digit detection. **I have confirmed that this now works.**

#### Version of OctoPrint
OctoPrint 1.3.10rc4 running on OctoPi 0.15.0

#### Operating System running OctoPrint
Raspbian Stretch on Raspberry Pi 3B
`Linux charming-pascal 4.14.62-v7+ #1134 SMP Tue Aug 14 17:10:10 BST 2018 armv7l GNU/Linux`

#### Printer model & used firmware incl. version
Robo C2, Marlin 1.1.7-C2 (Github)

#### Browser and version of browser, operating system running browser
Safari 12.0.1 on macOS Mojave 10.14.1

#### Link to octoprint.log
[octoprint.log](https://pastebin.com/raw/ic4dSBsb)

#### Link to contents of terminal tab or serial.log

```
[...]
Send: M115
Recv: ok
[...]
```

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

OutsourcedGuru, 2018-12-10:
Yeah, that's the terror of regex; it's almost as bad as Fermat's Last Theorem. _Prove_ that it won't break some strange firmware out there.

On the plus side, all of this appears to be just filtering for a view for the user rather than actually changing anything. The downside risk is that someone might see temperature commands when they didn't want to see them or conversely, someone like me wouldn't see something which they expected to see.

I wonder if there's a way of saying `(space OR a single character (NOT N OR is a space))` rather than just a space before that `T` recognition. `( |(^N| ))` maybe but I'd have to play in a terminal for a bit to confirm.

FanDjango, 2019-01-19:
You want the original rexex followed by a "AND NOT ^FIRMWARE_NAME:" for example.
What you really want is a two staged regex, as the support in regex for "AND NOT" is negligible, you cannot really do it in one regex.
Couldn't one allow chaining of one or more regex'es by using a separator? If octoprint would then execute them in order using the result of the previous one (pipe them), you could solve this problem.

FanDjango, 2019-01-21:
I've checked all the postives, they all match.

```
(Send: (N\d+\s+)?M105)|(Recv:\s+(ok\s+)?(B|T\d*):\d+)
```
[image]

And the negative:

[image]

Works for me

foosel, 2019-01-28:
I've adjusted the default regex in the above commit. On `maintenance`, soon `devel`, to be shipped with 1.3.11.

foosel, 2019-01-28:
Had to revert the change as I forgot the advanced ok case:

    Recv: ok N5993 P15 B15 T:59.2 /0.0 B:31.8 /0.0 T0:59.2 /0.0 @:0 B@:100:

which now broke.

commit dd692d6f (2019-01-28): :wrench: Adjust default regex of temperature terminal filter

Should no longer match lines like

    FIRMWARE_NAME:Marlin 1.1.7-C2 (Github) SOURCE_CODE_URL:https://github.com/Robo3D/Marlin-C2 PROTOCOL_VERSION:C2 MACHINE_TYPE:RoboC2 EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff EMERGENCY_CODES:M108,M112,M410

Fixes #2945
commit 904b3c6c (2019-01-28): :wrench: Adjust default regex of temperature terminal filter (2nd try)

Should no longer match lines like

    FIRMWARE_NAME:Marlin 1.1.7-C2 (Github) SOURCE_CODE_URL:https://github.com/Robo3D/Marlin-C2 PROTOCOL_VERSION:C2 MACHINE_TYPE:RoboC2 EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff EMERGENCY_CODES:M108,M112,M410

but should pass unit tests.

Fixes #2945
closed 2019-05-14 by foosel