OctoPrint/OctoPrint #2946: No more messages in websocket since update to v1.3.10

anthonyst91, 2018-12-10:
Hello Gina,

Since the update from v1.3.09 to v1.3.10, there are no more messages in the websocket.

Steps to reproduce:
- open websocket "ws://192.168.1.30/sockjs/websocket" using a simple web socket client
- wait a few seconds: connection is established
- only message {"connected": {"display_version": "1.3.10", "apikey": "18B6761E426549FB89C7CD6B9E7D33CB", "branch": "HEAD -> master", "plugin_hash": "3c9f178fb30bec4b9b0461942f398d8a", "config_hash": "bcf6b1c5543b308f316e3c02bb81d915", "version": "1.3.10", "safe_mode": false, "debug": false}} is showing
- there are no more messages after this one

May something has changed in OctoPrint v1.3.10 about the web socket? 

Best regards,

Anthony

jubaleth, 2018-12-10:
Hi @anthonyst91,

There is nothing that needs correcting on the OctoPrint side. You need to send an auth message in the socket... This has been documented (http://docs.octoprint.org/en/maintenance/api/push.html  -- note: the master branch of the docs doesn't seem to be updated, that will be looked into tomorrow) and a notification sent out to third party devs (who opted in) warning of this. I'll point out that there have been 4 RCs released over the last month, this is really something you should have already seen in preparation for the stable release.

closed 2018-12-10 by jubaleth

anthonyst91, 2018-12-10:
Hi @jubaleth,

Thanks for your quick feedback. Indeed I did not get the info. My bad, I had to register to these emails before...

Unfortunately I had to test my app with the latest 1.3.10 RC and did not noticed this "issue". Everything was still working fine, even on fresh installs. Today I've re-installed a Raspberry Pi with a fresh install of Raspbian + the lastest 1.3.10 stable, and the behavior starts to occur. I don't understand why...

Moreover, after updating to v1.3.10 stable, some of my users noticed that the websocket was still working fine (whereas I did not updated the app to login & send the auth message over the socket). That's strange too!