OctoPrint/OctoPrint #2947: [Request] Disable printer_safety_check warning for Monoprice Select Mini V2 with firmware v4.1

9R, 2018-12-10:
**Is your feature request related to a problem? Please describe.**
The printer safety warning is displayed on all Monoprice Select Mini printers that match MODEL=="m200"
At least the latest firmware version of the v2 has thermal_runaway protection. I have tested this by disconnection the thermistor wire from the control board while heating up the extruder. The extruder will continue heating up for about 60 seconds, then the printer will stop heating, turn the fan to 100%, disconnect the serial connection ignore all further commands & will only become operational after a power cycle. I have attached the serial log output of the process below. During the test I monitored the extruder temperature with an handheld infrared thermometer and can confirm, that heating actually stopped.

**Describe the solution you'd like**
Check the firmware version on M200 printers and do not display the warning about missing thermal runaway protection if the firmware version is 4.1 or above.
According to the firmware change log, the protection was added in 4.0 but to be safe I suggest to wait until someone with a printer with that version confirms that it is indeed included and functional before removing the warning for printers with v4.0.

**Describe alternatives you've considered**

**Additional context**

Relevant excepts from the attached log:

Printer model and version string
```
Send: N1 M115*39
Recv: NAME. Malyan	VER: 4.1	MODEL: M200	HW: HH02
```

Heating starts
```
Send: N8 M104 S180*100
Recv: ok N8 P15 B15
Send: N9 M105*46
Recv: ok N9 P15 B15 T:41.4 /180.0 B:31.1 /0.0 T0:41.4 /180.0 @:127 B@:0
Send: N10 M105*22
Recv: ok N10 P15 B15 T:41.2 /180.0 B:31.0 /0.0 T0:41.2 /180.0 @:127 B@:0
Send: N11 M105*23
Recv: ok N11 P15 B15 T:41.7 /180.0 B:30.7 /0.0 T0:41.7 /180.0 @:127 B@:0
Send: N12 M105*20
Recv: ok N12 P15 B15 T:43.7 /180.0 B:30.8 /0.0 T0:43.7 /180.0 @:127 B@:0
Send: N13 M105*21
Recv: ok N13 P15 B15 T:46.5 /180.0 B:31.0 /0.0 T0:46.5 /180.0 @:127 B@:0
Send: N14 M105*18
Recv: ok N14 P15 B15 T:50.2 /180.0 B:30.6 /0.0 T0:50.2 /180.0 @:127 B@:0
```

disconnecting thermistor cable
```
Send: N25 M105*16
Recv: ok N25 P15 B15 T:109.7 /180.0 B:30.8 /0.0 T0:109.7 /180.0 @:127 B@:0
Send: N26 M105*19
Recv: ok N26 P15 B15 T:115.1 /180.0 B:30.7 /0.0 T0:115.1 /180.0 @:127 B@:0
Send: N27 M105*18
Recv: ok N27 P15 B15 T:26.2 /180.0 B:31.0 /0.0 T0:26.2 /180.0 @:127 B@:0
Send: N28 M105*29
Recv: ok N28 P15 B15 T:26.0 /180.0 B:30.3 /0.0 T0:26.0 /180.0 @:127 B@:0
```

thermal runaway protection take over and stops the printer
```
Recv: ok N39 P15 B15 T:25.7 /180.0 B:30.7 /0.0 T0:25.7 /180.0 @:127 B@:0
Send: N40 M105*19
Recv: ok N40 P15 B15 T:26.0 /180.0 B:30.8 /0.0 T0:26.0 /180.0 @:127 B@:0
Recv: Error:Heating failed, system stopped! Heater_ID: 0
Changing monitoring state from "Operational" to "Error: Heating failed, system stopped! Heater_ID: 0"
Changing monitoring state from "Error: Heating failed, system stopped! Heater_ID: 0" to "Offline (Error: Heating failed, system stopped! Heater_ID: 0)"
Connection closed, closing down monitor
```

mattyams, 2018-12-10:
I second this.

9R, 2018-12-12:
Wow, that was quick!
Just tested with `maintenance` branch. Warning message is gone. Thx

closed 2018-12-12 by 9R