OctoPrint/OctoPrint #2948: Connect button loads as disconnect, but cannot connect/disconnect to printer

tdub415, 2018-12-11:
#### What were you doing?

1. Go to octoprint via octoprint.local or IP address
2. Click on connection
3. Scroll down to connect/disconnect button
4. See error: defaults to 'disconnect' with auto-connect not enabled. Tried enabling auto-connect but does not work, clicking on disconnect does nothing, cannot connect to printer.

#### What did you expect to happen?
On reboot after updating, to be able to click the connect button to connect to printer
#### What happened instead?
On reboot, unable to connect to printer.  Button is showing 'disconnect' but printer never connected
#### Did the same happen when running OctoPrint in safe mode?
Yes, the same issue occurs in safe mode.

#### Version of OctoPrint
1.3.10

#### Operating System running OctoPrint
Raspbian on a Pi3 (not OctoPi) uname -a result: Linux octoprint 4.14.30-v7+ #1102 SMP Mon Mar 26 16:45:49 BST 2018 armv7l GNU/Linux

#### Printer model & used firmware incl. version
Prusa Mk3, 3.4.1

#### Browser and version of browser, operating system running browser
Chrome Version 70.0.3538.110, Windows 10

#### Link to octoprint.log
https://pastebin.com/CvQq3QTk

#### Link to contents of terminal tab or serial.log
Terminal tab is completely blank

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:
https://i.imgur.com/uO3VNc6.png

I have read the FAQ.

tdub415, 2018-12-11:
`/#temp:1 This page includes a password or credit card input in a non-secure context. A warning has been added to the URL bar. For more information, see https://goo.gl/zmWq3m.
3/#temp:1 Active resource loading counts reached to a per-frame limit while the tab is in background. Network requests will be delayed until a previous loading finishes, or the tab is foregrounded. See https://www.chromestatus.com/feature/5527160148197376 for more details
packed_libs.js?eebed41b:10084 Active resource loading counts reached to a per-frame limit while the tab is in background. Network requests will be delayed until a previous loading finishes, or the tab is foregrounded. See https://www.chromestatus.com/feature/5527160148197376 for more details
packed_core.js?8f3c09b2:14700 Starting dependency resolution...
packed_core.js?8f3c09b2:14810 ... dependency resolution done
packed_core.js?8f3c09b2:15113 Initial application setup done, connecting to server...
packed_core.js?8f3c09b2:13096 Connected to the server
packed_core.js?8f3c09b2:15091 Finalizing application startup
packed_core.js?8f3c09b2:14975 Going to bind 37 view models...
packed_core.js?8f3c09b2:15028 Did not bind view model UsageViewModel to target #wizard_plugin_tracking since it does not exist
packed_core.js?8f3c09b2:15028 Did not bind view model CuraViewModel to target #wizard_plugin_cura since it does not exist
packed_core.js?8f3c09b2:15028 Did not bind view model SoftwareUpdateViewModel to target #softwareupdate_confirmation_dialog since it does not exist
packed_core.js?8f3c09b2:15028 Did not bind view model SoftwareUpdateViewModel to target #wizard_plugin_softwareupdate since it does not exist
packed_core.js?8f3c09b2:2267 User tdub415 logged in
packed_core.js?8f3c09b2:15067 ... binding done
packed_core.js?8f3c09b2:15087 Application startup complete`

tdub415, 2018-12-11:
https://pastebin.com/BVMaKE2z sorry, formatting didnt copy over

foosel, 2018-12-12:
Have you tried what's outlined in [the FAQ entry I linked](https://discourse.octoprint.org/t/the-web-interface-seems-stuck-not-responsive-broken/5675)? It looks like you've run into what's described under **Missing socket authentication** and following the commands described there will fix it.