OctoPrint/OctoPrint #2950: Octoprint GUI asks for a reload even though there is no new server (but after a client sleep)

Mirarkitty, 2018-12-12:
#### What were you doing?

Opening Octoprint interface tab in my browser.

1. open octoprint in browser
2. set computer to sleep (not tested: pick another tab first)
3. wake up computer
4. octoprint now asks PLEASE RELOAD even though nothing has changed on server

#### What did you expect to happen?

Octoprint behaving nicely without PLEASE RELOAD.

#### What happened instead?

Octoprint asks PLEASE RELOAD.

#### Did the same happen when running OctoPrint in safe mode?

Not tested. Doesn't seem relevant.

#### Version of OctoPrint

#### Operating System running OctoPrint

OctoPrint 1.3.10 running on OctoPi 0.15.0PE 

#### Browser and version of browser, operating system running browser

Mozilla Firefox 63 under Linux.

#### Link to contents of Javascript console in the browser

Will check next time it happens.

#### Screenshot(s)/video(s) showing the problem:

[image]

...I have read the FAQ.

foosel, 2018-12-12:
Thank you for reporting this!

Sadly your report lacks some of the information that is needed in order to look into this issue:

  * whether you have been able to reproduce this issue in safe mode too
  * unmodified full length [`octoprint.log`](https://discourse.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299/1)
  * the contents of [your JS error console](https://webmasters.stackexchange.com/questions/8525/how-to-open-the-javascript-console-in-different-browsers)

(If you need help with locating this stuff, please take a look [into the Contribution Guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#what-should-i-include-in-a-bug-report) which explains everything - and if anything is still unclear then just ask)

Could you please provide that so this issue can be analyzed and ideally reproduced and fixed? Thank you!

Mirarkitty, 2018-12-13:
If you think safe mode and octoprint.log will help; I think it's a client javascript issue. I will provide.

Mirarkitty, 2018-12-15:
...it seemed more useful that I'll start debugging. So far,

```
            if (versionChanged || pluginsChanged || configChanged) {
                console.log("dataupdater",versionChanged,pluginsChanged,configC\
hanged);
                showReloadOverlay();
            }

```
yielded false false true. I'll pursue some more.

Mirarkitty, 2018-12-17:
It does reconnect and the config hash is changing. I'll try to see why, is it suppose to change on reconnect?
`The connection to ws://172.16.2.28/sockjs/050/4hrww4qd/websocket was interrupted while the page was loading. packed_libs.js:24036:9
onconnect hash 81528b6f491e636669947280a5791b7e 77469fdc74e2995dd13556843c32208c packed_core.js:13683:6
Connected to the server packed_core.js:13722:9
dataupdater false false true packed_core.js:13715:3
config hash 81528b6f491e636669947280a5791b7e 77469fdc74e2995dd13556843c32208c packed_core.js:13716:3
`

(blah) hash (old) (new)

foosel, 2018-12-17:
> I'll try to see why, is it suppose to change on reconnect?

No, not unless something has updated the config in the meantime. The hash is basically a hash over the currently effective config. That usually shouldn't change all the time. The thing is that a change in the config can have consequences for the generated HTML code (also in the code of third party plugins, hence I cannot just use a whitelist here to limit detected changes to only those affecting the HTML code). Hence the prompt to reload.

> It does reconnect (I'm not sure what's in a connection, but the computer is sleeping for more than an hour)

Ah, sorry, I forgot that this was happening after sending the computer to sleep.

Mirarkitty, 2018-12-17:
where is `data["config_hash"];` generated? (And which packet does it come in?)

Mirarkitty, 2018-12-17:
Never mind, I think I found it...
```
        @property
        def config_yaml(self):
                import yaml
                return yaml.safe_dump(self._config)

        @property
        def config_hash(self):
                import hashlib
                hash = hashlib.md5()
                hash.update(self.config_yaml)
                return hash.hexdigest()

```

Lets see if I can dump the two configs and see what's changed...

Mirarkitty, 2018-12-18:
Moving issue there. https://github.com/fabianonline/OctoPrint-Telegram/issues/156

Mirarkitty, 2018-12-18:
Thanks for the help.

closed 2018-12-18 by Mirarkitty

mark-hahn, 2019-03-11:
I'm having the same problem and I'm not running telegram.

jneilliii, 2019-03-11:
Do you have and if my power plugins installed?

Mirarkitty, 2019-03-16:
Easiest is to find the config file, copy it, next time it asks for a reload do a `diff` and see what changed in the config. It will point you directly to the issue.

TheKrush, 2019-10-29:
In case anyone else is using it I believe PrintTimeGenius modifies the config after every print and has this say issue https://github.com/eyal0/OctoPrint-PrintTimeGenius/issues/164