OctoPrint/OctoPrint #2951: OctoPrint is missing errors, continues to happily bang out gcode.

AndKe, 2018-12-12:
#### What were you doing?
Printing, or so I thought for 40 minutes.

#### What did you expect to happen?
I expected the OctoPrint to stop on error(s)  - displaying, or at least keeping the original error in terminal

#### What happened instead?
Then it was "done", suspiciously fast, I could see in the terminal that most of the commands were ignored by the printer. Printer stopped, surely telling a reason, but now the error is lost, no clue as to what went wrong.

[image]

#### Did the same happen when running OctoPrint in safe mode?
no, this is more about  detecting error strings, than an OctoPrint fail that would require safe mode..

#### Version of OctoPrint
1.3.10

#### Operating System running OctoPrint
OctoPi, Linux

#### Printer model & used firmware incl. version
Marlin, 1.1.8  / rather fresh build.

#### Browser and version of browser, operating system running browser
Chromium

#### Link to octoprint.log
https://gist.github.com/AndKe/2e2450f0464af378b5862db44d1169b8

#### Link to contents of terminal tab or serial.log
"2018-12-12 08:27:51,949 - serial.log is currently not enabled, you can enable it via Settings > Serial Connection > Log communication to serial.log"

I guess enabling it could help..

#### Link to contents of Javascript console in the browser
not relevant

#### Screenshot(s)/video(s) showing the problem:

attached above

foosel, 2018-12-12:
Thank you for reporting this!

Sadly your report lacks some of the information that is needed in order to look into this issue:

  * contents of the terminal tab, or even better a [`serial.log` (please enable that first though, it's not being written by default)](https://discourse.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299/1)
  * since this could be an issue related to specific GCODE file properties, one or more GCODE files with which to reproduce this

(If you need help with locating this stuff, please take a look [into the Contribution Guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#what-should-i-include-in-a-bug-report) which explains everything - and if anything is still unclear then just ask)

Could you please provide that so this issue can be analyzed and ideally reproduced and fixed? Thank you!

foosel, 2018-12-13:
Setting it to awaiting information until we have a `serial.log`.

Patronics, 2018-12-13:
I've had a similar problem, specifically with the prusa i3 mk3 when it has the "mintemp error" when starting a print when the printer is below a temperature threshold (below which the firmware is not sure if it's a valid temperature reading or a failed sensor). I don't have any logs to upload, sorry, but I hope this helps a bit.

Patronics, 2018-12-13:
Ok, I've got the error again for me, here's the serial.log [serial.log-2.download.zip](https://github.com/foosel/OctoPrint/files/2677322/serial.log-2.download.zip). (apparently octoprint slightly messes up the size estimate or something, so macOS considers it a failed download, but it's still viewable in a text editor fine.)

AndKe, 2018-12-13:
@Patronics that's nasty, your log does not have a clear error, just one OK , and the next line is the "stopped due to errors" message,  that is a bug(missing error) in firmware....  

your firmware is based on Marlin 
FIRMWARE_NAME:Prusa-Firmware 3.4.1 based on Marlin FIRMWARE_URL:https://github.com/prusa3d/Prusa-Firmware PROTOCOL_VERSION:1.0 MACHINE_TYPE:Prusa i3 MK3 EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000

- don't know what version of Marlin, ... maybe it's fixed now ?

Patronics, 2018-12-13:
Yeah, it says on the printer's LCD "mintemp error" meaning the temperature's below a threshold that it trust the temperature sensor (15°C apparently), but it's weird it doesn't give that detail to octoprint. It's running the latest firmware available from Prusa.

AndKe, 2018-12-13:
I'll check current master tomorrow, and make a PR with proper "ERROR:" if needed.
however, I am printing at a cold storage place outside house, and think I have seen clear minerror in OctoPrint at least once, (forgor to reduce mintemp settings on new printer, and it was only 2°C.)

AndKe, 2018-12-14:
@foosel which of these strings would you detect as a problem ?

```
#define MSG_ERR_EEPROM_WRITE                "Error writing to EEPROM!"

#define MSG_STOP_BLTOUCH                    "STOP called because of BLTouch error - restart with M999"
#define MSG_STOP_UNHOMED                    "STOP called because of unhomed error - restart with M999"
#define MSG_KILL_INACTIVE_TIME              "KILL caused by too much inactive time - current command: "
#define MSG_KILL_BUTTON                     "KILL caused by KILL button/pin"
```
let me guess, OctoPrint will stop on STOP* and KILL*  ?

foosel, 2018-12-14:
It will stop on anything that [is actually marked as an error and considered fatal](https://reprap.org/wiki/G-code#Replies_from_the_RepRap_machine_to_the_host_computer), so prefixed with either `Error:` or `!!` and not one of the non fatal strings listed there (which I collected from various firmwares). 

Btw, make sure that you haven't disable error handling altogether for whatever reason:

[image]

AndKe, 2018-12-20:
I reproduced mintemp error on current Marlin 1.1.9  
MINTEMP call kill()  , which even disconnects the serial port,
```
Recv: Error:MINTEMP triggered, system stopped! Heater_ID: 0
Recv: Error:Printer halted. kill() called!
```
 and OctoPi handles that as expected:
[image]
[image]

foosel, 2019-01-10:
@AndKe I'm a bit confused now - is this now solved through a firmware fix or what? If not I still need a full `serial.log` from you, @AndKe, because I can't know if it's the same or different from what @Patronics saw.

AndKe, 2019-01-10:
Yes, after enabling full logging, I updated Marlin to current master 1.1.9 (was release 1.1.8 before) , and now I have never seen any unhandeled kill since. 
I will close this case for now.

closed 2019-01-10 by AndKe