OctoPrint/OctoPrint #2952: Communication timeouts with Azteeg X5 GT (LPC1769) running recent Marlin

alexxy, 2018-12-13:
#### What did you expect to happen?

Normal printing

#### What happened instead?

Random communication timeouts between Marlin and Octoprint running on orangepi

#### Did the same happen when running OctoPrint in safe mode?

Didnt tested

#### Version of OctoPrint

1.3.9, 1.3.10
#### Operating System running OctoPrint
Linux

#### Printer model & used firmware incl. version
CoreXY, Marlin firmware

#### Browser and version of browser, operating system running browser
Doesnt matter (usualy Chromium, Linux)

#### Link to octoprint.log
Logs are at https://github.com/MarlinFirmware/Marlin/issues/12591

#### Link to contents of terminal tab or serial.log
Same here https://github.com/MarlinFirmware/Marlin/issues/12591

alexxy, 2018-12-13:
And i tested serial communication with python script. Seems like it doesnt have issues...

foosel, 2018-12-13:
Thank you for reporting this!

Sadly your report lacks some of the information that is needed in order to look into this issue:

  * whether you have been able to reproduce this issue in safe mode too
  * contents of the terminal tab, or even better a [`serial.log` (please enable that first though, it's not being written by default)](https://discourse.octoprint.org/t/where-can-i-find-octoprints-and-octopis-log-files/299/1)
  * screenshots or maybe a video showing the problem

(If you need help with locating this stuff, please take a look [into the Contribution Guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#what-should-i-include-in-a-bug-report) which explains everything - and if anything is still unclear then just ask)

Could you please provide that so this issue can be analyzed and ideally reproduced and fixed? Thank you!

On a first rough look, it looks like your firmware might be sending an `Error:` message with non-ASCII characters that's causing issues here, same as what you ran into with pronsole. Communication between hosts and printers is supposed to be ASCII only.

alexxy, 2018-12-13:
About py scrtips. I run them for last 24 hours and didnt get any timeouts....

foosel, 2018-12-13:
> But i dont see why i need to attach video of printer what waits for input due to serial timeouts since its simply doesnt move head =)

Because I'm not seeing your printer, I'm not seeing your OctoPrint instance, and visuals can really help with debugging.

In any case...

Your pi scripts do not implement resend handling at all. They also ignore any `Error` messages sent by the printer. They are not valid benchmarks for a working printer-host-communication since they ignore most of the protocol requirements. 

First thing in your `serial.log` that looks broken:

```
2018-12-05 17:31:34,232 - Recv: Error:���� �������������� ������������������
```

Your printer is sending an `Error` with non ASCII data there. This is also what's causing the encoding error in `octoprint.log` and my guess is also in pronsole. You need to figure out why it's doing that and what that error is actually about. In any case, the firmware should never send non ASCII data, so this is a firmware bug.

Further down we encounter this:

```
2018-12-05 17:36:00,501 - Unexpected error while reading serial port, please consult octoprint.log for details: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:2605
```

This means you've either unplugged the printer or your connection is unreliable. If it's the latter, this needs to be looked into and fixed.

The communication timeouts look like the ping pong protocol implementation is running into a possible race condition somewhere, but it's pretty impossible to say for certain since you don't have extended oks enabled and the oks therefore lack the line number they are acknowledging. If you could reproduce this with this option enabled in your firmware, that might help to ascertain that this is in fact a bug on OctoPrint's side and not your firmware stopping to respond out of the blue - right now I'm unable to say and I'd rather make sure before I try to debug further.

Finally, your `octoprint.log` states that you are in fact not running 1.3.9 or 1.3.10 but the `devel` branch and 1.4.0.dev1332:

```
Starting OctoPrint 1.4.0.dev1332+gb01e327f
```

Please see if you can even reproduce the issue under current stable 1.3.10. 

1.4.0.dev is currently even more bleeding edge than usual since I haven't been able to test it myself for ages thanks to 1.3.10 preparations eating up all my available time.

alexxy, 2018-12-13:
Some of logs are from 1.3.10rc. Some from 1.3.9. I can try to reproduce it with 1.3.10. (I did clean install of it)

alexxy, 2018-12-13:
> Because I'm not seeing your printer, I'm not seeing your OctoPrint instance, and visuals can really help with debugging.
>
> In any case...
>
> Your pi scripts do not implement resend handling at all. [...]

Im not sure what actualy causing this.

> Further down we encounter this:
>
> ```
> 2018-12-05 17:36:00,501 - Unexpected error while reading serial port, please consult octoprint.log for details: SerialException: 'device reports readiness to [...]

Printer was manualy reseted. About hw reliability. I checked cables, and changed usb cable to 100% working. just to be sure that its not a cable problem.

> The communication timeouts look like the ping pong protocol implementation is running into a possible race condition somewhere, but it's pretty impossible to say for certain since you don't have [...]

I'm also thinking about some kind of races. But i cannot gues from where they come from

> Finally, your `octoprint.log` states that you are in fact not running 1.3.9 or 1.3.10 but the `devel` branch and 1.4.0.dev1332:
> 
> ```
> Starting OctoPrint 1.4.0.dev1332+gb01e327f
> ```
> Please see if you can even reproduce the issue under current stable 1.3.10.

After i see this issue with 1.3.9 i tryed 1.3.10rc2,rc3 

> 
> 1.4.0.dev is currently even more bleeding edge than usual since I haven't been able to test it myself for ages thanks to 1.3.10 preparations eating up all my available time.

foosel, 2018-12-13:
> Some of logs are from 1.3.10rc. Some from 1.3.9.

I went through the logs you linked above. The first one was still at 1.3.9. Then you upgraded to 1.4.0.dev. I don't see any 1.3.10 in the mix:

```
$ grep "OctoPrint 1.3.9" *.log.*
octoprint.log.2018-12-05:2018-12-05 14:19:35,179 - octoprint.server - INFO - OctoPrint 1.3.9
$ grep "OctoPrint 1.3.10" *.log.*
$ grep "OctoPrint 1.4.0" *.log.*
octoprint.log.2018-12-05:2018-12-05 15:20:42,644 - octoprint.startup - INFO - Starting OctoPrint 1.4.0.dev1332+gb01e327f
octoprint.log.2018-12-06:2018-12-06 15:20:48,086 - octoprint.server - INFO - OctoPrint 1.4.0.dev1332+gb01e327f
octoprint.log.2018-12-07:2018-12-07 15:20:48,299 - octoprint.server - INFO - OctoPrint 1.4.0.dev1332+gb01e327f
octoprint.log.2018-12-08:2018-12-08 15:20:48,483 - octoprint.server - INFO - OctoPrint 1.4.0.dev1332+gb01e327f
octoprint.log.2018-12-09:2018-12-09 15:20:48,661 - octoprint.server - INFO - OctoPrint 1.4.0.dev1332+gb01e327f
octoprint.log.2018-12-10:2018-12-10 15:20:48,849 - octoprint.server - INFO - OctoPrint 1.4.0.dev1332+gb01e327f
```

> After i see this issue with 1.3.9 i tryed 1.3.10rc2,rc3

Again, your logs tell a different story. After 1.3.9 you immediately switched to 1.4.0.dev on December 5th around 15:20 o'clock and stayed there.

> [non ASCII `Error`]

Is that error reproducible? If so I suggest to first of all make sure that you have your printer set to english - sometimes translated messages make it into the serial messages, which - since those have to be machine consumable and also have to adhere to ASCII - is obviously a bad idea, but it still happens. If it then shows up in english and ASCII, that might give us something to work with (in case that error is even related, it's not possible to say right now).

> > [communication timeouts]
> 
>I'm also thinking about some kind of races. But i cannot gues from where they come from

Hence my suggestion to recompile the firmware with advanced ok and see if that gives us more data to work with.

foosel, 2018-12-13:
> > Your pi scripts do not implement resend handling at all. They also ignore any Error messages sent by the printer. They are not valid benchmarks for a working printer-host-communication since they ignore most of the protocol requirements.
> 
> Just for some context, the scripts were designed to fail on any communication error on purpose, they are purely to test whether there is any error in the serial communication link, such as loss of data in the internal buffers. I had alexxy run them because his logs showed resend requests which because the link is native USB CDC should never happen, unlike the usual serial or cdc-serial bridge connections that are fragile.

Oops, missed that... Thanks for the clarification @p3p. I guessed something like this was the idea behind them, but felt it was important to point out that they only test the link level, not the protocol level.

alexxy, 2018-12-13:
> I went through the logs you linked above. The first one was still at 1.3.9. Then you upgraded to 1.4.0.dev. I don't see any 1.3.10 in the mix:
>
> ```
> $ grep "OctoPrint 1.3.9" *.log.* [...]

Its 1.3.10rcX compiled from git tag. Something like this
git checkout -b 1.3.10rc3 1.3.10rc3

> Is that error reproducible? If so I suggest to first of all make sure that you have your printer set to english - sometimes translated messages make it into the serial messages, which - since those [...]

Communication timeouts are random (thats why i thought about some kind of race). But it can be reproduced with long prints

> > > [communication timeouts]
> > 
> > 
> > I'm also thinking about some kind of races. But i cannot gues from where they come from
> 
> Hence my suggestion to recompile the firmware with advanced ok and see if that gives us more data to work with.

Will try.

foosel, 2018-12-14:
> I have Anet A8 and 1.3.10 same problem :( even i change in Marlin MSG_OK to uppercase.
> 
> still timeout :( in safe mode too

To quote [the contribution guidelines](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#what-should-i-do-before-submitting-a-bug-report):

> **Very important:** Please make absolutely sure that if you find a bug that looks like
> it is the same as your's, it actually behaves the same as your's. E.g. if someone gives steps
> to reproduce his bug that looks like your's, reproduce the bug like that if possible,
> and only add a "me too" if you actually can reproduce the same
> issue. Also **provide all information** as [described below](#what-should-i-include-in-a-bug-report)
> and whatever was additionally requested over the course of the ticket
> even if you "only" add to an existing ticket. The more information available regarding a bug, the higher
> the chances of reproducing and solving it. But "me too" on an actually unrelated ticket
> makes it more difficult due to on top of having to figure out the original problem
> there's now also a [red herring](https://en.wikipedia.org/wiki/Red_herring) interfering - so please be
> very diligent here!

Also 

> even i change in Marlin MSG_OK to uppercase.

Congratulations, you just broke your firmware's implementation of [the protocol](https://reprap.org/wiki/G-code#Replies_from_the_RepRap_machine_to_the_host_computer). Roll that back please, that's definitely breaking things.

foosel, 2018-12-14:
> Its 1.3.10rcX compiled from git tag. Something like this
> git checkout -b 1.3.10rc3 1.3.10rc3

Please don't do that. That confuses the version number reporting (as could be clearly seen). Just install 1.3.10 stable, reproduce and we'll take it from there. I need you to be on an official, released, current version here.

alexxy, 2018-12-14:
Ok. I changed OctoVersion to release one

```
2018-12-14 21:48:38,883 - octoprint.startup - INFO - ******************************************************************************
2018-12-14 21:48:38,888 - octoprint.startup - INFO - Starting OctoPrint 1.3.10
2018-12-14 21:48:38,889 - octoprint.startup - INFO - ******************************************************************************
```
I slightly modyfyed script for spiral test, so now it generates gcode that i will send via octoprint to printer.
So i get ~7.5M file with gcode.

alexxy, 2018-12-15:
Hmm.. Seems like i cannot reproduce issue with 1.3.10 and syntetic test (not real printing). Will try with some real object

alexxy, 2018-12-19:
I found that is `garbage` in console.
Its message in russian (set as display locale) "Не удалось прощупать"
So its message that says Probing failed

And it comes from https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/module/probe.cpp#L813

closed 2018-12-28 by GitIssueBot