OctoPrint/OctoPrint #2953: Total number of layers not displayed in GCode Viewer

biggreenogre, 2018-12-14:
#### What were you doing?
Printing
1. Start print
2. Switch to GCode Viewer
3. Look for total number of layers to print

#### What did you expect to happen?
Total number of layers to print would be visible

#### What happened instead?
Total number of layers to print was not visible

#### Did the same happen when running OctoPrint in safe mode?
Probably

#### Version of OctoPrint
1.3.10

#### Operating System running OctoPrint
OctoPi

#### Printer model & used firmware incl. version
XYZ Da Vinci 1.0A

#### Browser and version of browser, operating system running browser
Chrome, OSX High Sierra

There is a reference in the changelog for 1.3.10:
"Removed printed/visited layer counts from the GCODE viewer since it was confusing people more than helping them."

I do understand the removal of "visited" count, it doesn't make any sense to me, even after several years of using Octoprint.
However, the "printed' count (along with "current layer") was the most useful piece of information on the page.
I know the math isn't that hard, subtract the first layer height from the total height then divide by the normal layer height, fudge it a bit because it's not exact then add the first layer back in and subtract the number of layers printed and you have an approximation of the number of layers remaining. Of course, Slic3r's new variable layer height feature makes THAT impossible.

Could we have the total number of layers to print back? It was the "visited" count that was confusing.

foosel, 2018-12-14:
See bfd847f0a63658ab27b9f3e287c5361f11a670ba (`maintenance`, soon `devel`, to be released with 1.3.11)

biggreenogre, 2018-12-14:
Thank You!
Wow, you're good!

foosel, 2019-05-15:
1.3.11 was released yesterday.

closed 2019-05-15 by foosel