OctoPrint/OctoPrint #2955: action:resume don't work after M600 continue

Alissonverd, 2018-12-15:
#### What were you doing?
Changing filament with LCD2004 operations. (runout or M600)

1 - Start print;
2 - Cut or remove filament to trigger the runout sensor;
3 - Sensor triggered, firmware will auto park head.
4 - Wait filament runout "M600" proceduce;
6 - Replace filament;
7 - In LCD, set to purge and continue..
8 - Firmware will resume XYZ position.
**9 - Problem here: Octoprint will pause again.**

#### What did you expect to happen?
In step 9, Octoprint read //action:resume and don't pause again.. So, after firmware resume position and send action command, the printer continue print without need click in octoprint resume button.

#### What happened instead?
In step 8, firmware resume position head and send //action:resume, so, Octoprint pause print and wait user click in interface Resume button.

#### Did the same happen when running OctoPrint in safe mode?
Same problem.

#### Version of OctoPrint
OctoPrint 1.3.10 running on OctoPi 0.15.1

#### Operating System running OctoPrint
OctoPi 0.15.1

#### Printer model & used firmware incl. version
Marlin 2.0
FILAMENT_RUNOUT_SENSOR
NOZZLE_PARK_FEATURE
ADVANCED_PAUSE_FEATURE
EMERGENCY_PARSER
ACTION_ON_PAUSE "pause"
ACTION_ON_RESUME "resume"

#### Browser and version of browser, operating system running browser
Chrome and Edge - Windows 10

#### Link to contents of terminal tab or serial.log
Send: N5260 G1 X144.798 Y154.616 E58.21488*111
Recv: ok
Send: N5261 G1 X144.945 Y154.282 E58.22163*106
Recv: echo:enqueueing "M600"
Recv: //action:pause
Pausing on request of the printer...
Changing monitoring state from "Printing" to "Pausing"
Recv:  T:215.47 /215.00 B:100.18 /100.00 @:70 B@:43
Recv: echo:busy: processing
Recv:  T:214.88 /215.00 B:100.20 /100.00 @:81 B@:39
Recv: echo:busy: processing
Recv:  T:216.09 /215.00 B:100.00 /100.00 @:49 B@:71
Recv: echo:Insert filament and press button (or M108)
Recv: echo:busy: paused for user
Recv:  T:215.78 /215.00 B:99.93 /100.00 @:58 B@:83
Recv: echo:busy: processing
Recv:  T:213.19 /215.00 B:100.02 /100.00 @:105 B@:53
Recv: echo:busy: processing
Recv:  T:213.56 /215.00 B:100.01 /100.00 @:98 B@:63
**_Recv: //action:resume
Resuming on request of the printer..._**
Recv: ok
Send: N5262 M400*20
Recv: ok
Send: N5263 M114*21
Recv: X:144.94 Y:154.28 Z:0.40 E:58.22 Count A:47876 B:-1494 Z:686
Recv: ok
**_Send: N5264 M117 Print paused*86
Recv: ok
Changing monitoring state from "Pausing" to "Paused"_**
Recv:  T:214.06 /215.00 B:100.04 /100.00 @:89 B@:59
Recv:  T:214.19 /215.00 B:100.08 /100.00 @:90 B@:52
Recv:  T:214.69 /215.00 B:100.10 /100.00 @:83 B@:50

Alissonverd, 2018-12-16:
Dear @foosel , please check my modification on code, now works as expected.
To keep the behavior:
Changing monitoring state from "Printing" to "Pausing"
Changing monitoring state from "Pausing" to "Paused"

I added self._changeState(self.STATE_PAUSED) on end of this if.
Now the printer dont pause again after action:resume and work fine!

Could you add this modification in the next version?

Thank you

comm.py

> 			
			elif pause and self.isPrinting():
				if not self._pauseWaitStartTime:
					self._pauseWaitStartTime = time.time()

				**self._changeState(self.STATE_PAUSING)**

				if self.isSdFileSelected() and local_handling:
					self.sendCommand("M25",
					                 part_of_job=True,
					                 tags=tags | {"trigger:comm.set_pause", "trigger:pause"}) # pause print

				if not local_handling:
					with self._suppress_scripts_mutex:
						self._suppress_scripts.add("pause")

				def _on_M400_sent():
					# we don't call on_print_job_paused on our callback here
					# because we do this only after our M114 has been answered
					# by the firmware
					self._record_pause_data = True

					with self._pause_mutex:
						if self._pause_position_timer is not None:
							self._pause_position_timer.cancel()
							self._pause_position_timer = None
						self._pause_position_timer = ResettableTimer(self._timeout_intervals.get("positionLogWait", 10.0),
						                                             self._pause_preparation_failed)
						self._pause_position_timer.daemon = True
						self._pause_position_timer.start()
					self.sendCommand("M114",
					                 part_of_job=True,
					                 tags=tags | {"trigger:comm.set_pause",
					                              "trigger:pause",
					                              "trigger:record_position"})

				if self._log_position_on_pause:
					self.sendCommand("M400",
					                 on_sent=_on_M400_sent,
					                 part_of_job=True,
					                 tags=tags | {"trigger:comm.set_pause",
					                              "trigger:pause",
					                              "trigger:record_position"})
					self._continue_sending()
				else:
					self._pause_preparation_done(check_timer=False, suppress_script=not local_handling)

			**self._changeState(self.STATE_PAUSED)**

Alissonverd, 2018-12-16:
Update:
There is still strange behavior. In my first post, we have a Print paused after action:resume, and it still continues..
**Send: N5264 M117 Print paused*86**

So, I put in GCODES Scripts:
After print job is paused: M117 ### Pause script
Before print job is resumed: M117 ### Resume script

Whats happen with scripts?
Even though printing now automatically resumes, octoprint continues injecting a pause and runs the GCODE Scripts.
I could not figure out the source of this pause > N3621 M117 Print paused*85 
but I believe that after the action:resume, we should no longer receive these calls do pause.

> 2018-12-16 19:56:38,174 - Recv: //action:resume
2018-12-16 19:56:38,175 - Resuming on request of the printer...
2018-12-16 19:56:38,177 - Changing monitoring state from "Paused" to "Resuming"
2018-12-16 19:56:38,221 - Recv: ok
2018-12-16 19:56:38,225 - Send: N3621 M117 Print paused*85
2018-12-16 19:56:38,232 - Recv: ok
2018-12-16 19:56:38,276 - Send: N3624 M117 ### Pause script*73
2018-12-16 19:56:38,282 - Recv: ok
2018-12-16 19:56:38,284 - Changing monitoring state from "Resuming" to "Paused"
2018-12-16 19:56:38,295 - Send: M117 ### Resume script
2018-12-16 19:56:38,302 - Recv: ok
2018-12-16 19:56:38,305 - Changing monitoring state from "Paused" to "Printing"
2018-12-16 19:56:38,318 - Send: N3625 G1 X143.1 Y142.388 E88.14419*106
2018-12-16 19:56:38,330 - Recv: ok

foosel, 2018-12-17:
It's a bit more tricky than just adding a line.

The thing is that OctoPrint is still in the middle of processing the initial pause ("Printing -> Pausing") when the resume command arrives. The pause commands and subsequent pause however are already enqueued. So this needs handling, to abort further pause processing in case of a resume.

Alissonverd, 2018-12-17:
Thank you!

I try use action:resumed and action:paused on firmware.. Now dont execute the octoprint gcode scripts, but I receive the same Send: N3621 M117 Print paused85.. The printer still pausing after resume and need use Resume button of web interface.
The behavior is not 100% right.

Thank for your attention.

foosel, 2019-03-11:
On closer look I've decided against clearing the queue since it might cause issues in certain scenarios where the resume script defined by the user relies on things that have happened in the pause script.

I've done these things instead:

  1. Made sure that `PAUSING` state will also allow a resume to proceed, not just `PAUSED` as it was until now (commit c6b4eee)
  2. Only switch to `PAUSED` on end of pause handling if we were still in `PAUSING` (commit e748b52). 
  3. Only switch to `PRINTING` on end of resume handling if we were still in `RESUMING` (commit a1e8694).

2 allows an `action:resume` or `action:resumed` received during pausing to be enqueued. OctoPrint will continue through the pause script, then once that's all been sent proceed with the resume script.

3 does the same for `action:pause` or `action:paused` during resuming.

The combination of all of that should take care of the inconsistencies observed here, without causing potential trouble due to script dependencies.

Commits pushed to `maintenance`, soon `devel`, will be part of 1.3.11.

Alissonverd, 2019-03-11:
Thank you!
I will close this topic. Waiting for 1.3.11 release.

Best regards,
Alisson

commit e748b520 (2019-03-11): 🐛 Don't change to PAUSED state if not PAUSING

Fixes #2955
closed 2019-03-11 by Alissonverd